introlab/rtabmap_ros #169: exception during external odometry

jgoppert, 2017-05-05:
I'm currently using the PX4 autopilot as odometry input based on sensor_fusion.launch

```bash
terminate called after throwing an instance of 'UException'
  what():  [FATAL] (2017-05-04 20:29:43.009) Odometry.cpp:271::process() Condition (dt>0.0 || (dt == 0.0 && previousVelocityTransform_.isNull())) not met! [dt=-0.479163 previous transform=xyz=0.009769,-0.007867,-0.063735 rpy=-0.009064,-0.010757,0.000381]
================================================================================REQUIRED process [rtabmap/visual_odometry-20] has died!
process has died [pid 2427, exit code -6, cmd /opt/ros/kinetic/lib/rtabmap_ros/rgbd_odometry --delete_db_on_start rgb/image:=/camera/rgb/image_raw depth/image:=/camera/depth/image_raw rgb/camera_info:=/camera/rgb/camera_info odom:=/vo __name:=visual_odometry __log:=/home/jgoppert/.ros/log/9feda03e-3129-11e7-80da-3413e83564e8/rtabmap-visual_odometry-20.log].
log file: /home/jgoppert/.ros/log/9feda03e-3129-11e7-80da-3413e83564e8/rtabmap-visual_odometry-20*.log
Initiating shutdown!
```

I'm sending /vo from rtabmap to the px4 as a vision position estimate, and that is all working correctly. The dt  < 0 seems to suggest that even though I'm rerouting the vo to a different topic and subscribing to the output of mavros, that some of the vo messages are still being sent to rtabmap.

```xml
  <group ns="rtabmap">
    <!-- Visual Odometry -->
    <node pkg="rtabmap_ros" type="rgbd_odometry" name="visual_odometry" output="$(arg rtabmap_output)" args="$(arg rtabmap_args)" required="true">
      <remap from="rgb/image"       to="$(arg rgb_topic)"/>
      <remap from="depth/image"     to="$(arg depth_topic)"/>
      <remap from="rgb/camera_info" to="$(arg camera_info_topic)"/>
      <remap from="odom"            to="/vo"/>

      <param name="frame_id"               type="string" value="$(arg frame_id)"/>
      <param name="odom_frame_id"          type="string" value="$(arg odom_frame_id)"/>
      <param name="publish_tf"             type="bool"   value="false"/>
      <param name="publish_null_when_lost" type="bool"   value="false"/>
      <param name="guess_from_tf"          type="bool"   value="false"/>

      <param name="Odom/FillInfoData"      type="string" value="true"/>
      <param name="Odom/ResetCountdown"    type="string" value="1"/>
      <param name="Vis/FeatureType"        type="string" value="6"/> 
      <param name="OdomF2M/MaxSize"        type="string" value="1000"/>
    </node>
    <!-- SLAM -->
    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="$(arg rtabmap_output)" args="$(arg rtabmap_args)" required="true">
      <param name="frame_id"        type="string" value="$(arg frame_id)"/>
 
      <remap from="rgb/image"       to="$(arg rgb_topic)"/>
      <remap from="depth/image"     to="$(arg depth_topic)"/>
      <remap from="rgb/camera_info" to="$(arg camera_info_topic)"/>
      <remap from="odom"            to="/mavros/local_position/odom"/>
      
      <param name="Kp/DetectorStrategy"    type="string" value="6"/> <!-- use same features as odom -->

      <!-- localization mode -->
      <param name="Mem/InitWMWithAllNodes" type="string" value="$(arg localization)"/> 
      <param     if="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="false"/>
      <param unless="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="true"/>
      
    </node>
```

[image]

matlabbe, 2017-05-05:
Hi,

In your setup, `rtabmap` doesn't receive `/vo`. The error is coming from `rgbd_odometry` node. A [negative dt](https://github.com/introlab/rtabmap/blob/45ca3500c653436f3b559ba502c38439775719ed/corelib/src/Odometry.cpp#L271) means that `rgbd_odometry`'s input topics are received in wrong order, e.g., current frame has stamp older than previous frame.

The timestamp is taken from depth or rgb image (highest timestamp between both): https://github.com/introlab/rtabmap_ros/blob/ea859291ed8b690946265d2f9fcec436f58a02bf/src/nodelets/rgbd_odometry.cpp#L248

It may be a synchronization issue. If your rgb-d sensor outputs rgb/depth/camera_info with exactly the same timestamp, I recommend to use approx_sync=false:
```xml
<node pkg="rtabmap_ros" type="rgbd_odometry" name="visual_odometry" output="$(arg rtabmap_output)" args="$(arg rtabmap_args)" required="true">
      <remap from="rgb/image"       to="$(arg rgb_topic)"/>
      <remap from="depth/image"     to="$(arg depth_topic)"/>
      <remap from="rgb/camera_info" to="$(arg camera_info_topic)"/>
      <remap from="odom"            to="/vo"/>

      <param name="approx_sync" type="bool" value="false"/>
      ...
</node>
```

Otherwise, the commit above will just ignore the problem if odometry doesn't need `dt` for its computation (in your setup, `dt` is not used anyway).

cheers,
Mathieu

jgoppert, 2017-05-06:
Thanks for the info and removing the assert.

closed 2017-05-06 by jgoppert