introlab/rtabmap_ros #172: rgbd_odometry invalid Quaternion

Notou, 2017-05-11:
Hi Mathieu,

I am using the rgbd_odom module to compute the odometry.

Sometimes, I get this error: 
`[ERROR] [1493751218.599111632]: Ignoring transform for child_frame_id "cam0" from authority "unknown_publisher" because of an invalid quaternion in the transform (-0.062242 0.062456 -0.102315 0.990842)` (The quaternion in question changes when I move)
It goes in bursts as if it was seeing something it didn't like...

matlabbe, 2017-05-11:
What your TF tree looks like?
```bash
$ rosrun tf view_frames
```

Notou, 2017-05-12:
My tf tree looks fine: [frames.pdf](https://github.com/introlab/rtabmap_ros/files/995371/frames.pdf)
The map frame is called world, that's on purpose :) 

Here is my config file and my launch file just in case: [rtabmap.ini.txt](https://github.com/introlab/rtabmap_ros/files/995379/rtabmap.ini.txt) [rgbd_odom.launch.txt](https://github.com/introlab/rtabmap_ros/files/995384/rgbd_odom.launch.txt)

matlabbe, 2017-05-12:
With rqt_console, from which node is this error thrown? Maybe related to http://answers.ros.org/question/257498/rviz-ignoring-transforms/ or https://github.com/ccny-ros-pkg/imu_tools/issues/67

When rgbd_odometry is sending the pose [here](https://github.com/introlab/rtabmap_ros/blob/18d09f0405a2e626034feade52dda9c0333ecfc9/src/OdometryROS.cpp#L413) on TF, we may explicitly normalize the quaternion like https://github.com/ccny-ros-pkg/imu_tools/issues/67 before it is published. If this works, I may update the code of [transformToGeometryMsg](https://github.com/introlab/rtabmap_ros/blob/a92fac1ea3a39557b5fcae81f7fa2ccb09638c32/src/MsgConversion.cpp#L67-L77) to explicitly normalize the quaternion after conversion from eigen.

cheers,
Mathieu

Notou, 2017-05-15:
From rqt_console, the error is this:
```
Node: /rtabmap/rgbd_odometry
Time: 20:00:36.849946307 (2017-05-02)
Severity: Error
Published Topics: /rosout, /rtabmap/odom, /rtabmap/odom_info, /rtabmap/odom_last_frame, /rtabmap/odom_local_map, /rtabmap/odom_local_scan_map, /tf

Ignoring transform for child_frame_id "cam0" from authority "unknown_publisher" because of an invalid quaternion in the transform (-0.083852 -0.025377 0.768331 0.634037)

Location:
/tmp/binarydeb/ros-indigo-tf2-0.5.15/src/buffer_core.cpp::253
```

The issue might very well be that my tf2 is not up to date because with the values in error:
0.083852²+0.025377²+0.768331²+0.634037²= 1,000010593
This is not in the tolerance margin of 10^-6 that was put in January but since the rviz issue tf2 had a commit that raised the tolerance margin to 10^-3 so that should be good.

I'll check that my tf2 package is up to date and get back to you

Notou, 2017-05-19:
Okay, so the binaries don't seem to have been updated yet but I don't really want to risk screwing my setup reinstalling tf2 from source so I'll try to avoid it if I can.

I've edited my code so that I normalize the transform before publishing it:

```
		ROS_DEBUG("Pre normalisation, norm=%e", (poseMsg.transform.rotation.x * poseMsg.transform.rotation.x + poseMsg.transform.rotation.y * poseMsg.transform.rotation.y + poseMsg.transform.rotation.z * poseMsg.transform.rotation.z + poseMsg.transform.rotation.w * poseMsg.transform.rotation.w));
		long double recipNorm = 1 / sqrt(poseMsg.transform.rotation.x * poseMsg.transform.rotation.x + poseMsg.transform.rotation.y * poseMsg.transform.rotation.y + poseMsg.transform.rotation.z * poseMsg.transform.rotation.z + poseMsg.transform.rotation.w * poseMsg.transform.rotation.w);
		poseMsg.transform.rotation.x *= recipNorm;
		poseMsg.transform.rotation.y *= recipNorm;
		poseMsg.transform.rotation.z *= recipNorm;
		poseMsg.transform.rotation.w *= recipNorm;
		ROS_DEBUG("Post normalisation, norm=%e", (poseMsg.transform.rotation.x * poseMsg.transform.rotation.x + poseMsg.transform.rotation.y * poseMsg.transform.rotation.y + poseMsg.transform.rotation.z * poseMsg.transform.rotation.z + poseMsg.transform.rotation.w * poseMsg.transform.rotation.w));
```
I've added that [here](https://github.com/introlab/rtabmap_ros/blob/18d09f0405a2e626034feade52dda9c0333ecfc9/src/OdometryROS.cpp#L410) and it works, the quaternions get normalized properly:
```

[DEBUG] [1493775335.989783465]: Pre normalisation, norm=1.000008e+00
[DEBUG] [1493775335.990027299]: Post normalisation, norm=1.000000e+00
```

The pre nomalisation debug ligne showed me something that I haven't been able to explain:
The norm slowly increases with time, gaining about 10^-6 every 5-10s (not regular).
This is with the camera staying still on my desk.

This may be because of the uncertainties adding up somewhere in the code...

I will test what happens when the normalisation is put in the transformToGeometryMsg but I'm not sure which function I should put it in [transformToGeometryMsg](https://github.com/introlab/rtabmap_ros/blob/a92fac1ea3a39557b5fcae81f7fa2ccb09638c32/src/MsgConversion.cpp#L67-L77) or [transformFromGeometryMsg](https://github.com/introlab/rtabmap_ros/blob/a92fac1ea3a39557b5fcae81f7fa2ccb09638c32/src/MsgConversion.cpp#L80-L93)?
I guess it doesn't really matter if the goal here is to have regular normalisations...

commit 47e6c8df (2017-05-23): fixed #172
closed 2017-05-23 by matlabbe