introlab/rtabmap_ros #174: Image transform about Realsense R200 on UAV

hdudami, 2017-05-29:
Hello,
I want to ask a question, that is I want to use Realsense R200 camera on UAV, and run rtabmap_ros on my computer. I have a Rraspberry Pi on UAV. How can I get the camera node information in real time? Should I use wifi or a picture transmitter?

Looking forward for you answer!
Thank you

matlabbe, 2017-05-29:
Th ePi and your computer should be connected through ROS. If so, images can be streamed from the Pi to your computer via WiFi with normal topics. You may have to create [republish](http://wiki.ros.org/image_transport#republish) nodes on the computer to subscribe only once to Pi topics. 

See this tutorial for example: http://wiki.ros.org/rtabmap_ros/Tutorials/RemoteMapping

cheers,
Mathieu

hdudami, 2017-05-30:
Dear Mathieu
Thanks for your answer,

I do it as the tutorials use a kinect,and I connect successful!
I also have questions.  UAV fly faster than hand-held . If I use wifi ,the speed of stream and map create  can't follow the UAV .I may lose the image and map.  T^T
And the tutorials of Remote Mapping create a launch file ( freenect_throttle.launch )about kinect, if I use another RGB-D camera or stereo camera, can I use the same launch files?

Thanks

matlabbe, 2017-05-30:
Yes you can remap the topic accordingly to your RGB-D camera (e.g., realsense_throttle.launch). The frame rate can be changed to (e.g., 10, 15 Hz or more). Note that throttling may not be required if we can already set the maximum framerate of the realsense camera driver. Beware that increasing the frame rate would use more WiFi bandwidth, so if the network cannot sustain as much data, there would be problems.

cheers,
Mathieu

szx0112, 2017-07-04:
Hi, I am interested to know how is the performance of using r200 on UAV for outdoor environment, willing to know how is your test (If you did). It looks like r200 works best at short distance, and you mentioned about the speed of flight compare to the mapping process can be another issue.