introlab/rtabmap_ros #176: Rtabmap_ros

trabbani12, 2017-06-01:
I'm running Rtabmap_Ros on an Nvidia Jetson Tx2 with a Realsense ZR300. 

I keep getting this warning and I'm not quite sure how to fix it. I'm new to ROS so this might be a simple fix, I'm not sure. I've done everything according to the ROS Wiki Tutorial. These are the terminal commands and outputs of each terminal:

Terminal 1:
[image]
[image]

Terminal 2: 
[image]
[image]

Am I using the incorrect terminal commands? 

Thanks!

matlabbe, 2017-06-06:
Please copy/paste text instead of taking screenshot, it is hard to refer at some lines afterward. It seems odometry started but somewhat stopped to run. Verify that camera topics are still published ( at high Hz):
```bash
$ rostopic hz /camera/rgb/image_rect_color
$ rostopic hz /camera/depth_registered/sw_registered/image_rect_raw
$ rostopic hz /camera/rgb/camera_info
```
cheers

kammce, 2017-07-17:
When I run each of these commands I receive "no new message" from rostopic.

kammce, 2017-07-17:
I get **no new message** for each of those rostopic hz commands. I am using the commands found

matlabbe, 2017-07-19:
That means the realsense driver node is not running. Make sure the sensor is plugged (on usb3 idealy).