introlab/rtabmap_ros #179: Remote Mapping error about openni

hdudami, 2017-06-13:
Hello,
Today I use an odroid xu4 on my robot, to connect a kinect1. My odroid is ubunt mate 16.04 and ros kinetic. I do $sudo apt-get install ros-kinetic-freenect-*    And I also install OpenNI for arm .When I do as the tutorial about Remote Mapping:  $roslaunch freenect_throttle.launch rate:=5  I got the following errors:

[ERROR] [1455208288.833791480]: Failed to load nodelet [/camera/data_throttle] of type [rtabmap_ros/data_throttle] even after refreshing the cache: Failed to load library /home/odroid/catkin_ws/devel/lib//librtabmap_ros.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libOpenNI.so.0: cannot open shared object file: No such file or directory)
[ERROR] [1455208288.834015357]: The error before refreshing the cache was: Failed to load library /home/odroid/catkin_ws/devel/lib//librtabmap_ros.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libOpenNI.so.0: cannot open shared object file: No such file or directory)

I search my odroid and I don not find the file "libOpenNI.so.0" I think is that my odroid is arm, 32bits, so I can't do as computer : $sudo apt-get install ros-kinetic-openni-launch. How can I solve this problem?

Looking forward for your answer,
Thanks!

matlabbe, 2017-06-13:
Hi, 

Is `rtabmap` library built from source? If you do "`$ rtabmap`" alone, is rtabmap crash on start? `libOpenNI.so.0` would be a dependency coming from PCL library. Either PCL installed on Odroid is not built with OpenNI and that kinetic binaries of rtabmap is built thinking it has, or you may try to rebuild PCL without OpenNI, then build rtabmap against this PCL version. 

cheers,
Mathieu

hdudami, 2017-06-15:
Hi Mathieu,

I reinstall ros-kinetic-full and reinstall pcl with the following step:
```javascript
$ sudo apt-get install ros-kinetic-desktop-full
$ sudo apt-get install libpcl-dev pcl-tools
```
And remove OpenNI ,then I rebuild rtabmap has the following errors:

```javascript
In file included from /usr/include/ni/XnOS.h:27:0,
                 from /usr/include/pcl-1.7/pcl/io/openni_camera/openni.h:47,
                 from /usr/include/pcl-1.7/pcl/io/openni_camera/openni_depth_image.h:44,
                 from /home/odroid/rtabmap/corelib/src/../include/rtabmap/core/CameraRGBD.h:42,
                 from /home/odroid/rtabmap/corelib/src/CameraThread.cpp:31:
/usr/include/ni/XnPlatform.h:73:3: error: #error OpenNI Platform Abstraction Layer - Unsupported Platform!
  #error OpenNI Platform Abstraction Layer - Unsupported Platform!
   ^
In file included from /usr/include/pcl-1.7/pcl/io/openni_camera/openni.h:47:0,
                 from /usr/include/pcl-1.7/pcl/io/openni_camera/openni_depth_image.h:44,
                 from /home/odroid/rtabmap/corelib/src/../include/rtabmap/core/CameraRGBD.h:42,
                 from /home/odroid/rtabmap/corelib/src/CameraThread.cpp:31:
```
When I search the file "libOpenNI.so.0", I can't find it either.Should I must install OpenNI? Or can I instead it with libopenni2?

Thanks,
Fan

matlabbe, 2017-06-15:
Openni is installed with libpcl-dev, but it does support the platform:
```
/usr/include/ni/XnPlatform.h:73:3: error: #error OpenNI Platform Abstraction Layer - Unsupported Platform!
  #error OpenNI Platform Abstraction Layer - Unsupported Platform!
```
You will have to uninstall openni and libpcl-dev, then rebuild form source PCL without openni dependency.