introlab/rtabmap_ros #180: error while rtabmapviz tries to load librtabmap_core.so

arogowiec, 2017-06-28:
Hi! 
This is probably a bug introduced by new `rtabmap` release (13.0). When I try to launch `rtabmapviz` node I got following error:
> error while loading shared libraries: librtabmap_core.so.0.12: cannot open shared object file: No such file or directory

which causes `rtabmapviz` node to die.

matlabbe, 2017-06-28:
Did you update/rebuild rtabmap_ros too? You may also need to do remove the CMakeCache.txt in your catkin build folder before doing catkin_make to make sure cmake links on 0.13 and not 0.12 libraries.

arogowiec, 2017-06-29:
I updated `rtabmap_ros` too. Moreover I've carefully removed all files and directories from previous `rtabmap` versions. Next I've rebuilded `rtabmap`, installed it under `/usr/local/` directory; removed `build` and `devel` folders from my catkin workspace and rebuilded it entirely. However now I'm facing following warning from `CMake`:
> WITH rviz
> CMake Warning at rtabmap_ros/CMakeLists.txt:318 (MESSAGE):
>   Found RTAB-Map built without its GUI library.  Node rtabmapviz will not be
>   built!

while trying to build my catkin workspace.

Furthermore I've noticed that I need to append rtabmap installed libraries directory to environment variable `LD_LIBRARY_PATH` in order to run standalone app, since without this I got errors that `rtabmap` can't load `librtabmap_gui.so.0.13` shared library.

matlabbe, 2017-06-29:
For the warning telling rtabmapviz will be not built, that would be fixed by https://github.com/introlab/rtabmap/commit/5646751f885a716820eb1a96f7e8651b1fa5874b

For LD_LIBRARY_PATH, there is a comment on this page: https://github.com/introlab/rtabmap_ros#ros-distribution

However, I think running "sudo ldconfig" or closing/reopen terminal solve this problem too.

cheers,
Mathieu

arogowiec, 2017-07-03:
Thanks a lot! Now it all works well! ;)

cheers,
Adam

closed 2017-07-03 by arogowiec