introlab/rtabmap_ros #181: Can not get feasible results:  /rtabmap/rgbd_odometry: Did not receive data since 5 seconds!

szx0112, 2017-07-01:
I follow the tutorial 1 using r200 with rtabmap. The warn says it cannot receive three messages in topic:

   /camera/rgb/image_rect_color,
   /camera/depth_registered/image_raw,
   /camera/rgb/camera_info

I use rostopic echo and find they are all empty. My realsense_ros package works fine and the rtabmap stand alone application also works fine. Any comment is appreciate.

SUMMARY
========

PARAMETERS
 * /points_xyzrgb/approx_sync: True
 * /points_xyzrgb/decimation: 2.0
 * /points_xyzrgb/voxel_size: 0.02   /camera/rgb/image_rect_color,
   /camera/depth_registered/image_raw,
   /camera/rgb/camera_info
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /rtabmap/rgbd_odometry/approx_sync: True
 * /rtabmap/rgbd_odometry/config_path: 
 * /rtabmap/rgbd_odometry/frame_id: camera_link
 * /rtabmap/rgbd_odometry/ground_truth_base_frame_id: 
 * /rtabmap/rgbd_odometry/ground_truth_frame_id: 
 * /rtabmap/rgbd_odometry/odom_frame_id: odom
 * /rtabmap/rgbd_odometry/queue_size: 10
 * /rtabmap/rgbd_odometry/wait_for_transform_duration: 0.2
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/approx_sync: True
 * /rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/database_path: ~/.ros/rtabmap.db
 * /rtabmap/rtabmap/frame_id: camera_link
 * /rtabmap/rtabmap/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/ground_truth_frame_id: 
 * /rtabmap/rtabmap/map_frame_id: map
 * /rtabmap/rtabmap/odom_frame_id: 
 * /rtabmap/rtabmap/odom_tf_angular_variance: 1.0
 * /rtabmap/rtabmap/odom_tf_linear_variance: 1.0
 * /rtabmap/rtabmap/queue_size: 10
 * /rtabmap/rtabmap/subscribe_depth: True
 * /rtabmap/rtabmap/subscribe_scan: False
 * /rtabmap/rtabmap/subscribe_scan_cloud: False
 * /rtabmap/rtabmap/subscribe_stereo: False
 * /rtabmap/rtabmap/subscribe_user_data: False
 * /rtabmap/rtabmap/wait_for_transform_duration: 0.2

NODES
  /rtabmap/
    rgbd_odometry (rtabmap_ros/rgbd_odometry)
    rtabmap (rtabmap_ros/rtabmap)
  /
    points_xyzrgb (nodelet/nodelet)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[rtabmap/rgbd_odometry-1]: started with pid [17193]
process[rtabmap/rtabmap-2]: started with pid [17194]
process[rviz-3]: started with pid [17195]
process[points_xyzrgb-4]: started with pid [17196]
[ INFO] [1498869693.656653906]: Starting node...
[ INFO] [1498869693.711844985]: Initializing nodelet with 4 worker threads.
[ INFO] [1498869693.748123608]: Initializing nodelet with 4 worker threads.
[ INFO] [1498869695.304191510]: /rtabmap/rtabmap(maps): map_filter_radius          = 0.000000
[ INFO] [1498869695.304275364]: /rtabmap/rtabmap(maps): map_filter_angle           = 30.000000
[ INFO] [1498869695.304322863]: /rtabmap/rtabmap(maps): map_cleanup                = true
[ INFO] [1498869695.304361353]: /rtabmap/rtabmap(maps): map_negative_poses_ignored = false
[ INFO] [1498869695.304397029]: /rtabmap/rtabmap(maps): map_negative_scan_ray_tracing = true
[ INFO] [1498869695.304434008]: /rtabmap/rtabmap(maps): cloud_output_voxelized     = true
[ INFO] [1498869695.304469789]: /rtabmap/rtabmap(maps): cloud_subtract_filtering   = false
[ INFO] [1498869695.304512914]: /rtabmap/rtabmap(maps): cloud_subtract_filtering_min_neighbors = 2
[ INFO] [1498869695.311121404]: /rtabmap/rtabmap(maps): octomap_tree_depth         = 16
[ INFO] [1498869695.311212862]: /rtabmap/rtabmap(maps): octomap_occupancy_thr      = 0.500000
[ INFO] [1498869695.470692189]: rtabmap: frame_id      = camera_link
[ INFO] [1498869695.470800209]: rtabmap: map_frame_id  = map
[ INFO] [1498869695.470855001]: rtabmap: tf_delay      = 0.050000
[ INFO] [1498869695.470897656]: rtabmap: tf_tolerance  = 0.100000
[ INFO] [1498869695.470937187]: rtabmap: odom_sensor_sync   = false
[ INFO] [1498869696.500273265]: Setting RTAB-Map parameter "Mem/IncrementalMemory"="true"
[ INFO] [1498869696.503469959]: Setting RTAB-Map parameter "Mem/InitWMWithAllNodes"="false"
[ INFO] [1498869697.379035363]: 
/rtabmap/rgbd_odometry subscribed to (approx sync):
   /camera/rgb/image_rect_color,
   /camera/depth_registered/image_raw,
   /camera/rgb/camera_info
[ INFO] [1498869699.178752511]: RTAB-Map detection rate = 1.000000 Hz
[ INFO] [1498869699.179113186]: rtabmap: Deleted database "/home/ubuntu/.ros/rtabmap.db" (--delete_db_on_start or -d are set).
[ INFO] [1498869699.179177977]: rtabmap: Using database from "/home/ubuntu/.ros/rtabmap.db".
[ INFO] [1498869699.580797342]: rtabmap: Database version = "0.13.0".
[ INFO] [1498869699.696574303]: /rtabmap/rtabmap: queue_size    = 10
[ INFO] [1498869699.696664354]: /rtabmap/rtabmap: rgbd_cameras = 1
[ INFO] [1498869699.696714406]: /rtabmap/rtabmap: approx_sync   = true
[ INFO] [1498869699.696895238]: Setup depth callback
[ INFO] [1498869699.964988926]: 
/rtabmap/rtabmap subscribed to (approx sync):
   /rtabmap/odom,
   /camera/rgb/image_rect_color,
   /camera/depth_registered/image_raw,
   /camera/rgb/camera_info
[ INFO] [1498869699.991171378]: rtabmap 0.13.0 started...
[ WARN] [1498869702.380134138]: /rtabmap/rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. 
/rtabmap/rgbd_odometry subscribed to (approx sync):
   /camera/rgb/image_rect_color,
   /camera/depth_registered/image_raw,
   /camera/rgb/camera_info
[ WARN] [1498869704.965805418]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmap subscribed to (approx sync):
   /rtabmap/odom,
   /camera/rgb/image_rect_color,
   /camera/depth_registered/image_raw,
   /camera/rgb/camera_info
[ WARN] [1498869707.380826367]: /rtabmap/rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. 
/rtabmap/rgbd_odometry subscribed to (approx sync):
   /camera/rgb/image_rect_color,
   /camera/depth_registered/image_raw,
   /camera/rgb/camera_info

matlabbe, 2017-07-01:
Hi, 

in the [tutorial](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping), for r200, the depth image should be remapped to `/camera/depth_registered/sw_registered/image_rect_raw`:
```bash
$ roslaunch realsense_camera r200_nodelet_rgbd.launch
$ roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start" depth_topic:=/camera/depth_registered/sw_registered/image_rect_raw
```
cheers,
Mathieu

szx0112, 2017-07-05:
It works. Thank you. I have a followup question: I find that the performance of mapping during my test is pretty poor, not as smooth as in the demo video. For example, I have to keep very very very .... slow movement in order to not get the RED background (which is lost tracking I believe), slight move a little fast will cause errors/warns happens. I think it may related to camera setup, so I did some camera configuration but up to now no good results. My goal is to attach it on small robots for indoor/outdoor mappings. Could you give me some hints?

matlabbe, 2017-07-05:
You can try the second approach to generate a depth image. Also you can use Vis/EstimationType=1:

```bash
$ roslaunch realsense_camera r200_nodelet_rgbd.launch

# To get registration without empty lines
$ rosrun rtabmap_ros pointcloud_to_depthimage cloud:=/camera/depth/points camera_info:=/camera/rgb/camera_info image_raw:=/camera/depth/points/image_raw image:=/camera/depth/points/image _approx:=false _fill_holes_size:=2

 # Using depth registered from `rtabmap_ros/pointcloud_to_depthimage`
 $ roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start --Vis/EstimationType 1" depth_topic:=/camera/depth/points/image_raw
```

cheers,
Mathieu

szx0112, 2017-07-08:
Thanks for your reply. I play around with the second approach, it indeed increase the robustness of movement. However, I find the map in rtabmapviz does not registered. Pics below show screenshots with slowly rotating r200.

[image]
[image]

I put the camera with laptop on a chart so it can move at regularly speed. One problem may related to the features in images, if there is a flat blank image (e.g. a white wall or a empty space)  in the middle of the video stream, the rtabmapviz will get stuck (background turn in red). 

I find this #102, which is similar to what I am thinking out. So I add those parameters into rtabmap.launch and it looks better now, but still the map does not registered.

My goal is to finally use it for outdoor mappings like this video: [](https://www.youtube.com/watch?v=Mnn-fpEBtBk&t=65s) with a ground vehicle or UAV. Is there any specific setting for this applications?

Thanks,

matlabbe, 2017-07-10:
See [Lost Odometry](https://github.com/introlab/rtabmap/wiki/Kinect-mapping#lost-odometry-red-screens) section on this page for the red backgrounds.

Looking at the depth image overlay in your screenshot, the sensor seems to have very low range and is very sparse. I tried my example above with "--GFTT/QualityLevel 0.001 --GFTT/MinDistance 3 --Kp/DetectorStrategy 6" to get more features extracted for the low resolution images, thus getting more matches (odometry quality around 200).

In the video you are referring, they may activated auto odometry reset (e.g., "--Odom/ResetCountdown 1") after a number of tracking failure(s). A new map is created when this happens, a relocation with the previous maps is required to merge them.

cheers,
Mathieu

szx0112, 2017-07-11:
I think I need to tune the parameters in r200_ros first to have it capture good depth images because I tried the RTAB-MAP standalone application and it works just fine. 

Thanks,

szx0112, 2017-07-12:
Hi, I am still wondering in the pic above, the loop closure detection window is empty.. Is it happens because of the bad image quality?

Thanks,

matlabbe, 2017-07-13:
Oh I missed that, that means rtabmap node is not running or rtabmapviz doesn't receive mapData from rtabmap node. What is the output on your terminal? You should see rtabmap update log each second.

szx0112, 2017-07-13:
Thanks for reply. I get the following log when I put camera towards a good feature scene. It doesn't present any warning/errors but still no map data I think. I am currently update the version of both rtabmap and rtabmap_ros to the newest, not sure if version mismatch or old version will have any issue.

[ INFO] [1499907708.317757257]: Odom: quality=82, std dev=0.003461m|0.003461rad, update time=0.124020s
[ INFO] [1499907708.446787519]: Odom: quality=82, std dev=0.003702m|0.003702rad, update time=0.123123s
[ INFO] [1499907708.566025239]: Odom: quality=82, std dev=0.003638m|0.003638rad, update time=0.114031s
[ INFO] [1499907708.718590779]: Odom: quality=91, std dev=0.003096m|0.003096rad, update time=0.133870s
[ INFO] [1499907708.862339854]: Odom: quality=94, std dev=0.003511m|0.003511rad, update time=0.136458s
[ INFO] [1499907709.008709823]: Odom: quality=74, std dev=0.004209m|0.004209rad, update time=0.128491s
[ INFO] [1499907709.149404628]: Odom: quality=85, std dev=0.003791m|0.003791rad, update time=0.128264s
[ INFO] [1499907709.283244914]: Odom: quality=82, std dev=0.002906m|0.002906rad, update time=0.128219s
[ INFO] [1499907709.414867581]: Odom: quality=82, std dev=0.002297m|0.002297rad, update time=0.126156s
[ INFO] [1499907709.539746977]: Odom: quality=98, std dev=0.003699m|0.003699rad, update time=0.120313s
[ INFO] [1499907709.693255159]: Odom: quality=83, std dev=0.003438m|0.003438rad, update time=0.146680s
[ INFO] [1499907709.841793737]: Odom: quality=92, std dev=0.003565m|0.003565rad, update time=0.140008s
[ INFO] [1499907709.987828504]: Odom: quality=89, std dev=0.003746m|0.003746rad, update time=0.134824s
[ INFO] [1499907710.122390966]: Odom: quality=86, std dev=0.003361m|0.003361rad, update time=0.128837s
[ INFO] [1499907710.251102017]: Odom: quality=93, std dev=0.004132m|0.004132rad, update time=0.123693s
[ INFO] [1499907710.374921640]: Odom: quality=96, std dev=0.003266m|0.003266rad, update time=0.119237s
[ INFO] [1499907710.529949954]: Odom: quality=76, std dev=0.003981m|0.003981rad, update time=0.146946s
[ INFO] [1499907710.663000567]: Odom: quality=88, std dev=0.003235m|0.003235rad, update time=0.124925s
[ INFO] [1499907710.796148886]: Odom: quality=90, std dev=0.003153m|0.003153rad, update time=0.122416s
[ INFO] [1499907710.923096217]: Odom: quality=83, std dev=0.003814m|0.003814rad, update time=0.121002s
[ INFO] [1499907711.052248354]: Odom: quality=93, std dev=0.003522m|0.003522rad, update time=0.123568s
[ INFO] [1499907711.185290947]: Odom: quality=73, std dev=0.003215m|0.003215rad, update time=0.127659s
[ INFO] [1499907711.344675439]: Odom: quality=79, std dev=0.003280m|0.003280rad, update time=0.153769s
[ INFO] [1499907711.516259776]: Odom: quality=88, std dev=0.002645m|0.002645rad, update time=0.165715s
[ INFO] [1499907711.654496968]: Odom: quality=83, std dev=0.003888m|0.003888rad, update time=0.133673s
[ INFO] [1499907711.790736851]: Odom: quality=93, std dev=0.003655m|0.003655rad, update time=0.131395s
[ INFO] [1499907711.941576904]: Odom: quality=81, std dev=0.003265m|0.003265rad, update time=0.145210s
[ INFO] [1499907712.078745782]: Odom: quality=86, std dev=0.002939m|0.002939rad, update time=0.127173s
[ INFO] [1499907712.218606646]: Odom: quality=86, std dev=0.003139m|0.003139rad, update time=0.134730s
[ INFO] [1499907712.379276430]: Odom: quality=82, std dev=0.002938m|0.002938rad, update time=0.153559s
[ INFO] [1499907712.543232407]: Odom: quality=86, std dev=0.003008m|0.003008rad, update time=0.158737s
[ INFO] [1499907712.713722389]: Odom: quality=86, std dev=0.002995m|0.002995rad, update time=0.163338s

Thanks,

matlabbe, 2017-07-13:
Looks like only odometry is running. What is the top output? (rtabmap initialization stuff). What commands did your use?

szx0112, 2017-07-13:
I follow the second method mentioned above (first method actually shows similar results), I can find there is a error related to stereo image (in bold) when a run the rtabmap.

**first run**
**roslaunch realsense_camera r200_nodelet_rgbd.launch**
... logging to /home/ubuntu/.ros/log/666d477a-676a-11e7-af12-00044b65bf8e/roslaunch-tegra-ubuntu-19767.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:47779/

SUMMARY

PARAMETERS
 * /camera/camera_nodelet_manager/num_worker_threads: 4
 * /camera/depth_rectify_depth/interpolation: 0
 * /camera/disparity_depth/max_range: 4.0
 * /camera/disparity_depth/min_range: 0.5
 * /camera/disparity_registered_sw/max_range: 4.0
 * /camera/disparity_registered_sw/min_range: 0.5
 * /camera/driver/base_frame_id: camera_link
 * /camera/driver/camera_type: R200
 * /camera/driver/color_fps: 30
 * /camera/driver/color_frame_id: camera_rgb_frame
 * /camera/driver/color_height: 480
 * /camera/driver/color_optical_frame_id: camera_rgb_optica...
 * /camera/driver/color_width: 640
 * /camera/driver/depth_fps: 30
 * /camera/driver/depth_frame_id: camera_depth_frame
 * /camera/driver/depth_height: 360
 * /camera/driver/depth_optical_frame_id: camera_depth_opti...
 * /camera/driver/depth_width: 480
 * /camera/driver/enable_color: True
 * /camera/driver/enable_depth: True
 * /camera/driver/enable_pointcloud: False
 * /camera/driver/enable_tf: True
 * /camera/driver/ir2_frame_id: camera_ir2_frame
 * /camera/driver/ir_frame_id: camera_ir_frame
 * /camera/driver/mode: manual
 * /camera/driver/serial_no: 
 * /camera/driver/usb_port_id: 
 * /camera/points_xyzrgb_sw_registered/queue_size: 100
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /camera/
    camera_nodelet_manager (nodelet/nodelet)
    depth_metric (nodelet/nodelet)
    depth_metric_rect (nodelet/nodelet)
    depth_points (nodelet/nodelet)
    depth_rectify_depth (nodelet/nodelet)
    depth_registered_sw_metric_rect (nodelet/nodelet)
    disparity_depth (nodelet/nodelet)
    disparity_registered_sw (nodelet/nodelet)
    driver (nodelet/nodelet)
    ir_rectify_ir (nodelet/nodelet)
    points_xyzrgb_sw_registered (nodelet/nodelet)
    register_depth_rgb (nodelet/nodelet)
    rgb_debayer (nodelet/nodelet)
    rgb_rectify_color (nodelet/nodelet)
    rgb_rectify_mono (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [19778]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 666d477a-676a-11e7-af12-00044b65bf8e
process[rosout-1]: started with pid [19791]
started core service [/rosout]
[... 109 lines cut ...]
[ INFO] [1499909318.621785431]: /rtabmap/rtabmap(maps): map_negative_scan_ray_tracing = true
[ INFO] [1499909318.621822410]: /rtabmap/rtabmap(maps): cloud_output_voxelized     = true
[ INFO] [1499909318.621859024]: /rtabmap/rtabmap(maps): cloud_subtract_filtering   = false
[ INFO] [1499909318.621894440]: /rtabmap/rtabmap(maps): cloud_subtract_filtering_min_neighbors = 2
[ INFO] [1499909318.666818703]: /rtabmap/rtabmap(maps): octomap_tree_depth         = 16
[ INFO] [1499909318.666918285]: /rtabmap/rtabmap(maps): octomap_occupancy_thr      = 0.500000
[ INFO] [1499909319.269702990]: rtabmap: frame_id      = camera_link
[ INFO] [1499909319.269834811]: rtabmap: map_frame_id  = map
[ INFO] [1499909319.269897570]: rtabmap: tf_delay      = 0.050000
[ INFO] [1499909319.269941216]: rtabmap: tf_tolerance  = 0.100000
[ INFO] [1499909319.269980746]: rtabmap: odom_sensor_sync   = false
**[points_xyzrgb-4] process has died [pid 21338, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet standalone rtabmap_ros/point_cloud_xyzrgb left/image:=/stereo_camera/left/image_rect_color right/image:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info right/camera_info:=/stereo_camera/right/camera_info rgb/image:=/camera/rgb/image_rect_color depth/image:=/camera/depth/points/image_raw rgb/camera_info:=/camera/rgb/camera_info cloud:=voxel_cloud __name:=points_xyzrgb __log:=/home/ubuntu/.ros/log/666d477a-676a-11e7-af12-00044b65bf8e/points_xyzrgb-4.log].**
log file: /home/ubuntu/.ros/log/666d477a-676a-11e7-af12-00044b65bf8e/points_xyzrgb-4*.log
[ INFO] [1499909323.682309003]: Setting RTAB-Map parameter "Mem/IncrementalMemory"="true"
[ INFO] [1499909323.687516123]: Setting RTAB-Map parameter "Mem/InitWMWithAllNodes"="false"
[ INFO] [1499909324.428584874]: Update odometry parameter "Vis/EstimationType"="1" from arguments
[ INFO] [1499909326.229903979]: 
/rtabmap/rgbd_odometry subscribed to (approx sync):
   /camera/rgb/image_rect_color,
   /camera/depth/points/image_raw,
   /camera/rgb/camera_info
[ INFO] [1499909326.624039665]: Odom: quality=0, std dev=99.995000m|99.995000rad, update time=0.102322s
[ INFO] [1499909326.726850210]: Odom: quality=122, std dev=0.003533m|0.003533rad, update time=0.095916s
[ INFO] [1499909326.865204828]: Odom: quality=122, std dev=0.003686m|0.003686rad, update time=0.129443s
[ INFO] [1499909327.005541034]: Odom: quality=118, std dev=0.003326m|0.003326rad, update time=0.127595s

matlabbe, 2017-07-13:
There are missing logs about to which topics rtabmap node is subscribing (they can appear while the odometry is running). Can you do also after?
```bash
$ rostopic hz /rtabmap/mapData
```

For the points_xyzrgb nodelet error, I'll check tomorrow. It is started only if `"rviz:=true`" argument is set when launching `rtabmap.launch`, though you don't set it !?!

szx0112, 2017-07-13:
Oh, I changed enable rviz as default and forget to make it back. The topic /rtabmap/mapData shows no message:

ubuntu@tegra-ubuntu:~$ rostopic hz /rtabmap/mapData
subscribed to [/rtabmap/mapData]
no new messages
no new messages
no new messages
no new messages
no new messages
...

Changing rtabmapviz back also presents the error:

[ INFO] [1499911722.464751217]: Odom: quality=100, std dev=0.003636m|0.003636rad, update time=0.118849s
**[rtabmap/rtabmap-2] process has died [pid 1514, exit code -11, cmd /home/ubuntu/catkin_ws/devel/lib/rtabmap_ros/rtabmap rgb/image:=/camera/rgb/image_rect_color depth/image:=/camera/depth_registered/sw_registered/image_rect rgb/camera_info:=/camera/rgb/camera_info left/image_rect:=/stereo_camera/left/image_rect_color right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info right/camera_info:=/stereo_camera/right/camera_info scan:=/scan scan_cloud:=/scan_cloud user_data:=/user_data user_data_async:=/user_data_async __name:=rtabmap __log:=/home/ubuntu/.ros/log/0ccf0c0c-6770-11e7-88c3-00044b65bf8e/rtabmap-rtabmap-2.log].
log file: /home/ubuntu/.ros/log/0ccf0c0c-6770-11e7-88c3-00044b65bf8e/rtabmap-rtabmap-2*.log**
[ INFO] [1499911722.605778890]: Odom: quality=98, std dev=0.003928m|0.003928rad, update time=0.130766s

I attached the screen shot of rtabmapviz, rqt_graph and tf_views below:

[image]

[image]

[image]

matlabbe, 2017-07-13:
rtabmap node is crashing on start, rtabmapviz looks fine. Is there any error/fatal/assert message before "`[rtabmap/rtabmap-2] process has died`"?

Also, can you open the standalone application without crash?
```bash
$ rtabmap
```

To debug more, you can add "`launch_prefix:="xterm -e gdb -ex run --args"`" when launching `rtabmap.launch`. You will see gdb terminals for rtabmap, rgbd_odometry and rtabmapviz. Look in the one of rtabmap node, when it crashes, type "bt" (for backtrace) then copy paste the log here.

cheers,
Mathieu

szx0112, 2017-07-13:
The standalone applications works as it supposed to be.

Below lists the logging of gdb backtrace of rtabmap node. I also put a screen shot of rtabmap node running on gdb for reference. Let me know if I provide the wrong info. Thanks.

[image]

#0  __GI___libc_free (mem=0x200000001) at malloc.c:2951
#1  0x0000007fb79ebaa0 in rtabmap::util3d::voxelize(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGB> > const&, boost::shared_ptr<std::vector<int, std::allocator<int> > > const&, float) () from /usr/local/lib/librtabmap_core.so.0.13
#2  0x0000007fb7b20888 in pcl::PointCloud<pcl::PointXYZRGB>::Ptr rtabmap::OccupancyGrid::segmentCloud<pcl::PointXYZRGB>(pcl::PointCloud<pcl::PointXYZRGB>::Ptr const&, boost::shared_ptr<std::vector<int, std::allocator<int> > > const&, rtabmap::Transform const&, cv::Point3_<float> const&, boost::shared_ptr<std::vector<int, std::allocator<int> > >&, boost::shared_ptr<std::vector<int, std::allocator<int> > >&, boost::shared_ptr<std::vector<int, std::allocator<int> > >*) const () from /usr/local/lib/librtabmap_core.so.0.13
#3  0x0000007fb7b1ef08 in rtabmap::OccupancyGrid::createLocalMap(rtabmap::Signature const&, cv::Mat&, cv::Mat&, cv::Point3_<float>&) const () from /usr/local/lib/librtabmap_core.so.0.13
#4  0x0000007fb790a1d8 in rtabmap::Memory::createSignature(rtabmap::SensorData const&, rtabmap::Transform const&, rtabmap::Statistics*) () from /usr/local/lib/librtabmap_core.so.0.13
#5  0x0000007fb790d1f4 in rtabmap::Memory::update(rtabmap::SensorData const&, rtabmap::Transform const&, cv::Mat const&, std::vector<float, std::allocator<float> > const&, rtabmap::Statistics*) () from /usr/local/lib/librtabmap_core.so.0.13
#6  0x0000007fb78d052c in rtabmap::Rtabmap::process(rtabmap::SensorData const&, rtabmap::Transform, cv::Mat const&, std::vector<float, std::allocator<float> > const&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, float, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, float> > > const&) () from /usr/local/lib/librtabmap_core.so.0.13
#7  0x0000007f860a89ac in rtabmap_ros::CoreWrapper::process(ros::Time const&, rtabmap::SensorData const&, rtabmap::Transform const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, cv::Mat const&) ()
   from /home/ubuntu/catkin_ws/devel/lib//librtabmap_ros.so
#8  0x0000007f860a66d4 in rtabmap_ros::CoreWrapper::commonDepthCallbackImpl(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, boost::shared_ptr<rtabmap_ros::UserData_<std::allocator<void> > const> const&, std::vector<boost::shared_ptr<cv_bridge::CvImage const>, std::allocator<boost::shared_ptr<cv_bridge::CvImage const> > > const&, std::vector<boost::shared_ptr<cv_bridge::CvImage const>, std::allocator<boost::shared_ptr<cv_bridge::CvImage const> > > const&, std::vector<sensor_msgs::CameraInfo_<std::allocator<void> >, std::allocator<sensor_msgs::CameraInfo_<std::allocator<void> > > > const&, boost::shared_ptr<sensor_msgs::LaserScan_<std::allocator<void> > const> const&, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&, boost::shared_ptr<rtabmap_ros::OdomInfo_<std::allocator<void> > const> const&) () from /home/ubuntu/catkin_ws/devel/lib//librtabmap_ros.so
#9  0x0000007f860a507c in rtabmap_ros::CoreWrapper::commonDepthCallback(boost::shared_ptr<nav_msgs::Odometry_<std::allocator<void> > const> const&, boost::shared_ptr<rtabmap_ros::UserData_<std::allocator<void> > const> const&, std::vector<boost::shared_ptr<cv_bridge::CvImage const>, std::allocator<boost::shared_ptr<cv_bridge::CvImage const> > > const&, std::vector<boost::shared_ptr<cv_bridge::CvImage const>, std::allocator<boost::shared_ptr<cv_bridge::CvImage const> > > const&, std::vector<sensor_msgs::CameraInfo_<std::allocator<void> >, std::allocator<sensor_msgs::CameraInfo_<std::allocator<void> > > > const&, boost::shared_ptr<sensor_msgs::LaserScan_<std::allocator<void> > const> const&, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&, boost::shared_ptr<rtabmap_ros::OdomInfo_<std::allocator<void> > const> const&) () from /home/ubuntu/catkin_ws/devel/lib//librtabmap_ros.so
#10 0x0000007f8601c9d8 in rtabmap_ros::CommonDataSubscriber::commonSingleDepthCallback(boost::shared_ptr<nav_msgs::Odometry_<std::allocator<void> > const> const&, boost::shared_ptr<rtabmap_ros::UserData_<std::allocator<void> > const> const&, boost::shared_ptr<cv_bridge::CvImage const> const&, boost::shared_ptr<cv_bridge::CvImage const> const&, sensor_msgs::CameraInfo_<std::allocator<void> > const&, boost::shared_ptr<sensor_msgs::LaserScan_<std::allocator<void> > const> const&, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&, boost::shared_ptr<rtabmap_ros::OdomInfo_<std::allocator<void> > const> const&) () from /home/ubuntu/catkin_ws/devel/lib//librtabmap_ros.so
#11 0x0000007f86157c78 in rtabmap_ros::CommonDataSubscriber::depthOdomCallback(boost::shared_ptr<nav_msgs::Odometry_<std::allocator<void> > const> const&, boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&, boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&, boost::shared_ptr<sensor_msgs::CameraInfo_<std::allocator<void> > const> const&) () from /home/ubuntu/catkin_ws/devel/lib//librta [... cut ...]
#46 0x0000007fb7d6cedc in nodelet::detail::CallbackQueue::callOne() () from /opt/ros/kinetic/lib/libnodeletlib.so
#47 0x0000007fb7d6eb5c in nodelet::detail::CallbackQueueManager::workerThread(nodelet::detail::CallbackQueueManager::ThreadInfo*) () from /opt/ros/kinetic/lib/libnodeletlib.so
#48 0x0000007fb73b4cf8 in ?? () from /usr/lib/aarch64-linux-gnu/libboost_thread.so.1.58.0
#49 0x0000007fb737dfc4 in start_thread (arg=0x7fb73b4c30) at pthread_create.c:335
#50 0x0000007fb7531290 in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S

matlabbe, 2017-07-13:
thx, I [added](https://github.com/introlab/rtabmap/commit/85f7c1e73c54a9aec4ac4b10c29fcab80e6821f7) more checks to make sure the cloud passed to rtabmap::util3d::voxelize() is not empty. Not sure why the cloud used to create the occupancy grid is empty though. If mapping can work now, you can send me if you want the resulting database (`~/.ros/rtabmap.db`) to see what could create empty clouds (maybe all null depth values).

cheers ,
Mathieu

szx0112, 2017-07-13:
Hi, unfortunately, after the update I still fail to get the loop closure detection. The rtabmap node crash after a couple of seconds:

[image]

BTW, I find that tf_static does not contain any message when type: rostopic hz /tf_static in bash. Will this be a problem based on this thread: http://answers.ros.org/question/203452/no-data-from-rtabmap_ros-rtabmap/

I can send you the rtabmap.db file but I don't know how. Can you send me the links or other way to upload it?

Thanks,

matlabbe, 2017-07-20:
A  link to dropbox or google drive.  If you bactrace the error, is it the same place?

zxp771, 2018-05-24:
Hi @matlabbe 

could you please help me to figure out these question?
I use the intel RTF drone which has the intel realsense R200 camera.
I follow the the tutorial, for r200.run the rtabmap
$ roslaunch realsense_camera r200_nodelet_rgbd.launch
$ roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start" depth_topic:=/camera/depth_registered/sw_registered/image_rect_raw

but the terminal always notes the [ WARN] [1498869707.380826367]: /rtabmap/rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set.

there just once the rtabmap work but I got nothing form the odometry.

I also tried the second method you suggest but it always notes the same warning. 

Thanks

matlabbe, 2018-05-25:
Are the topics listed in the warning really published? Are they all published at the same rate?

closed 2020-06-07 by szx0112