introlab/rtabmap_ros #183: ZR300 - crash in commonCallBack(): rgb and depth images with different resolutions

capstayn, 2017-07-12:
On ubuntu 16.04,
ros kinetic
rtabmap 0.13
rtabmap_ros 0.13

When I run rgbd_odometry on a rosbag with topics taken from zr300, it crashes straight away with the following message: 

> [FATAL] (2017-07-12 15:48:16.737) rgbd_odometry.cpp:222::commonCallback() Condition (imageWidth % depthWidth == 0 && imageHeight % depthHeight == 0 && imageWidth/depthWidth == imageHeight/depthHeight) not met! [rgb=640x480 depth=480x360]

In another computer, ubuntu 14.04, ros indigo, with the binaries of rtabmap (0.11) taken from the repositories, it works flawlessly, though.

Should I assume that I should enforce the same resolutions in the rgb and depth images? I'm a bit baffled by the fact that it works in previous versions of rtabmap.

matlabbe, 2017-07-12:
Are you replaying the same rosbag in both versions? Not sure why 0.11 worked, but it should have asserted too as rgb size is not equal to depth size. RGB and depth should have the same resolution (0.11) or RGB dimension should be a multiple of the depth dimension (>0.11).

Following this [tutorial](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping), you should not use `/camera/depth/image_raw` topic (which is 480x360 **not registered** depth) but the depth image generated from one of the two approaches (`/camera/depth_registered/sw_registered/image_rect_raw` or `/camera/depth/points/image` are 640x480). If you recorded in the bag the point clouds (`/camera/depth/points`), you can use the second approach that will regenerate a registered depth image. If you didn't record the point clouds of ZR300, you will have to [project the depth image to a point cloud](http://wiki.ros.org/depth_image_proc#depth_image_proc.2BAC8-point_cloud_xyz), then feed the point cloud to `pointcloud_to_depthimage` node of the tutorial (second approach) to reconstruct the registered depth image.

cheers,
Mathieu

capstayn, 2017-07-12:
Thanks for your response Mathieu, and your bits of advice. I'll double check and will get back to you. 
Pedro