introlab/rtabmap_ros #184: process has died 【pid 15601, exit code -11】

hildaaw, 2017-07-14:
I have tried both step A and B from #28 , and i am still experiencing "process has died" during startup..

This is the error:
### [rtabmap/rgbd_odometry-1] process has died [pid 15601, exit code -11, cmd /home/ia/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry rgb/image:=/camera/rgb/image_rect_color depth/image:=/camera/depth_registered/image_raw rgb/camera_info:=/camera/rgb/camera_info __name:=rgbd_odometry __log:=/home/ia/.ros/log/29cd68a4-6869-11e7-8ddc-d43d7e05ed8d/rtabmap-rgbd_odometry-1.log].
log file: /home/ia/.ros/log/29cd68a4-6869-11e7-8ddc-d43d7e05ed8d/rtabmap-rgbd_odometry-1*.log

[rtabmap/rtabmap-2] process has died [pid 15612, exit code -11, cmd /home/ia/catkin_ws/devel/lib/rtabmap_ros/rtabmap rgb/image:=/camera/rgb/image_rect_color depth/image:=/camera/depth_registered/image_raw rgb/camera_info:=/camera/rgb/camera_info left/image_rect:=/stereo_camera/left/image_rect_color right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info right/camera_info:=/stereo_camera/right/camera_info scan:=/scan scan_cloud:=/scan_cloud user_data:=/user_data user_data_async:=/user_data_async __name:=rtabmap __log:=/home/ia/.ros/log/29cd68a4-6869-11e7-8ddc-d43d7e05ed8d/rtabmap-rtabmap-2.log].
log file: /home/ia/.ros/log/29cd68a4-6869-11e7-8ddc-d43d7e05ed8d/rtabmap-rtabmap-2*.log

[rtabmap/rtabmapviz-3] process has died [pid 15615, exit code -11, cmd /home/ia/catkin_ws/devel/lib/rtabmap_ros/rtabmapviz -d ~/.ros/rtabmap_gui.ini rgb/image:=/camera/rgb/image_rect_color depth/image:=/camera/depth_registered/image_raw rgb/camera_info:=/camera/rgb/camera_info left/image_rect:=/stereo_camera/left/image_rect_color right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info right/camera_info:=/stereo_camera/right/camera_info scan:=/scan scan_cloud:=/scan_cloud __name:=rtabmapviz __log:=/home/ia/.ros/log/29cd68a4-6869-11e7-8ddc-d43d7e05ed8d/rtabmap-rtabmapviz-3.log].
log file: /home/ia/.ros/log/29cd68a4-6869-11e7-8ddc-d43d7e05ed8d/rtabmap-rtabmapviz-3*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...

and this is the back trace command:
[image]

I am currently using Ubuntu14.04.3. I have tried to open rtabmap standalone, and it works without any errors. 
Any hint is deeply appreciated.

matlabbe, 2017-07-15:
Hi,

See https://github.com/introlab/rtabmap_ros/issues/121#issuecomment-253537021 or this [post](http://official-rtab-map-forum.67519.x6.nabble.com/rtabmapviz-crashes-td995.html) about `boost::math::lanczos_initializer()` crash in libpcl. You can try rebuilding the rtabmap library without c++11 support, so without g2o or GTSAM:
```bash
$ cd rtabmap/build
$ cmake -DWITH_G2O=OFF -DWITH_GTSAM=OFF ..
$ make
$ make install
```

cheers,
Mathieu

hildaaw, 2017-07-18:
Hi,

I tried the codes above but it didn't work, i got the same errors.
However i seen the comments from #121 , thus i delete & reinstall rtabmap and rtabmap_ros, and rtabmap_ros finally works!
Thank you for your time and effort! Appreciate it :)

closed 2017-07-18 by hildaaw