introlab/rtabmap_ros #185: ros wrapper occasionally generates goal pose for move_base that is rejected

fishcu, 2017-07-14:
Hi,

I've run into the issue that RTABMAP will occasionally produce a goal pose for move_base that is rejected with the error:
`Quaternion is invalid... for navigation the z-axis of the quaternion must be close to vertical.`

I have modified move_base to check what the quaternion was. The output is:
` q(w, x, y, z) = 0.647693, 0.413554 0.344697 0.539121`

This is clearly an orientation that is not vertical. I have followed the source code of move_base to check what it does. 
`tf_q.normalize();

    tf::Vector3 up(0, 0, 1);

    double dot = up.dot(up.rotate(tf_q.getAxis(), tf_q.getAngle()));

    if(fabs(dot - 1) > 1e-3){
      ROS_ERROR("Quaternion is invalid... for navigation the z-axis of the quaternion must be close to vertical.");
...`

According to move_base, the resulting axis is (0.5428, 0.4524, 0.7076), and the angle is 0.866. This seems correct. Obviously, the dot product with the up vector will not be near 1.

RTABMAP receives the goal from my application (via "/rtabmap/goal"), which I have confirmed does not correspond to the goal that RTABMAP publishes to move_base.

I have seen that by sending the same goal pose to /rtabmap/goal, RTABMAP will produce an identical (wrong) quaternion as orientation. This would hint that it is not a memory trashing error, but rather some transformation is wrong somewhere. However, I cannot reproduce the error consistently. It seems to happen more once the map ( / the SLAM graph) is bigger.

Any ideas how to fix this? In the meantime, I may just publish to move_base's "goal" topic directly, however I would like to profit from RTABMAP's ability to republish the goal after a loop closure.

Here's the full output of RTABMAP and move_base:
`[ INFO] [1500043662.008014316, 3463.940000000]: Planning: set goal xyz=11.096650,6.433499,0.000000 rpy=0.000000,-0.000000,0.000000
[ INFO] [1500043662.011605386, 3463.940000000]: Planning: Time computing path = 0.003591 s
[ INFO] [1500043662.011904715, 3463.940000000]: Planning: Path successfully created (size=235)
[ INFO] [1500043662.012576802, 3463.950000000]: Publishing next goal: 1633 -> xyz=3.151535,7.817153,0.299990 rpy=1.137004,0.000605,1.388726
[ INFO] [1500043662.013381761, 3463.950000000]: Global path: [1733 1730 1727 1726 1725 1724 1722 1721 1720 1719 1718 1717 1716 1715 1714 1713 1712 1711 1710 1709 1708 1707 1705 1704 1703 1702 1701 1700 1699 1698 1697 1696 1695 1694 1693 1692 1691 1690 1689 1688 1687 1686 1682 1681 1680 1679 1678 1677 1675 1673 1671 1669 1668 1667 1666 1665 1664 1663 1662 1661 1660 1659 1658 1657 1656 1655 1654 1653 1652 1651 1650 1649 1648 1647 1646 1645 1644 1643 1642 1641 1640 1639 1638 1637 1636 1635 1634 1633 1632 1631 1630 1629 1628 1627 1622 1620 1617 1615 1613 1611 1610 1609 1608 1607 1606 1605 1604 1603 1602 1601 1600 1599 1598 1597 1596 1592 1591 1590 1589 1588 1585 1583 1581 1580 1579 1578 1577 1576 1575 1574 1573 1572 1571 1570 1567 1566 1565 1564 1561 1559 1557 1555 1554 1553 1552 1551 1550 1549 1548 1547 1546 1545 1544 1543 1542 1538 1537 1536 1535 1534 1532 1530 1528 1526 1525 1524 1522 1520 1519 1518 1517 1516 1515 1514 1513 1512 1511 1510 1509 1506 1505 1504 1503 1502 1501 1498 1496 1494 1492 1491 1490 1489 1488 1487 1486 1485 1484 1483 1482 1478 1477 1476 1475 1474 1471 1469 1467 1465 1464 1463 1462 1461 1460 1459 1458 1457 1456 1455 1454 1453 1452 1451 1450 1449 1445 1444 1443 1442 1441 1438 1435 1433 1431 1430 1429]
[ERROR] [1500043662.181586855, 3464.110000000]: Quaternion is invalid... for navigation the z-axis of the quaternion must be close to vertical.
[ERROR] [1500043662.181644217, 3464.110000000]: The quaternion was: 0.647693, 0.413554 0.344697 0.539121
[ERROR] [1500043662.183323127, 3464.090000000]: Planning: move_base failed for some reason. Aborting the plan...
`

matlabbe, 2017-07-20:
It is because you may have a 3d map, then the nodes in the graph has 6dof. Set Reg/Force3DoF to true to create a 2d map. If you want to keep a 6dof map/graph, you may create a node subscribing to goal from rtabmap and drop pitch,roll rotation from the quaternion, then republish the goal to move base. This way actionlib cannot be used between rtabmap and move_base... To still use actionlib, you would have to modify CoreWrapper where the goal is published.

Cheers,
Mathieu

fishcu, 2017-07-29:
Thank you, that makes sense.

closed 2017-07-29 by fishcu