introlab/rtabmap_ros #187: Xtion PRO live : demo_two_kinects.launch

hildaaw, 2017-07-19:
Hi,
recently i have tried to run **demo_two_kinects.launch** . 
Since i want this file to work on 2 Xtion Pro live cameras instead of  2 Kinects cameras, I tried manipulate the file from:
` 
  <include file="$(find freenect_launch)/launch/freenect.launch">
    <arg name="depth_registration" value="True" />
    <arg name="camera" value="camera1" />
    <arg name="device_id" value="#1" />
  </include>
  <include file="$(find freenect_launch)/launch/freenect.launch">
    <arg name="depth_registration" value="True" />
    <arg name="camera" value="camera2" />
    <arg name="device_id" value="#2" />
  </include>`

To:
`  
  <include file="$(find openni2_launch)/launch/openni2.launch">
    <arg name="depth_registration" value="True" />
    <arg name="camera" value="camera1" />
    <arg name="device_id" value="#1" />
  </include>
  <include file="$(find openni2_launch)/launch/openni2.launch">
    <arg name="depth_registration" value="True" />
    <arg name="camera" value="camera2" />
    <arg name="device_id" value="#2" />
  </include>
`
After that i tried to run this launch file and i received these warnings:
_[ WARN] [1500457424.840177650]: Camera calibration file /home/ia/.ros/camera_info/rgb_PS1080_PrimeSense.yaml not found.

[ WARN] [1500457428.380126422]: /rtabmap/rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. 
/rtabmap/rgbd_odometry subscribed to (approx sync):
   /camera1/rgbd_image,
   /camera2/rgbd_image_

Even though i received these warnings, my two cameras are connected successfully. 
May I know what do i need to add or remove so that it can calibrate?

Thank you in advance!

matlabbe, 2017-07-26:
Can you verify if the topics mentioned are published?
```bash
$ rostopic hz /camera1/rgbd_image
$ rostopic hz /camera2/rgbd_image
```

hildaaw, 2017-07-27:
I have tried to varify the topics mentioned and i received this error:
**ERROR: Unable to communicate with master!**
Assuming that these items are not in my computer, i updated rtabmap and rtabmap_ros, and still receive the same error.

Severn-Vergil, 2017-07-27:
I guess I have a similar problem. #190 
The topics /camera1/rgbd_image and /camera2/rgbd_image are not published.

matlabbe, 2017-07-28:
So rgbd_image topics are advertised but not published. We may need to look at if the rgbd_sync inputs are published:
```bash
$ rostopic hz /camera1/rgb/image_rect_color
$ rostopic hz /camera1/depth_registered/image_raw
$ rostopic hz /camera1/rgb/camera_info

$ rostopic hz /camera2/rgb/image_rect_color
$ rostopic hz /camera2/depth_registered/image_raw
$ rostopic hz /camera2/rgb/camera_info
```
I [updated](https://github.com/introlab/rtabmap_ros/commit/5088768a2ecf4577dd0b7a20fc865c5f86bee1cc) rgbd_sync code to output warnings if inputs are not received for convenience.

cheers,
Mathieu

hildaaw, 2017-07-31:
I have updated rtabmap_ros and look into the rgbd_sync input and these are the results: 

```
WARNING: topic [/camera1/rgb/image_rect_color] does not appear to be published yet
WARNING: topic [/camera1/depth_registered/image_raw] does not appear to be published yet
WARNING: topic [/camera1/rgb/camera_info] does not appear to be published yet

WARNING: topic [/camera2/rgb/image_rect_color] does not appear to be published yet
WARNING: topic [/camera2/depth_registered/image_raw] does not appear to be published yet
WARNING: topic [/camera2/rgb/camera_info] does not appear to be published yet

```

matlabbe, 2017-07-31:
That means the camera drivers are not working. You can also use `rqt_image_view` to actually see the raw images that are supposed to be published.

hildaaw, 2017-08-01:
My apologies. It seems that i made a mistake, i forgotten to run the launch file, that's why the camera drivers isn't running. After launching `demo_two_kinects.launch` i got these:

> subscribed to [/camera1/rgb/image_rect_color]
> average rate: 4.782
> 	min: 0.033s max: 0.737s std dev: 0.30459s window: 5
> subscribed to [/camera1/depth_registered/image_raw]
> subscribed to [/camera1/rgb/camera_info]
> average rate: 16.684
> 	min: 0.033s max: 0.168s std dev: 0.04429s window: 15
> 
> subscribed to [/camera2/rgb/image_rect_color]
> average rate: 29.850
> 	min: 0.033s max: 0.034s std dev: 0.00053s window: 30
> subscribed to [/camera2/depth_registered/image_raw]
> subscribed to [/camera2/rgb/camera_info]
> average rate: 29.851
> 	min: 0.033s max: 0.034s std dev: 0.00046s window: 30
>

hildaaw, 2017-08-16:
Update: I have tried the software on XBOX 360 and i received the same error.

`... logging to /home/ia/.ros/log/5408978e-8230-11e7-b28b-d43d7e05ed8d/roslaunch-ia-MS-7759-23600.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ia-MS-7759:43401/

SUMMARY
========

PARAMETERS
 * /camera1/camera1_nodelet_manager/num_worker_threads: 4
 * /camera1/depth_rectify_depth/interpolation: 0
 * /camera1/depth_registered_rectify_depth/interpolation: 0
 * /camera1/disparity_depth/max_range: 4.0
 * /camera1/disparity_depth/min_range: 0.5
 * /camera1/disparity_registered_hw/max_range: 4.0
 * /camera1/disparity_registered_hw/min_range: 0.5
 * /camera1/disparity_registered_sw/max_range: 4.0
 * /camera1/disparity_registered_sw/min_range: 0.5
 * /camera1/driver/data_skip: 0
 * /camera1/driver/debug: False
 * /camera1/driver/depth_camera_info_url: 
 * /camera1/driver/depth_frame_id: camera1_depth_opt...
 * /camera1/driver/depth_registration: True
 * /camera1/driver/device_id: #1
 * /camera1/driver/diagnostics_max_frequency: 30.0
 * /camera1/driver/diagnostics_min_frequency: 30.0
 * /camera1/driver/diagnostics_tolerance: 0.05
 * /camera1/driver/diagnostics_window_time: 5.0
 * /camera1/driver/enable_depth_diagnostics: False
 * /camera1/driver/enable_ir_diagnostics: False
 * /camera1/driver/enable_rgb_diagnostics: False
 * /camera1/driver/rgb_camera_info_url: 
 * /camera1/driver/rgb_frame_id: camera1_rgb_optic...
 * /camera2/camera2_nodelet_manager/num_worker_threads: 4
 * /camera2/depth_rectify_depth/interpolation: 0
 * /camera2/depth_registered_rectify_depth/interpolation: 0
 * /camera2/disparity_depth/max_range: 4.0
 * /camera2/disparity_depth/min_range: 0.5
 * /camera2/disparity_registered_hw/max_range: 4.0
 * /camera2/disparity_registered_hw/min_range: 0.5
 * /camera2/disparity_registered_sw/max_range: 4.0
 * /camera2/disparity_registered_sw/min_range: 0.5
 * /camera2/driver/data_skip: 0
 * /camera2/driver/debug: False
 * /camera2/driver/depth_camera_info_url: 
 * /camera2/driver/depth_frame_id: camera2_depth_opt...
 * /camera2/driver/depth_registration: True
 * /camera2/driver/device_id: #2
 * /camera2/driver/diagnostics_max_frequency: 30.0
 * /camera2/driver/diagnostics_min_frequency: 30.0
 * /camera2/driver/diagnostics_tolerance: 0.05
 * /camera2/driver/diagnostics_window_time: 5.0
 * /camera2/driver/enable_depth_diagnostics: False
 * /camera2/driver/enable_ir_diagnostics: False
 * /camera2/driver/enable_rgb_diagnostics: False
 * /camera2/driver/rgb_camera_info_url: 
 * /camera2/driver/rgb_frame_id: camera2_rgb_optic...
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /rtabmap/rgbd_odometry/Odom/FillInfoData: true
 * /rtabmap/rgbd_odometry/Odom/Strategy: 0
 * /rtabmap/rgbd_odometry/OdomF2M/MaxSize: 1000
 * /rtabmap/rgbd_odometry/Vis/CorNNType: 3
 * /rtabmap/rgbd_odometry/Vis/FeatureType: 6
 * /rtabmap/rgbd_odometry/Vis/InlierDistance: 0.02
 * /rtabmap/rgbd_odometry/Vis/MaxDepth: 4.0
[... 301 lines cut ...]
[camera1/disparity_registered_sw-20] killing on exit
[camera1/disparity_depth-19] killing on exit
[camera1/depth_registered_metric-18] killing on exit
[camera1/depth_registered_hw_metric_rect-17] killing on exit
[camera1/points_xyzrgb_hw_registered-16] killing on exit
[camera1/depth_registered_rectify_depth-15] killing on exit
[camera1/depth_registered_sw_metric_rect-14] killing on exit
[camera1/points_xyzrgb_sw_registered-13] killing on exit
[camera1/register_depth_rgb-12] killing on exit
[camera1/depth_points-11] killing on exit
[camera1/depth_metric-10] killing on exit
[camera1/depth_metric_rect-9] killing on exit
[camera1/depth_rectify_depth-8] killing on exit
[camera1/ir_rectify_ir-7] killing on exit
[camera1/rgb_rectify_color-6] killing on exit
[camera1/rgb_rectify_mono-5] killing on exit
[camera1/rgb_debayer-4] killing on exit
[camera1/driver-3] killing on exit
[camera1/camera1_nodelet_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

matlabbe, 2017-08-16:
The default of parameter `Vis/EstimationType` has been changed to 1 (2D->3D PnP estimation) in recent versions. `Vis/EstimationType=1` cannot be used with multiple cameras. Set explicitly `Vis/EstimationType=0` to `rgbd_odometry` and `rtabmap` nodes:

```xml
<param name="Vis/EstimationType" type="string" value="0"/>
```
I updated [demo_two_kinects.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/demo/demo_two_kinects.launch) with this change.

cheers,
Mathieu

mariogazziro, 2018-07-30:
Hi Mathieu

Can you provide a launch script for 2 orbbec-astra sensors ? i am already try to mix it but without any success.

thanks a lot!

matlabbe, 2018-08-02:
Well, did you try [demo_two_kinects.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/demo/demo_two_kinects.launch)? by changing `freenect_launch` to `astra_launch` instead:
https://github.com/introlab/rtabmap_ros/blob/bc4b7a860e41948e77554023c3efd26ced949b50/launch/demo/demo_two_kinects.launch#L7-L16

this instead:
```xml
  <include file="$(find astra_launch)/launch/astra.launch">
    <arg name="depth_registration" value="True" />
    <arg name="camera" value="camera1" />
    <arg name="device_id" value="#1" />
  </include>
  <include file="$(find astra_launch)/launch/astra.launch">
    <arg name="depth_registration" value="True" />
    <arg name="camera" value="camera2" />
    <arg name="device_id" value="#2" />
  </include>
```

If so, can you show the resulting tf tree? and warnings/errors of the terminal?
```bash
$ rosrun tf view_frames
$ evince frames.pdf
```

matlabbe, 2018-08-04:
Indeed, the default parameters are not set to scan objects moving independently from the scene. You could set the maximum depth of extracted features under "Maximum feature depth." of Visual Registration panel ( or "Vis/MaxDepth" parameter) and also the "Maximum words depth" of Vocabulary panel (or "Kp/MaxDepth" parameter).

When exporting the clouds, you can also set a maximum depth for the generated points to filter the background.

mariogazziro, 2018-08-04:
I did exactly what you said in previous sample (i already minded about isolating object from the scene) but it doenst work. the scrrenshot is the best result before the cloud become a completly mess. i think there are not enough inliers. there are someway to overcome this ? even if we lose in details, my interest is to close the mesh.

matlabbe, 2018-08-07:
You may use ICP odometry instead. Here is an example using the RTAB-Map standalone version and a Kinect:
https://youtu.be/es0x0bAdEIs

Parameters used: [person_scan.ini.txt](https://github.com/introlab/rtabmap_ros/files/2267838/person_scan.ini.txt)
Main parameters changed are in Odom, Reg and Icp groups. The input scan from the kinect was downsampled by 4 and limited up to 1.5 meter. RTAB-Map should be built with libpointmatcher to give similar results.

I added a small box on behind my neck to add geometry because tracking got lost or had very large drift when seeing only the back of the chair. However, there is still a small drift when the camera is only seeing parallel planes. I was not perfectly immobile while turning the chair either, which can add some errors to the scan. 

cheers,
Mathieu

mariogazziro, 2018-08-07:
wow.. very impressive.. i will try your parameters in our equipment ( https://www.youtube.com/watch?v=0y0jxQCE9Bs ). thank you very much Mathieu.  and if you have any ideas to solve the problem without using the small box would be nice, cause we need to evaluate postural and perimeter analysis, and the box could imply problems in perimeter measure (but if we need to put some object between the perimeter lines isn't a problem at all, let me try here).

matlabbe, 2018-08-08:
With a person standing, I don't think adding more geometry is required as the body has a lot of distinctive curves. The problem here was really the flat back of the chair, causing more drift or making loosing odometry based on the geometry. 

Looking at your equipment, you may don't need SLAM at all if you can know the current rotation of the body (current angle of the platform) accordingly to the cameras. You will then avoid the drift problem when there is not enough geometry (i.e., you can have perfect odometry independently of what the object geometry looks like).

mariogazziro, 2018-08-08:
yes, i can get the exactly info about rotation (with arduino or even with a raspberrypi), but how can send this information to RTAB-MAP ? do i need to write a custom kernel driver  ? or i can simple write this info in a shared file ?

matlabbe, 2018-08-08:
You can create a [TF broadcaster](http://wiki.ros.org/tf/Tutorials) to publish a fake odometry transform (`odom -> base_link`), replacing `rgbd_odometry` node. That transform will change accordingly as the platform is rotating. Set `rtabmap` node's parameter `odom_frame_id` to that frame (`odom`), and set `Kp/MaxFeatures` to `-1` to disable loop closure detection (we don't need it as we assume there is no drift).

Assuming that the camera links are children of a fixed frame called `base_link`, in rviz the frame `base_link` will look like turning around `odom` frame (which would be at the center of the rotating platform), similar to video I posted.

mariogazziro, 2018-08-09:
wow.. sure i will do this! thanks Mathieu.. I will post the results here soon.

mariogazziro, 2018-08-29:
Hi Mathieu

I finally install an absolute encoder (magnetic) on my rotary table and adjust the turtle sim (script below) to test it as you can see in my videos. 

but now i am not sure how to mix up this together.. in other words, how to exactly replace the odom info in my 2 sensor tree...

#!/usr/bin/env python  
import math
import serial
import roslib
roslib.load_manifest('learning_tf')
import rospy

import tf
import turtlesim.msg

port = serial.Serial("/dev/ttyUSB0", baudrate=9600, timeout=3.0)

def handle_turtle_pose(msg, turtlename):
    br = tf.TransformBroadcaster()
    angle = 115
    angle = int(port.readline())
    br.sendTransform((msg.x, msg.y, 0),
                     #tf.transformations.quaternion_from_euler(0, 0, msg.theta),
                     tf.transformations.quaternion_from_euler(0, 0, math.radians(angle)),
                     rospy.Time.now(),
                     turtlename,
                     "world")

if __name__ == '__main__':
    rospy.init_node('turtle_tf_broadcaster')
    turtlename = rospy.get_param('~turtle')
    rospy.Subscriber('/%s/pose' % turtlename,
                     turtlesim.msg.Pose,
                     handle_turtle_pose,
                     turtlename)
    rospy.spin()

[encoder_turtle_frames.pdf](https://github.com/introlab/rtabmap_ros/files/2333424/encoder_turtle_frames.pdf)
[frames_2_astras.pdf](https://github.com/introlab/rtabmap_ros/files/2333425/frames_2_astras.pdf)

https://youtu.be/2o18tDgwUHw

https://youtu.be/hG3WQNmQXXw

matlabbe, 2018-08-30:
What you want is to provide a TF representing the camera moving around the rotary table, not the table accordingly to its center. Here the `/odom` frame is fixed at the center of the table and the camera (`/camera_link`) is rotating at a fixed distance _d_ from the center of the table accordingly to angle of your absolute encoder:
[image]

In reality the table is rotating, but in this setup we change the referential so that the camera appears to  rotate around the table.