introlab/rtabmap_ros #189: Did not receive data since 5 seconds!

mertmzk, 2017-07-24:
Hi all,
I have no success to run rtabmap with my two webcams stereo setup camera. 

here my rqt_graph

[image]

and the warning mesage with approx_sync:=false

> [ WARN] [1500890325.598924032]: /rtabmap/rtabmapviz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rtabmapviz subscribed to (exact sync):
   /rtabmap/odom,
   /stereo_camera/left/image_rect_color,
   /stereo_camera/right/image_rect,
   /stereo_camera/left/camera_info,
   /stereo_camera/right/camera_info,
   /rtabmap/odom_info

if I make it approx_sync:=true here the error:

> [ WARN] [1500890522.000583488]: odometry: Could not get transform from base_link to yuyv (stamp=1500890521.696677) after 0.200000 seconds ("wait_for_transform_duration"=0.200000)! Error=". canTransform returned after 0.201409 timeout was 0.2."

Can anyone help me to solve the problem please?

Thanks

matlabbe, 2017-07-27:
What is your launch file or command line used? "base_link" is the default `frame_id`, you may want to change it to "yuyv" for example. Well, I don't recommend using to seperate webcams to do stereo, as synchronization is crucial between left and right camera. In general, stereo cameras are synchronized by the hardware, so that exact synchronization can be used.

cheers,
Mathieu

TothMateGit, 2017-09-30:
I have also the same problem. I started to create a stereo system with two cameras, but I've also got the same warning messages.
I created an approx synchronizer node to avoid the problem mentioned by matlabbe, but that solution did not help me out, and I opened another ROS topic here:
[https://answers.ros.org/question/271979/3d-reconstruction-with-stereo-cameras/](url)

I'm afraid about we can not fix the problem, and I should to use a real stereo camera and a frame grabber, shame.

mertmzk, 2017-10-25:
Hi @TothMateGit 

I fixed the problem as I was not producing proper tf's for the base_link. Make sure that you publish tf's for right and left camera as child stereo camera as parent. And connect stereo_camera tf to base_link.

I hope it helps.

TothMateGit, 2017-10-25:
Yes, the problem is the bad tf generation and tf tree, tf connection between the nodes and topics. We found the solution to fix this issue as well, if you want more information about the solution please check the following links:
[ROS answers](https://answers.ros.org/question/271979/3d-reconstruction-with-stereo-cameras/)
[Stack Overflow](https://stackoverflow.com/questions/46530509/ros-rtabmap-does-not-create-point-cloud)

ken012git, 2019-03-20:
I have the same issue here. 

I am using realsense RS200 as my RGBD camera. I was trying to map the color and depth from RS200 to RTAB-MAP, but I got this warning when I run my rtabmap launch file

```
[ WARN] [1553082244.099429564, 7862.743000000]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmap subscribed to (approx sync):
   /rtabmap/odom,
   /realsense/camera/color/image_raw,
   /realsense/camera/depth/image_raw,
   /realsense/camera/color/camera_info
```

Here is the result I run `ros topic hz` and both color and depth topics are published, but I still get the warning and see nothing in my Rivz.

```
$ rostopic hz /realsense/camera/color/image_raw /realsense/camera/depth/image_raw                                                                                                    
subscribed to [/realsense/camera/color/image_raw]
subscribed to [/realsense/camera/depth/image_raw]
WARNING: may be using simulated time
              topic                  rate   min_delta   max_delta   std_dev    window
=====================================================================================
/realsense/camera/color/image_raw   22.59   0.034       0.052       0.004351   20    
/realsense/camera/depth/image_raw   22.43   0.037       0.057       0.005641   20    

              topic                  rate   min_delta   max_delta   std_dev    window
=====================================================================================
/realsense/camera/color/image_raw   22.34   0.034       0.054       0.004011   42    
/realsense/camera/depth/image_raw   22.37   0.036       0.057       0.004697   42    

              topic                  rate   min_delta   max_delta   std_dev    window
=====================================================================================
/realsense/camera/color/image_raw   22.1    0.034       0.054       0.003899   62    
/realsense/camera/depth/image_raw   22.13   0.036       0.057       0.004471   62    

              topic                  rate   min_delta   max_delta   std_dev    window
=====================================================================================
/realsense/camera/color/image_raw   22.28   0.026       0.054       0.004334   84    
/realsense/camera/depth/image_raw   22.25   0.031       0.058       0.005172   84    

              topic                  rate   min_delta   max_delta   std_dev    window
=====================================================================================
/realsense/camera/color/image_raw   22.2    0.026       0.057       0.004835   105   
/realsense/camera/depth/image_raw   22.18   0.031       0.062       0.005577   105   

              topic                  rate   min_delta   max_delta   std_dev    window
=====================================================================================
/realsense/camera/color/image_raw   22.23   0.026       0.058       0.005195   126   
/realsense/camera/depth/image_raw   22.24   0.031       0.07        0.005845   126   

              topic                  rate   min_delta   max_delta   std_dev    window
=====================================================================================
/realsense/camera/color/image_raw   22.26   0.026       0.058       0.005144   132   
/realsense/camera/depth/image_raw   22.27   0.031       0.07        0.005781   132   

```

I provide my rtabmap launch file here. I changed the value of frame_id to `origin` here, since the point cloud shown in Rviz with fixed frame `origin`.
```
<launch>
  <!-- Mapping -->
  <group ns="rtabmap">

    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen">
          <param name="database_path"       type="string" value="rtabmap.db"/>
          <param name="frame_id"            type="string" value="origin"/>
          <param name="subscribe_depth"     type="bool"   value="true"/>
          <param name="subscribe_scan"      type="bool"   value="false"/>
          <param name="map_negative_poses_ignored" type="bool" value="true"/>

[... 16 lines cut ...]
          <param name="Vis/InlierDistance"           type="string" value="0.1"/>    <!-- 3D visual words correspondence distance -->
          <param name="RGBD/AngularUpdate"           type="string" value="0.1"/>    <!-- Update map only if the robot is moving -->
          <param name="RGBD/LinearUpdate"            type="string" value="0.1"/>    <!-- Update map only if the robot is moving -->
          <param name="RGBD/ProximityPathMaxNeighbors" type="string" value="0"/>
          <param name="Rtabmap/TimeThr"              type="string" value="700"/>
          <param name="Mem/RehearsalSimilarity"      type="string" value="0.30"/>
          <param name="Optimizer/Slam2D"             type="string" value="true"/>
          <param name="Reg/Force3DoF"                type="string" value="true"/>
          <param name="GridGlobal/MinSize"           type="string" value="20"/>
          <param name="RGBD/OptimizeMaxError"        type="string" value="0.1"/>

    </node>

  </group>
</launch>

```

Snapshots of my Rivz:
[image]

Rviz shows nothing when I subscribe to `rtabmap/mapData`.
[image]

Would someone point out where did I get wrong? Thanks!

matlabbe, 2019-03-23:
Did you check if the camera_info is also published?
`$ rostopic hz /realsense/camera/color/image_raw /realsense/camera/depth/image_raw  /realsense/camera/color/camera_info`

ken012git, 2019-03-23:
Hmm, it seems that camera_info is also published.

```
$ rostopic hz /realsense/camera/color/image_raw /realsense/camer
a/depth/image_raw /realsense/camera/color/camera_info
subscribed to [/realsense/camera/color/image_raw]
subscribed to [/realsense/camera/depth/image_raw]
subscribed to [/realsense/camera/color/camera_info]
WARNING: may be using simulated time
               topic                   rate   min_delta   max_delta   std_dev    window
=======================================================================================
/realsense/camera/color/image_raw     24.44   0.028       0.051       0.00662    24    
/realsense/camera/depth/image_raw     24.52   0.025       0.053       0.006234   24    
/realsense/camera/color/camera_info   24.44   0.028       0.051       0.006281   24    

               topic                   rate   min_delta   max_delta   std_dev    window
=======================================================================================
/realsense/camera/color/image_raw     24.34   0.028       0.051       0.005315   48    
/realsense/camera/depth/image_raw     24.4    0.025       0.053       0.005152   48    
/realsense/camera/color/camera_info   24.39   0.028       0.063       0.006108   47    

               topic                   rate   min_delta   max_delta   std_dev    window
=======================================================================================
/realsense/camera/color/image_raw     24.44   0.026       0.056       0.005403   71    
/realsense/camera/depth/image_raw     24.43   0.022       0.056       0.005632   71    
/realsense/camera/color/camera_info   24.45   0.028       0.063       0.005806   71    

               topic                   rate   min_delta   max_delta   std_dev    window
=======================================================================================
/realsense/camera/color/image_raw     24.42   0.026       0.056       0.00524    95    
/realsense/camera/depth/image_raw     24.45   0.022       0.056       0.005973   95    
/realsense/camera/color/camera_info   24.43   0.026       0.063       0.005572   95    

               topic                   rate   min_delta   max_delta   std_dev    window
=======================================================================================
/realsense/camera/color/image_raw     24.34   0.026       0.056       0.005376   119   
/realsense/camera/depth/image_raw     24.36   0.022       0.056       0.006017   119   
/realsense/camera/color/camera_info   24.34   0.026       0.063       0.005683   118   

               topic                   rate   min_delta   max_delta   std_dev    window
=======================================================================================
/realsense/camera/color/image_raw     24.23   0.026       0.056       0.005376   142   
/realsense/camera/depth/image_raw     24.23   0.022       0.056       0.006029   142   
/realsense/camera/color/camera_info   24.23   0.026       0.063       0.005671   141  
```

matlabbe, 2019-03-23:
What is rgbd_odometry telling? The other topic is /rtabmap/odom, maybe odometry is lost

ken012git, 2019-03-28:
Yes, you are right. There is no `/rtabmap/rgbd_odometry` and `/rtabmap/odom` under my topic list. How would I fix this?

```
$ rostopic hz /rtabmap/rgbd_odometry
WARNING: topic [/rtabmap/rgbd_odometry] does not appear to be published yet
$ rostopic hz /rtabmap/rgbd_odom
WARNING: topic [/rtabmap/rgbd_odom] does not appear to be published yet
```

Here is the result of `rostopic list`
```
$ rostopic list
/clicked_point
/clock
/gazebo/link_states
/gazebo/model_states
/gazebo/parameter_descriptions
/gazebo/parameter_updates
/gazebo/set_link_state
/gazebo/set_model_state
/gazebo_gui/parameter_descriptions
/gazebo_gui/parameter_updates
/initialpose
/map
/move_base_simple/goal
/realsense/camera/color/camera_info
/realsense/camera/color/image_raw
/realsense/camera/color/image_raw/compressed
/realsense/camera/color/image_raw/compressed/parameter_descriptions
/realsense/camera/color/image_raw/compressed/parameter_updates
/realsense/camera/color/image_raw/compressedDepth
/realsense/camera/color/image_raw/compressedDepth/parameter_descriptions
/realsense/camera/color/image_raw/compressedDepth/parameter_updates
/realsense/camera/color/image_raw/theora
/realsense/camera/color/image_raw/theora/parameter_descriptions
/realsense/camera/color/image_raw/theora/parameter_updates
/realsense/camera/depth/camera_info
/realsense/camera/depth/image_raw
/realsense/camera/depth/image_raw/compressed
/realsense/camera/depth/image_raw/compressed/parameter_descriptions
/realsense/camera/depth/image_raw/compressed/parameter_updates
/realsense/camera/depth/image_raw/compressedDepth
/realsense/camera/depth/image_raw/compressedDepth/parameter_descriptions
/realsense/camera/depth/image_raw/compressedDepth/parameter_updates
/realsense/camera/depth/image_raw/theora
/realsense/camera/depth/image_raw/theora/parameter_descriptions
/realsense/camera/depth/image_raw/theora/parameter_updates
/realsense/camera/depth/image_rect
/realsense/camera/depth/image_rect/compressed
/realsense/camera/depth/image_rect/compressed/parameter_descriptions
/realsense/camera/depth/image_rect/compressed/parameter_updates
/realsense/camera/depth/image_rect/compressedDepth
/realsense/camera/depth/image_rect/compressedDepth/parameter_descriptions
/realsense/camera/depth/image_rect/compressedDepth/parameter_updates
/realsense/camera/depth/image_rect/theora
/realsense/camera/depth/image_rect/theora/parameter_descriptions
/realsense/camera/depth/image_rect/theora/parameter_updates
/realsense/camera/depth_registered/points
/realsense/camera/ir/camera_info
/realsense/camera/ir/image_raw
/realsense/camera/ir/image_raw/compressed
/realsense/camera/ir/image_raw/compressed/parameter_descriptions
/realsense/camera/ir/image_raw/compressed/parameter_updates
/realsense/camera/ir/image_raw/compressedDepth
/realsense/camera/ir/image_raw/compressedDepth/parameter_descriptions
/realsense/camera/ir/image_raw/compressedDepth/parameter_updates
/realsense/camera/ir/image_raw/theora
/realsense/camera/ir/image_raw/theora/parameter_descriptions
/realsense/camera/ir/image_raw/theora/parameter_updates
/realsense/camera/ir2/camera_info
/realsense/camera/ir2/image_raw
/realsense/camera/ir2/image_raw/compressed
/realsense/camera/ir2/image_raw/compressed/parameter_descriptions
/realsense/camera/ir2/image_raw/compressed/parameter_updates
/realsense/camera/ir2/image_raw/compressedDepth
/realsense/camera/ir2/image_raw/compressedDepth/parameter_descriptions
/realsense/camera/ir2/image_raw/compressedDepth/parameter_updates
/realsense/camera/ir2/image_raw/theora
/realsense/camera/ir2/image_raw/theora/parameter_descriptions
/realsense/camera/ir2/image_raw/theora/parameter_updates
/rosout
/rosout_agg
/rtabmap/cloud_ground
/rtabmap/cloud_map
/rtabmap/cloud_obstacles
/rtabmap/global_path
/rtabmap/goal
/rtabmap/goal_node
/rtabmap/goal_out
/rtabmap/goal_reached
/rtabmap/grid_prob_map
/rtabmap/info
/rtabmap/initialpose
/rtabmap/labels
/rtabmap/local_path
/rtabmap/localization_pose
/rtabmap/mapData
/rtabmap/mapGraph
/rtabmap/mapPath
/rtabmap/move_base/cancel
/rtabmap/move_base/feedback
/rtabmap/move_base/goal
/rtabmap/move_base/result
/rtabmap/move_base/status
/rtabmap/octomap_binary
/rtabmap/octomap_empty_space
/rtabmap/octomap_full
/rtabmap/octomap_grid
/rtabmap/octomap_ground
/rtabmap/octomap_obstacles
/rtabmap/octomap_occupied_space
/rtabmap/proj_map
/rtabmap/scan_map
/standalone_nodelet/bond
/tf
/tf_static
```

As I subscribe some topic from `rtabmap` like  `rostopic hz /rtabmap/cloud_map`, I got `no new messages`
```
$ rostopic hz /rtabmap/cloud_map
subscribed to [/rtabmap/cloud_map]
WARNING: may be using simulated time
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
```

matlabbe, 2019-03-31:
Can you post the launch file you are using? It seems you don't have any nodes publishing odometry.

ken012git, 2019-04-03:
I am using the launch files from [realsense_gazebo_plugin](https://github.com/SyrianSpock/realsense_gazebo_plugin)

`realsense.launch`
```
<launch>
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="world_name" value="$(find realsense_gazebo_plugin)/worlds/sensor.world"/>
    <arg name="headless" value="true"/>
    <arg name="gui" value="true"/>
    <arg name="verbose" value="true"/>
  </include>
</launch>
```

`depth_proc.launch`
```
<launch>
  <param name="use_sim_time" value="true" />

  <arg name="rgb_camera_info" value="/realsense/camera/color/camera_info"/>
  <arg name="rgb_img_rect" value="/realsense/camera/color/image_raw"/>  <!--Rectified color image-->
  <arg name="depReg_imgraw" value="/realsense/camera/depth/image_raw"/>  <!--Raw depth image-->
  <arg name="depReg_imgrect" value="/realsense/camera/depth/image_rect"/>  <!--Raw depth image-->
  <arg name="out_cloud" value="/realsense/camera/depth_registered/points"/>

  <node pkg="nodelet" type="nodelet" name="standalone_nodelet" args="manager" output="screen"/>

  <!-- Convert depth from mm (in uint16) to meters -->
  <node pkg="nodelet" type="nodelet" name="convert_metric" args="load depth_image_proc/convert_metric standalone_nodelet">
    <remap from="image_raw" to="$(arg depReg_imgraw)"/>
    <remap from="image" to="$(arg depReg_imgrect)"/>
  </node>

  <!-- Construct point cloud of the rgb and depth topics -->
  <node pkg="nodelet" type="nodelet" name="points_xyzrgb" args="load depth_image_proc/point_cloud_xyzrgb standalone_nodelet --no-bond">
    <remap from="rgb/camera_info" to="$(arg rgb_camera_info)" />
    <remap from="rgb/image_rect_color" to="$(arg rgb_img_rect)"/>
    <remap from="depth_registered/image_rect" to="$(arg depReg_imgrect)"/>
    <remap from="depth_registered/points" to="$(arg out_cloud)"/>
  </node>

  <node pkg="tf" type="static_transform_publisher" name="fake_tf" args="0 0 0 0 0 -1.57 origin color 10"/>
</launch>
```

`realsense_rtabmap.launch`. Some modifications:
- `<param name="frame_id"            type="string" value="origin"/>`
- `<param name="subscribe_depth"     type="bool"   value="true"/>`
- `<param name="subscribe_scan"      type="bool"   value="false"/>`
- `<remap from="rgb/image"       to="/realsense/camera/color/image_raw"/>`
- `<remap from="depth/image"     to="/realsense/camera/depth/image_raw"/>`
- `<remap from="rgb/camera_info" to="/realsense/camera/color/camera_info"/>`
- `<param name="Grid/FromDepth"               type="string" value="true"/>`
```
<launch>
  <!-- Mapping -->
  <group ns="rtabmap">

    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen">
          <param name="database_path"       type="string" value="rtabmap.db"/>
          <param name="frame_id"            type="string" value="origin"/>
          <param name="subscribe_depth"     type="bool"   value="true"/>
          <param name="subscribe_scan"      type="bool"   value="false"/>
          <param name="map_negative_poses_ignored" type="bool" value="true"/>

          <!-- inputs -->
          <remap from="scan"            to="/scan"/>
          <remap from="rgb/image"       to="/realsense/camera/color/image_raw"/>
          <remap from="depth/image"     to="/realsense/camera/depth/image_raw"/>
          <remap from="rgb/camera_info" to="/realsense/camera/color/camera_info"/>
[... 216 lines cut ...]
   /rtabmap/odom,
   /realsense/camera/color/image_raw,
   /realsense/camera/depth/image_raw,
   /realsense/camera/color/camera_info
[ WARN] [1554256409.025077108, 1386.543000000]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their head
er are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size"
 parameter (current=30).
/rtabmap/rtabmap subscribed to (approx sync):
   /rtabmap/odom,
   /realsense/camera/color/image_raw,
   /realsense/camera/depth/image_raw,
   /realsense/camera/color/camera_info
```

4. Finally, I ran `$ rviz`. As Rviz is open, I select `origin` at `Fixed Frame` under `Global Options` and add `/realsense/camera/depth_registered/points` and `/realsense/camera/color/image_raw`
```
[ INFO] [1554256400.639810124]: rviz version 1.12.16
[ INFO] [1554256400.639882833]: compiled against Qt version 5.5.1
[ INFO] [1554256400.639902548]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1554256401.952788719, 1379.570000000]: Stereo is NOT SUPPORTED
[ INFO] [1554256401.953552355, 1379.570000000]: OpenGl version: 4.5 (GLSL 4.5).
```

Here is the snapshot of rviz.
[image]

matlabbe, 2019-04-04:
Can you check:
`$ rostopic hz /realsense/camera/color/image_raw /realsense/camera/depth/image_raw /realsense/camera/color/camera_info`

If there is no depth image, you would have to reconstruct one from the point cloud. To do that:
`$ rosrun rtabmap_ros pointcloud_to_depthimage cloud:=/realsense/camera/depth_registered/points camera_info:=/realsense/camera/color/camera_info image_raw:=/realsense/camera/depth/image_raw image:=/realsense/camera/depth/image _approx:=false _fill_holes_size:=2`

cheers,
Mathieu

ken012git, 2019-04-04:
Hmm...
I ran `rosrun rtabmap_ros pointcloud_to_depthimage cloud:=/realsense/camera/depth_registered/points camera_info:=/realsense/camera/color/camera_info image_raw:=/realsense/camera/depth/image_raw image:=/realsense/camera/depth/image _approx:=false _fill_holes_size:=2` before launch `realsense_rtabmap.launch`, but it seems that RTABMAP is still not receiving the camera data.

I also checked the depth image. There is the result when I ran `$ rostopic hz /realsense/camera/color/image_raw /realsense/camera/depth/image_raw /realsense/camera/color/camera_info` 
```
$ rostopic hz /realsense/camera/color/image_raw /realsense/camera/depth/image_raw /realsense/camera/color/camera_info
subscribed to [/realsense/camera/color/image_raw]
subscribed to [/realsense/camera/depth/image_raw]
subscribed to [/realsense/camera/color/camera_info]
WARNING: may be using simulated time
               topic                   rate   min_delta   max_delta   std_dev    window
=======================================================================================
/realsense/camera/color/image_raw     32.13   0.023       0.052       0.005504   33    
/realsense/camera/depth/image_raw     32.46   0.021       0.042       0.00548    33    
/realsense/camera/color/camera_info   33.03   0.024       0.038       0.003713   32    

               topic                   rate   min_delta   max_delta   std_dev    window
=======================================================================================
/realsense/camera/color/image_raw     32.81   0.021       0.052       0.005114   66    
/realsense/camera/depth/image_raw     32.96   0.021       0.042       0.004935   66    
/realsense/camera/color/camera_info   33.24   0.018       0.039       0.004452   65    

               topic                   rate   min_delta   max_delta   std_dev    window
=======================================================================================
/realsense/camera/color/image_raw     33.38   0.016       0.052       0.00493    100   
/realsense/camera/depth/image_raw     33.42   0.021       0.042       0.004577   100   
/realsense/camera/color/camera_info   33.68   0.018       0.039       0.004309   99    

               topic                   rate   min_delta   max_delta   std_dev    window
=======================================================================================
/realsense/camera/color/image_raw     33.56   0.016       0.052       0.004803   133   
/realsense/camera/depth/image_raw     33.6    0.021       0.043       0.004726   133   
/realsense/camera/color/camera_info   33.8    0.018       0.04        0.004289   132   

               topic                   rate   min_delta   max_delta   std_dev    window
=======================================================================================
/realsense/camera/color/image_raw     33.77   0.016       0.052       0.004716   167   
/realsense/camera/depth/image_raw     33.84   0.017       0.043       0.004638   167   
/realsense/camera/color/camera_info   33.92   0.018       0.04        0.004387   165   

               topic                   rate   min_delta   max_delta   std_dev    window
=======================================================================================
/realsense/camera/color/image_raw     33.81   0.016       0.052       0.004727   200   
/realsense/camera/depth/image_raw     33.86   0.017       0.043       0.004655   200   
/realsense/camera/color/camera_info   33.97   0.018       0.04        0.004321   199   

               topic                   rate   min_delta   max_delta   std_dev    window
=======================================================================================
/realsense/camera/color/image_raw     33.99   0.016       0.052       0.004629   234   
/realsense/camera/depth/image_raw     34.02   0.017       0.043       0.004589   234   
/realsense/camera/color/camera_info   34.13   0.018       0.04        0.004261   233   

               topic                   rate   min_delta   max_delta   std_dev    window
=======================================================================================
/realsense/camera/color/image_raw     34.1    0.016       0.052       0.004597   262   
/realsense/camera/depth/image_raw     34.13   0.014       0.043       0.004729   262   
/realsense/camera/color/camera_info   34.23   0.018       0.04        0.004296   261   

```

I can also subscribe to the depth image topic and visualize the depth map in Rviz. Here is the snapshot, lower left is the depth map:
[image]

Besides, I also tried to start from turtlebot and modified `kobuki_hexagons_r200.urdf.xacro` as follow would produce the same warning: `Did not receive data since 5 seconds!`

```
<?xml version="1.0"?>
<!--
    - Base      : kobuki
    - Stacks    : hexagons
[... 18 lines cut ...]
```

Warnings from turtlebot
```
[ WARN] [1554270428.847831910, 499.250000000]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=30).
/rtabmap/rtabmap subscribed to (approx sync):
   /odom,
   /camera/rgb/image_raw,
   /camera/depth/image_raw,
   /camera/rgb/camera_info
[... cut ...]
[ WARN] [1554270438.860224114, 509.250000000]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=30).
/rtabmap/rtabmap subscribed to (approx sync):
   /odom,
   /camera/rgb/image_raw,
   /camera/depth/image_raw,
   /camera/rgb/camera_info
```

matlabbe, 2019-04-04:
I just realized in the previous post that you launched only `rtabmap` node (without `rgbd_odometry` node), so /rtabmap/odom topic was indeed not published. In `realsense_rtabmap.launch`, to launch visual odometry, add the following in rtabmap namespace:
```xml
    <node name="rtabmap" pkg="rgbd_odometry" type="rgbd_odometry" output="screen">
          <param name="frame_id"            type="string" value="origin"/>
          <param name="approx_sync"        type="bool" value="false"/> 

          <remap from="rgb/image"       to="/realsense/camera/color/image_raw"/>
          <remap from="depth/image"     to="/realsense/camera/depth/image_raw"/>
          <remap from="rgb/camera_info" to="/realsense/camera/color/camera_info"/>
    </node>

```
In your last message, you are using the /odom topic as odometry, make sure it is published from the kuboki.

cheers,
Mathieu

ken012git, 2019-04-08:
It works for me. Thanks for catching the bug.

```
SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /rtabmap/rtabmap/Grid/FromDepth: true
 * /rtabmap/rtabmap/GridGlobal/MinSize: 20
 * /rtabmap/rtabmap/Icp/CorrespondenceRatio: 0.3
 * /rtabmap/rtabmap/Kp/MaxDepth: 4.0
 * /rtabmap/rtabmap/Mem/RehearsalSimilarity: 0.30
 * /rtabmap/rtabmap/Optimizer/Slam2D: true
 * /rtabmap/rtabmap/RGBD/AngularUpdate: 0.1
 * /rtabmap/rtabmap/RGBD/LinearUpdate: 0.1
 * /rtabmap/rtabmap/RGBD/OptimizeFromGraphEnd: false
 * /rtabmap/rtabmap/RGBD/OptimizeMaxError: 0.1
 * /rtabmap/rtabmap/RGBD/ProximityBySpace: true
 * /rtabmap/rtabmap/RGBD/ProximityPathMaxNeighbors: 0
 * /rtabmap/rtabmap/Reg/Force3DoF: true
 * /rtabmap/rtabmap/Reg/Strategy: 0
 * /rtabmap/rtabmap/Rtabmap/TimeThr: 700
 * /rtabmap/rtabmap/Vis/InlierDistance: 0.1
 * /rtabmap/rtabmap/Vis/MinInliers: 15
 * /rtabmap/rtabmap/database_path: rtabmap.db
 * /rtabmap/rtabmap/frame_id: origin
 * /rtabmap/rtabmap/map_negative_poses_ignored: True
 * /rtabmap/rtabmap/queue_size: 30
 * /rtabmap/rtabmap/subscribe_depth: True
 * /rtabmap/rtabmap/subscribe_scan: False
 * /rtabmap/rtabmap_odometry/approx_sync: False
 * /rtabmap/rtabmap_odometry/frame_id: origin

NODES
  /rtabmap/
    rtabmap (rtabmap_ros/rtabmap)
    rtabmap_odometry (rtabmap_ros/rgbd_odometry)

ROS_MASTER_URI=http://localhost:11311

process[rtabmap/rtabmap-1]: started with pid [11513]
process[rtabmap/rtabmap_odometry-2]: started with pid [11514]
[ INFO] [1554691905.638331717]: Starting node...
[ INFO] [1554691905.843780351]: Initializing nodelet with 8 worker threads.
[ INFO] [1554691905.917893956]: Initializing nodelet with 8 worker threads.
[ INFO] [1554691906.800374415, 4114.650000000]: Odometry: frame_id               = origin
[ INFO] [1554691906.800437312, 4114.650000000]: Odometry: odom_frame_id          = odom
[ INFO] [1554691906.800464724, 4114.650000000]: Odometry: publish_tf             = true
[ INFO] [1554691906.800486958, 4114.650000000]: Odometry: wait_for_transform     = true
[ INFO] [1554691906.800520627, 4114.650000000]: Odometry: wait_for_transform_duration  = 0.200000
[ INFO] [1554691906.800571422, 4114.650000000]: Odometry: initial_pose           = xyz=0.000000,0.000
000,0.000000 rpy=0.000000,-0.000000,0.000000
[ INFO] [1554691906.800594991, 4114.650000000]: Odometry: ground_truth_frame_id  = 
[ INFO] [1554691906.800616363, 4114.650000000]: Odometry: ground_truth_base_frame_id = origin
[ INFO] [1554691906.800638094, 4114.650000000]: Odometry: config_path            = 
[ INFO] [1554691906.800657615, 4114.650000000]: Odometry: publish_null_when_lost = true
[ INFO] [1554691906.800677883, 4114.650000000]: Odometry: guess_frame_id         = 
[ INFO] [1554691906.800700942, 4114.650000000]: Odometry: guess_min_translation  = 0.000000
[ INFO] [1554691906.800723207, 4114.650000000]: Odometry: guess_min_rotation     = 0.000000
[ INFO] [1554691906.839224333, 4114.688000000]: /rtabmap/rtabmap(maps): map_filter_radius          = 
0.000000
[ INFO] [1554691906.839281211, 4114.688000000]: /rtabmap/rtabmap(maps): map_filter_angle           = 
30.000000
[ INFO] [1554691906.839304592, 4114.688000000]: /rtabmap/rtabmap(maps): map_cleanup                = 
true
[ INFO] [1554691906.839327766, 4114.688000000]: /rtabmap/rtabmap(maps): map_negative_poses_ignored = 
true
[ INFO] [1554691906.839349361, 4114.688000000]: /rtabmap/rtabmap(maps): map_negative_scan_ray_tracing
[... 144 lines cut ...]
95s, Maps update=0.0002s pub=0.0004s (local map=2, WM=3)
[ INFO] [1554691960.706703398, 4165.770000000]: Odom: quality=229, std dev=0.032212m|0.064769rad, upd
ate time=0.166722s
[ INFO] [1554691960.875323287, 4165.926000000]: rtabmap (37): Rate=1.00s, Limit=0.700s, RTAB-Map=0.16
22s, Maps update=0.0002s pub=0.0003s (local map=2, WM=3)
[ INFO] [1554691966.128255827, 4170.782000000]: Odom: quality=225, std dev=0.010423m|0.052765rad, upd
ate time=0.186210s
[ INFO] [1554691966.286974088, 4170.943000000]: rtabmap (38): Rate=1.00s, Limit=0.700s, RTAB-Map=0.15
17s, Maps update=0.0001s pub=0.0003s (local map=2, WM=3)
[ INFO] [1554692019.737320417, 4222.591000000]: Odom: quality=214, std dev=0.011264m|0.050827rad, upd
ate time=0.219605s
[ INFO] [1554692019.849893498, 4222.684000000]: rtabmap (39): Rate=1.00s, Limit=0.700s, RTAB-Map=0.10
49s, Maps update=0.0002s pub=0.0002s (local map=2, WM=3)
[rtabmap/rtabmap_odometry-2] killing on exit
[rtabmap/rtabmap-1] killing on exit
rtabmap: Saving database/long-term memory... (located at /home/hychiang/.ros/rtabmap.db)
rtabmap: Saving database/long-term memory...done! (located at /home/hychiang/.ros/rtabmap.db, 7 MB)
shutting down processing monitor...
... shutting down processing monitor complete
done

```

Here is the output `ply` file from RTAB-MAP
[image]

ken012git, 2019-04-08:
@matlabbe Thank you for your help so far!
Also, I checked the turtlebot project again and followed your suggestion. As you suggested, I found rtab-map using the /odom topic as odometry published from the kuboki (from the imu and sensor defined in kobuki_gazebo.urdf.xacro) [[reference](http://wiki.ros.org/turtlebot_node)].

I tried to switch the rtab-map odometry from the `/odom` publiched by kuboki to `rgbd_odmetry` by setting the flag `rgbd_odometry` to true.

However, the odometry seems failed to be calculated from rgbd images due to not enough inliners from realsense r200 camera.

How do I fix this? 

Thanks.

```
<launch>
  <arg name="database_path"     default="rtabmap.db"/>
  <arg name="rgbd_odometry"     default="false"/>
  <arg name="rtabmapviz"        default="false"/>
  <arg name="localization"      default="false"/>
  <arg name="simulation"        default="false"/>
  <arg name="sw_registered"     default="false"/>
  <arg     if="$(arg localization)" name="args"  default=""/>
  <arg unless="$(arg localization)" name="args"  default="--delete_db_on_start"/>

  <arg     if="$(arg simulation)" name="rgb_topic"   default="/camera/rgb/image_raw"/>
  <arg unless="$(arg simulation)" name="rgb_topic"   default="/camera/rgb/image_rect_color"/>
  <arg     if="$(arg simulation)" name="depth_topic" default="/camera/depth/image_raw"/>
  <arg unless="$(arg simulation)" name="depth_topic" default="/camera/depth_registered/image_raw"/>
  <arg name="camera_info_topic" default="/camera/rgb/camera_info"/>

  <arg name="wait_for_transform"  default="0.2"/>
  <!--
      robot_state_publisher's publishing frequency in "turtlebot_bringup/launch/includes/robot.launch.xml"
      can be increase from 5 to 10 Hz to avoid some TF warnings.
  -->

  <!-- Navigation stuff (move_base) -->
  <include unless="$(arg simulation)" file="$(find turtlebot_bringup)/launch/3dsensor.launch">
     <arg     if="$(arg sw_registered)" name="depth_registration" value="false"/>
     <arg unless="$(arg sw_registered)" name="depth_registration" value="true"/>
  </include>
  <include file="$(find turtlebot_navigation)/launch/includes/move_base.launch.xml"/>

  <!-- Mapping -->
  <group ns="rtabmap">

    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="$(arg args)">
          <param name="database_path"       type="string" value="$(arg database_path)"/>
          <param name="frame_id"            type="string" value="base_footprint"/>
          <param name="wait_for_transform_duration"  type="double"   value="$(arg wait_for_transform)"/>
          <param name="subscribe_depth"     type="bool"   value="true"/>
          <param name="subscribe_scan"      type="bool"   value="false"/>
          <param name="map_negative_poses_ignored" type="bool" value="true"/>

          <!-- inputs -->
          <remap from="scan"            to="/scan"/>
          <remap from="rgb/image"       to="$(arg rgb_topic)"/>
          <remap from="depth/image"     to="$(arg depth_topic)"/>
          <remap from="rgb/camera_info" to="$(arg camera_info_topic)"/>
          <remap unless="$(arg rgbd_odometry)" from="odom" to="/odom"/>
          <!-- output -->
          <remap from="grid_map" to="/map"/>

          <!-- RTAB-Map's parameters: do "rosrun rtabmap rtabmap (double-dash)params" to see the list of available parameters. -->
          <param name="RGBD/ProximityBySpace"        type="string" value="true"/>   <!-- Local loop closure detection (using estimated position) with locations in WM -->
          <param name="RGBD/OptimizeFromGraphEnd"    type="string" value="false"/>  <!-- Set to false to generate map correction between /map and /odom -->
          <param name="Kp/MaxDepth"                  type="string" value="4.0"/>
          <param name="Reg/Strategy"                 type="string" value="0"/>      <!-- Loop closure transformation: 0=Visual, 1=ICP, 2=Visual+ICP -->
          <param name="Icp/CorrespondenceRatio"      type="string" value="0.3"/>
          <param name="Vis/MinInliers"               type="string" value="15"/>      <!-- 3D visual words minimum inliers to accept loop closure -->
          <param name="Vis/InlierDistance"           type="string" value="0.1"/>    <!-- 3D visual words correspondence distance -->
          <param name="RGBD/AngularUpdate"           type="string" value="0.1"/>    <!-- Update map only if the robot is moving -->
[... 400 lines cut ...]
ow inliers after bundle adjustment: 16<20"
[ INFO] [1554706985.302510847, 7.120000000]: Odom: quality=16, std dev=0.000000m|0.000000rad, update 
time=0.024186s
[ WARN] (2019-04-08 15:03:05.402) OdometryF2M.cpp:469::computeTransform() Registration failed: "Too l
ow inliers after bundle adjustment: 15<20"
[ INFO] [1554706985.402243256, 7.220000000]: Odom: quality=15, std dev=0.000000m|0.000000rad, update 
time=0.022933s
[ WARN] (2019-04-08 15:03:05.512) OdometryF2M.cpp:469::computeTransform() Registration failed: "Too l
ow inliers after bundle adjustment: 16<20"
[ INFO] [1554706985.513145989, 7.330000000]: Odom: quality=16, std dev=0.000000m|0.000000rad, update 
time=0.034191s
[ WARN] (2019-04-08 15:03:05.628) OdometryF2M.cpp:469::computeTransform() Registration failed: "Too l
ow inliers after bundle adjustment: 16<20"
[ INFO] [1554706985.629181949, 7.450000000]: Odom: quality=16, std dev=0.000000m|0.000000rad, update 
time=0.025295s
[ WARN] (2019-04-08 15:03:05.723) OdometryF2M.cpp:469::computeTransform() Registration failed: "Too l
ow inliers after bundle adjustment: 15<20"
[ INFO] [1554706985.723438859, 7.540000000]: Odom: quality=15, std dev=0.000000m|0.000000rad, update 
time=0.024002s
[ WARN] (2019-04-08 15:03:05.825) OdometryF2M.cpp:469::computeTransform() Registration failed: "Too l
ow inliers after bundle adjustment: 18<20"
[ INFO] [1554706985.825990287, 7.640000000]: Odom: quality=18, std dev=0.000000m|0.000000rad, update 
time=0.023376s

```

matlabbe, 2019-04-10:
Visual odometry won't work well in that kind of simulated environment (not a lot of discriminative visual features). I suggest to stick with kuboki odometry.

ju-mingyue, 2021-07-06:
@matlabbe 
[https://github.com/introlab/rtabmap/issues/747](url)
Unfortunately, I also encountered some problems when using rtabmap, so I hope you can provide me with some solutions, I am very grateful