introlab/rtabmap_ros #190: demo_two_kinects.launch failed in load nodelet /camera1/rgbd_sync and /camera2/rgbd_sync

Severn-Vergil, 2017-07-26:
Hi Mathieu,

I'm trying to run demo_two_kinects.launch, it works well on my previous notebook a week ago(rtabmap version 0.11.12), and I'm trying to run it on my new notebook, with new installed rtabmap and rtabmap_ros(rtabmap version 0.13.0).

The logs show that the nodelet /camera1/rgbd_sync not match with the type /rtabmap_ros/rgbd_sync, and I guess it can be modified from the param "group ns = "camera1" " from demo_two_kinects.launch, and I'm trying to fix it out.

The logs are listed below:

** roslaunch rtabmap_ros demo_two_kinects.launch  **
... logging to /home/severn/.ros/log/96f0851c-71ae-11e7-8dae-c9342e4caeee/roslaunch-severn-21286.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://severn:33065/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /rtabmap/rgbd_odometry/Odom/FillInfoData: true
 * /rtabmap/rgbd_odometry/Odom/Strategy: 0
 * /rtabmap/rgbd_odometry/OdomF2M/MaxSize: 1000
 * /rtabmap/rgbd_odometry/Vis/CorNNType: 3
 * /rtabmap/rgbd_odometry/Vis/FeatureType: 6
 * /rtabmap/rgbd_odometry/Vis/InlierDistance: 0.02
 * /rtabmap/rgbd_odometry/Vis/MaxDepth: 4.0
 * /rtabmap/rgbd_odometry/Vis/MinInliers: 20
 * /rtabmap/rgbd_odometry/frame_id: base_link
 * /rtabmap/rgbd_odometry/rgbd_cameras: 2
 * /rtabmap/rgbd_odometry/subscribe_rgbd: True
 * /rtabmap/rgbd_odometry/wait_for_transform: True
 * /rtabmap/rtabmap/Grid/FromDepth: false
 * /rtabmap/rtabmap/Vis/InlierDistance: 0.02
 * /rtabmap/rtabmap/Vis/MinInliers: 10
 * /rtabmap/rtabmap/frame_id: base_link
 * /rtabmap/rtabmap/gen_scan: True
 * /rtabmap/rtabmap/map_negative_poses_ignored: False
 * /rtabmap/rtabmap/map_negative_scan_empty_ray_tracing: False
 * /rtabmap/rtabmap/rgbd_cameras: 2
 * /rtabmap/rtabmap/subscribe_depth: False
 * /rtabmap/rtabmap/subscribe_rgbd: True
 * /rtabmap/rtabmap/wait_for_transform: True
 * /rtabmap/rtabmapviz/frame_id: base_link
 * /rtabmap/rtabmapviz/rgbd_cameras: 2
 * /rtabmap/rtabmapviz/subscribe_depth: False
 * /rtabmap/rtabmapviz/subscribe_odom_info: True
 * /rtabmap/rtabmapviz/subscribe_rgbd: True
 * /rtabmap/rtabmapviz/wait_for_transform: True

NODES
  /camera2/
    rgbd_sync (nodelet/nodelet)
  /camera1/
    rgbd_sync (nodelet/nodelet)
  /rtabmap/
    rgbd_odometry (rtabmap_ros/rgbd_odometry)
    rtabmap (rtabmap_ros/rtabmap)
    rtabmapviz (rtabmap_ros/rtabmapviz)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[camera1/rgbd_sync-1]: started with pid [21304]
process[camera2/rgbd_sync-2]: started with pid [21305]
process[rtabmap/rgbd_odometry-3]: started with pid [21306]
process[rtabmap/rtabmap-4]: started with pid [21311]
process[rtabmap/rtabmapviz-5]: started with pid [21318]
[... 164 lines cut ...]
process[camera1/disparity_registered_hw-20]: started with pid [19936]
process[robot_camera1_link-21]: started with pid [19946]
process[camera1_base_link-22]: started with pid [19954]
process[camera1_base_link1-23]: started with pid [19960]
process[camera1_base_link2-24]: started with pid [19962]
process[camera1_base_link3-25]: started with pid [19968]
[ INFO] [1501038168.352711371]: Initializing nodelet with 4 worker threads.
Warning: USB events thread - failed to set priority. This might cause loss of data...
Warning: USB events thread - failed to set priority. This might cause loss of data...
[ INFO] [1501038169.271017512]: Number devices connected: 2
[ INFO] [1501038169.271121650]: 1. device on bus 003:34 is a SensorKinect (2ae) from PrimeSense (45e) with serial id '0000000000000000'
[ INFO] [1501038169.271148572]: 2. device on bus 003:38 is a SensorKinect (2ae) from PrimeSense (45e) with serial id 'A00367D04979136A'
[ INFO] [1501038169.274536702]: Searching for device with bus@address = 3@34
[ INFO] [1501038170.314469921]: Opened 'SensorKinect' on bus 3:34 with serial number '0000000000000000'
[ INFO] [1501038170.455692430]: rgb_frame_id = 'camera1_rgb_optical_frame' 
[ INFO] [1501038170.455751329]: depth_frame_id = 'camera1_depth_optical_frame' 
[ WARN] [1501038170.476676677]: Camera calibration file /home/severn/.ros/camera_info/depth_0000000000000000.yaml not found.
[ WARN] [1501038170.476735874]: Using default parameters for IR camera calibration.
[ERROR] [1501038186.505226194]: Failed to load nodelet [/camera1/rgbd_sync] of type [rtabmap_ros/rgbd_sync] even after refreshing the cache: MultiLibraryClassLoader: Could not create object of class type rtabmap_ros::RGBDSync as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ERROR] [1501038186.505396001]: The error before refreshing the cache was: Failed to load library /home/severn/rtabmap_ws/catkin_ws/devel/lib//librtabmap_ros.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = dlopen: cannot load any more object with static TLS)

Thank you in advance, 
Severn
[image]
[image]

matlabbe, 2017-07-28:
`rgbd_sync` nodelets cannot be loaded, not sure why as they are loaded correctly on my system. Could be a compilation problem. What if you clean the catkin build directory and redo a catkin_make?

nathangeorge1, 2020-04-16:
I have the same/similar error, but cleaning the catkin build/devel directory didn't help me; this is the error on the Jetson Nano

```
[N=/rtabmap/rgbd_sync,F=Loader::load-300]: Failed to load nodelet [/rtabmap/rgbd_sync] of type [rtabmap_ros/rgbd_sync] even after refreshing the cache: MultiLibraryClassLoader: Could not create object of class type rtabmap_ros::RGBDSync as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[N=/rtabmap/rgbd_sync,F=Loader::load-301]: The error before refreshing the cache was: Failed to load library /opt/ros/melodic/lib//librtabmap_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /opt/ros/melodic/lib//librtabmap_plugins.so: undefined symbol: _ZN10costmap_2d12CostmapLayer9clearAreaEiiii)
[rtabmap/rgbd_sync-4] process has died [pid 27738, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet standalone rtabmap_ros/rgbd_sync /rtabmap/grid_map:=/map /cmd_vel/managed:=/motor_controller/command /cmd_vel:=/motor_controller/command_no_stamp rgb/image:=/zed/zed_node/rgb/image_rect_color depth/image:=/zed/zed_node/depth/depth_registered rgb/camera_info:=/zed/zed_node/rgb/camera_info rgbd_image:=rgbd_image __name:=rgbd_sync __log:=/home/nvidia/.ros/log/698aff04-800a-11ea-9abc-06159326cf59/rtabmap-rgbd_sync-4.log].
log file: /home/nvidia/.ros/log/698aff04-800a-11ea-9abc-06159326cf59/rtabmap-rgbd_sync-4*.log

```

There appears to be 2 slashes on one of the lines:

`/opt/ros/melodic/lib//librtabmap_plugins.so: undefined symbol`

nathangeorge1, 2020-04-16:
I fixed this error by doing `sudo apt update` and `sudo apt upgrade`.

closed 2020-04-17 by matlabbe

lifuguan, 2020-07-22:
So is there any solution for this issue? I've faced it in version 0.19.3.

derektan95, 2021-01-22:
I am not sure if this fixes anyone's issues. When I faced this issue on my Nvidia TX2 (while attempting to compile opencv with CUDA), I realized that I forgot to run the command: `sudo ldconfig -v`.

hank870907, 2022-07-05:
I am facing the same issue on Nvidia agx xavier!