introlab/rtabmap_ros #191: Turtlebot 2 with Kobuki : malformed sub-payload detected

alduxvm, 2017-07-28:
Hi,

Devices involved: Turtlebot 2 (Kobuki base), Kinect, Acer netbook.

Starting with a fresh install of Ubuntu 16.04 and ROS Kinetic (base). Then installing `ros-kinetic-turtlebot-bringup ros-kinetic-turtlebot-navigation ros-kinetic-turtlebot-teleop` the robots works fine as normally, no error detected.

Then we proceed to install `ros-kinetic-rtabmap-ros` and when launching the bringup minimal, we start having the following error:
```
[ERROR] [1501263662.514569924]: Kobuki : malformed sub-payload detected. [60][170][3C AA 55 4D 01 0F ]
```
That error appears regularly whenever the bringup minimal is launched. There is no way to get rid of such error until a fresh install of OS is performed. We tried with no success:
* Removing rtabmap
* Removing all turtlebot packages and installing them again
* autoremove of not used packages

Its not a hardware problem with my robot (or netbook) because if I fresh install the OS and do not install rtabmap, the problem does not appear. The malformed sub-payload only starts after installing rtabmap. 

Maybe something of the new commits or dependencies affected the entire current installation of rtabmap for turtlebot 2 + ros-kinetic? because a few months ago we did not had that issue.

Any ideas?

Thanks a lot!

matlabbe, 2017-07-28:
Just to be sure, rtabmap is not even started at all? just installed? Very strange. Maybe one of the dependencies cited [here](https://github.com/introlab/rtabmap_ros/blob/master/package.xml) could be the problem.

Other info: http://answers.ros.org/question/52203/kobuki-malformed-subpayload/

alduxvm, 2017-07-28:
Yes, rtabmap works, we can start mapping but after a while the 3D sensor fails and rtabmap fails as well and I need to restart everything... 

Yes, I went to that link when the problem start to arise, but the issue gets resolved with a fresh install of the OS+ROS, which is when I think it has to do with a dependency of rtabmap, but which one? still a mystery...

tiagopn, 2017-09-26:
Dear all. I solve this problem by updating Turtlebot firmware. My turtlebot was 1.1.3 and I update it to 1.1.4 (not the non tested latest one but a newer one).

It seems to have solved the problem.