introlab/rtabmap_ros #195: Armhf indigo builds are failing due to running out of memory

tfoote, 2017-08-10:
We don't have instrumentation of the memory usage. But there has not been a successful build since late july when a rebuild was triggered for new dependencies. And all of those builds that ran to the end seem to have been terminated for out of memory issues.

Looking at the last build it was a new release of geometry2 which might be causing increased memory usage that pushed this package over the limit.

matlabbe, 2017-08-11:
We can blacklist the armhf build as I didn't plan to update the Indigo version for backward compatibility issues (interface slightly different with latest code version). The latest version (which is on kinetic/lunar) has CoreWrapper.cpp split into multiple files to reduce memory used per file.

tfoote, 2017-08-11:
Ok, thanks I've blacklisted it in https://github.com/ros-infrastructure/ros_buildfarm_config/pull/90

closed 2017-08-11 by tfoote