introlab/rtabmap_ros #196: rtabmap installation repeatedly failing

indigomega, 2017-09-03:
Hey, I've tried to install rtabmap for ROS all day now on my Jetson TK1. The first problem was when I tried to apt-get install the indigo-ros package, to which I got E: Unable to locate package ros-indigo-rtabmap-ros. Ok... I just decided to continue following the instructions and just clone the code. Cmake then refused to work with me but I got it out of the way by installing PCL. Now, I can't do the 'make' command. I've tried 'make', 'make -j4', created a swapfile with an SD card, nothing. Here is the failure log:

`ubuntu@tegra-ubuntu:~/rtabmap/build$ make
[  6%] Built target rtabmap_utilite
[  6%] Built target res_tool
[  6%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/CameraRGBD.cpp.o
/home/ubuntu/rtabmap/corelib/src/CameraRGBD.cpp: In destructor 'virtual rtabmap::CameraRealSense::~CameraRealSense()':
/home/ubuntu/rtabmap/corelib/src/CameraRGBD.cpp:2012:19: error: 'rs::source' has not been declared
    dev_->stop(rs::source::all_sources);
                   ^
/home/ubuntu/rtabmap/corelib/src/CameraRGBD.cpp: In member function 'virtual bool rtabmap::CameraRealSense::init(const string&, const string&)':
/home/ubuntu/rtabmap/corelib/src/CameraRGBD.cpp:2071:18: error: 'rs::source' has not been declared
   dev_->stop(rs::source::all_sources);
                  ^
/home/ubuntu/rtabmap/corelib/src/CameraRGBD.cpp:2114:7: warning: unused variable 'computeOdometry' [-Wunused-variable]
  bool computeOdometry = false;
       ^
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/CameraRGBD.cpp.o] Error 1
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make: *** [all] Error 2
ubuntu@tegra-ubuntu:~/rtabmap/build$ cmake -DCMAKE_INSTALL_PREFIX=~/catkin_ws/devel ..
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- Found OpenMP
-- Found OpenCV: /usr/include/opencv;/usr/include
-- Found PCL: /usr/include/pcl-1.7;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/vtk-5.8
-- Found ZLIB: /usr/include
-- Found DC1394: /usr/include/dc1394
-- Found RealSense: /usr/local/include
-- Found octomap: /opt/ros/indigo/include
-- Found Pthreads
-- --------------------------------------------
-- Info :
--   Version : 0.13.2
--   CMAKE_INSTALL_PREFIX = /home/ubuntu/catkin_ws/devel
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -Wno-deprecated -fopenmp -std=c++11
--   With OpenCV 2 nonfree module (SIFT/SURF) = NO (not found, License: BSD)
--   With Freenect             = NO (libfreenect not found)
--   With OpenNI2              = NO (OpenNI2 not found)
--   With Freenect2            = NO (libfreenect2 not found)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--   With g2o                  = NO (WITH_G2O=OFF)
--   With GTSAM                = NO (WITH_GTSAM=OFF)
--   With VERTIGO              = NO (GTSAM or g2o required)
--   With cvsba                = NO (cvsba not found)
--   With libpointmatcher      = NO (libpointmatcher not found)
--   With ZED                  = NO (ZED sdk not found)
--   With RealSense            = YES (License: Apache-2)
--   With RealSenseSlam        = NO (WITH_REALSENSE_SLAM=OFF)
--   With OCTOMAP              = YES (License: BSD)
--   With CPUTSDF              = NO (CPUTSDF not found)
--   With libfovis             = NO (libfovis not found)
--   With libviso2             = NO (libviso2 not found)
--   With dvo_core             = NO (dvo_core not found)
--   With ORB_SLAM2            = NO (ORB_SLAM2 not found, make sure environment variable ORB_SLAM2_ROOT_DIR is set)
--   With Qt4                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/rtabmap/build
ubuntu@tegra-ubuntu:~/rtabmap/build$ make
[  6%] Built target rtabmap_utilite
[  6%] Built target res_tool
[  6%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/CameraRGBD.cpp.o
/home/ubuntu/rtabmap/corelib/src/CameraRGBD.cpp: In destructor 'virtual rtabmap::CameraRealSense::~CameraRealSense()':
/home/ubuntu/rtabmap/corelib/src/CameraRGBD.cpp:2012:19: error: 'rs::source' has not been declared
    dev_->stop(rs::source::all_sources);
                   ^
/home/ubuntu/rtabmap/corelib/src/CameraRGBD.cpp: In member function 'virtual bool rtabmap::CameraRealSense::init(const string&, const string&)':
/home/ubuntu/rtabmap/corelib/src/CameraRGBD.cpp:2071:18: error: 'rs::source' has not been declared
   dev_->stop(rs::source::all_sources);
                  ^
/home/ubuntu/rtabmap/corelib/src/CameraRGBD.cpp:2114:7: warning: unused variable 'computeOdometry' [-Wunused-variable]
  bool computeOdometry = false;
       ^
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/CameraRGBD.cpp.o] Error 1
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make: *** [all] Error 2
`

I already installed libRealSense on my TK1, but it always seems to fail when it builds the CameraRGBD file. I've also tried 'cmake -DWITH_G2O=OFF -DWITH_GTSAM=OFF ..' but I receive pretty much the same error. What the hell??? Please help!!

matlabbe, 2017-09-03:
Hi,

Did you try with WITH_REALSENSE=OFF? If you need realsense support in rtabmap, the error `'rs::source' has not been declared` may be that realsense library version you are using is not the same as the one I tested on. Normally, in the included [librealsense/rs.hpp](https://github.com/introlab/rtabmap/blob/b93ca148a83d0c1f3a2281a7a37718c9873d3d78/corelib/src/CameraRGBD.cpp#L67), the rs::source should be declared [here](https://github.com/IntelRealSense/librealsense/blob/8a5e5fb6f51707881f133fcb922a3bf95093b12e/include/librealsense/rs.hpp#L219).

cheers,
Mathieu

indigomega, 2017-09-04:
Oh my god you're a saint!! I managed to install the rtabmap package, but there is a problem with the ROS package (ugh...). Here is the fail log below:
`[ 54%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/nodelets/rgbd_odometry.cpp.o
/tmp/ccwUffEa.s: Assembler messages:
/tmp/ccwUffEa.s: Fatal error: can't close CMakeFiles/rtabmap_ros.dir/src/nodelets/rgbd_odometry.cpp.o: No space left on device
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/nodelets/rgbd_odometry.cpp.o] Error 1
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j1" failed
`
Strange, I thought the SD card swapfile would prove fine, and also system information shows I still have about 600 MB of RAM left to go at the point it foiles. Of course, I used the 'catkin_make -j1'.

Also, does the fact that I added -DWITH_REALSENSE=OFF mean that I can't use the R200 with rtabmap?

Thanks for the help!

indigomega, 2017-09-04:
I should add that the if I do not use -j1 in the catkin_make command the entire computer (the TK1, not my computer) freezes up and I have to manually reset it

indigomega, 2017-09-04:
Hey Matthieu,

I just figured it out, turns out I ran out of space in the root directory (oops lol). A quick google search helped me clean up a bit and I managed to install rtabmap for ROS. Thanks for your help! Still though, one question left, does the fact that I did not build the package with librealsense mean I will have to build it again? I'm not very sure, even though in your tutorial it doesn't seem to matter.

matlabbe, 2017-09-04:
Indeed, in ros tutorials, you don't need realsense built in rtabmap as you would use the ros package of the sensor to stream images to rtabmap node.

indigomega, 2017-09-04:
Excellent, just what I needed to know, thanks!

indigomega, 2017-09-04:
Great news! Tried to build some other packages today only for rtabmap to tell me octomap and costmap_2d are missing, after installing them this is the fail log:

`-- +++ processing catkin package: 'rtabmap_ros'
-- ==> add_subdirectory(rtabmap_ros)
-- Using these message generators: gencpp;genlisp;genpy
-- rtabmap_ros: 12 messages, 6 services
-- WITH costmap_2d
-- WITH octomap
-- WITH rviz
-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntu/catkin_ws/build/CMakeFiles/CMakeError.log".
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
`

I just don't get it, why does it keep failing?

indigomega, 2017-09-05:
OK wow... after some pretty intense investigations into the catkin_make error log I found that my arch did not have a gtest directory in /usr/src/. What I ended up doing is copying the entire directory from my computer which managed to install rtabmap (quite smoothly I'd say) on Lunar, pasting it on to a USB drive, and then pasting again in the /usr/src/ directory in the TK1. I really hope this is the end of my problems. Hope this ends up helping someone with a TK1 with Indigo ROS (21.4).

closed 2017-09-05 by indigomega