introlab/rtabmap_ros #198: Question about validation datasets

BrennoCaldato, 2017-09-09:
I'm currently facing a loop closure problem and I looked at the ground-truth of City Centre, New College and Lip6 datasets that you have provided at GitHub page, and I notice that the diagonal line of the ground-truth is set to 0. Why didn't you consider the diagonal images as a loop closure?

Thanks

matlabbe, 2017-09-10:
The diagonal doesn't correspond to real loop closures. Having these hypotheses as loop closures doesn't make sense, as the current image is always identical to itself. Note that for the UdeS_1Hz dataset (see ground truth [here](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/1/16/UdeS_1Hz.png)), I put some grey pixels on the diagonal so that loop closure detection recall/precision performance [is not affected](https://github.com/introlab/rtabmap/blob/4c2dad453d78eb2cca514cf0fe6d2f8932f80371/Matlab/ShowLogs/getPrecisionRecall.m#L47) (positively or negatively) if the loop closures are found on consecutive images. Such problem can be seen at 1:16 on this [video](https://www.youtube.com/watch?v=CAk-QGMlQmI), loop closures have been detected on just viewed locations. While these detections don't add errors to map, they cannot reduce the global error either, as they are not real loop closures. However, they can indeed be used to reduce local errors, but the main research on loop closure detection (to compare with other approaches) is more about loop closure where we moved out from a location and then we came back afterwards. 

cheers,
Mathieu

BrennoCaldato, 2017-09-12:
Thank you for the explanation!!