introlab/rtabmap_ros #199: make error from g2o

paulsammut, 2017-10-12:
I am getting a whole host of make errors stemming from g2o. 
`
/home/paul/rtabmap/corelib/src/OptimizerG2O.cpp: In member function ‘virtual std::map<int, rtabmap::Transform> rtabmap::OptimizerG2O::optimize(int, const std::map<int, rtabmap::Transform>&, const std::multimap<int, rtabmap::Link>&, std::__cxx11::list<std::map<int, rtabmap::Transform> >*, double*, int*)’:
/home/paul/rtabmap/corelib/src/OptimizerG2O.cpp:200:50: error: no matching function for call to ‘g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> >::BlockSolver(SlamLinearEigenSolver*&)’
    blockSolver = new SlamBlockSolver(linearSolver);
                                                  ^
In file included from /usr/local/include/g2o/core/block_solver.h:199:0,
                 from /home/paul/rtabmap/corelib/src/OptimizerG2O.cpp:42:
/usr/local/include/g2o/core/block_solver.hpp:40:1: note: candidate: g2o::BlockSolver<Traits>::BlockSolver(std::unique_ptr<typename Traits::LinearSolverType>) [with Traits = g2o::BlockSolverTraits<-1, -1>; typename Traits::LinearSolverType = g2o::LinearSolver<Eigen::Matrix<double, -1, -1> >]
 BlockSolver<Traits>::BlockSolver(std::unique_ptr<LinearSolverType> linearSolver)
 ^
/usr/local/include/g2o/core/block_solver.hpp:40:1: note:   no known conversion for argument 1 from ‘SlamLinearEigenSolver* {aka g2o::LinearSolverEigen<Eigen::Matrix<double, -1, -1> >*}’ to ‘std::unique_ptr<g2o::LinearSolver<Eigen::Matrix<double, -1, -1> >, std::default_delete<g2o::LinearSolver<Eigen::Matrix<double, -1, -1> > > >’
/home/paul/rtabmap/corelib/src/OptimizerG2O.cpp:208:50: error: no matching function for call to ‘g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> >::BlockSolver(SlamLinearCholmodSolver*&)’
    blockSolver = new SlamBlockSolver(linearSolver);
                                                  ^
In file included from /usr/local/include/g2o/core/block_solver.h:199:0,
                 from /home/paul/rtabmap/corelib/src/OptimizerG2O.cpp:42:
/usr/local/include/g2o/core/block_solver.hpp:40:1: note: candidate: g2o::BlockSolver<Traits>::BlockSolver(std::unique_ptr<typename Traits::LinearSolverType>) [with Traits = g2o::BlockSolverTraits<-1, -1>; typename Traits::LinearSolverType = g2o::LinearSolver<Eigen::Matrix<double, -1, -1> >]
 BlockSolver<Traits>::BlockSolver(std::unique_ptr<LinearSolverType> linearSolver)
 ^
/usr/local/include/g2o/core/block_solver.hpp:40:1: note:   no known conversion for argument 1 from ‘SlamLinearCholmodSolver* {aka g2o::LinearSolverCholmod<Eigen::Matrix<double, -1, -1> >*}’ to ‘std::unique_ptr<g2o::LinearSolver<Eigen::Matrix<double, -1, -1> >, std::default_delete<g2o::LinearSolver<Eigen::Matrix<double, -1, -1> > > >’
/home/paul/rtabmap/corelib/src/OptimizerG2O.cpp:217:50: error: no matching function for call to ‘g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> >::BlockSolver(SlamLinearCSparseSolver*&)’
    blockSolver = new SlamBlockSolver(linearSolver);
                                                  ^
In file included from /usr/local/include/g2o/core/block_solver.h:199:0,
                 from /home/paul/rtabmap/corelib/src/OptimizerG2O.cpp:42:
/usr/local/include/g2o/core/block_solver.hpp:40:1: note: candidate: g2o::BlockSolver<Traits>::BlockSolver(std::unique_ptr<typename Traits::LinearSolverType>) [with Traits = g2o::BlockSolverTraits<-1, -1>; typename Traits::LinearSolverType = g2o::LinearSolver<Eigen::Matrix<double, -1, -1> >]
 BlockSolver<Traits>::BlockSolver(std::unique_ptr<LinearSolverType> linearSolver)
 ^
/usr/local/include/g2o/core/block_solver.hpp:40:1: note:   no known conversion for argument 1 from ‘SlamLinearCSparseSolver* {aka g2o::LinearSolverCSparse<Eigen::Matrix<double, -1, -1> >*}’ to ‘std::unique_ptr<g2o::LinearSolver<Eigen::Matrix<double, -1, -1> >, std::default_delete<g2o::LinearSolver<Eigen::Matrix<double, -1, -1> > > >’
/home/paul/rtabmap/corelib/src/OptimizerG2O.cpp:225:50: error: no matching function for call to ‘g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> >::BlockSolver(SlamLinearPCGSolver*&)’
    blockSolver = new SlamBlockSolver(linearSolver);
                                                  ^
In file included from /usr/local/include/g2o/core/block_solver.h:199:0,
                 from /home/paul/rtabmap/corelib/src/OptimizerG2O.cpp:42:
/usr/local/include/g2o/core/block_solver.hpp:40:1: note: candidate: g2o::BlockSolver<Traits>::BlockSolver(std::unique_ptr<typename Traits::LinearSolverType>) [with Traits = g2o::BlockSolverTraits<-1, -1>; typename Traits::LinearSolverType = g2o::LinearSolver<Eigen::Matrix<double, -1, -1> >]
 BlockSolver<Traits>::BlockSolver(std::unique_ptr<LinearSolverType> linearSolver)
 ^
/usr/local/include/g2o/core/block_solver.hpp:40:1: note:   no known conversion for argument 1 from ‘SlamLinearPCGSolver* {aka g2o::LinearSolverPCG<Eigen::Matrix<double, -1, -1> >*}’ to ‘std::unique_ptr<g2o::LinearSolver<Eigen::Matrix<double, -1, -1> >, std::default_delete<g2o::LinearSolver<Eigen::Matrix<double, -1, -1> > > >’
/home/paul/rtabmap/corelib/src/OptimizerG2O.cpp:230:80: error: no matching function for call to ‘g2o::OptimizationAlgorithmGaussNewton::OptimizationAlgorithmGaussNewton(SlamBlockSolver*&)’
    optimizer.setAlgorithm(new g2o::OptimizationAlgorithmGaussNewton(blockSolver));
                                                                                ^
In file included from /home/paul/rtabmap/corelib/src/OptimizerG2O.cpp:45:0:
/usr/local/include/g2o/core/optimization_algorithm_gauss_newton.h:46:16: note: candidate: g2o::OptimizationAlgorithmGaussNewton::OptimizationAlgorithmGaussNewton(std::unique_ptr<g2o::Solver>)
       explicit OptimizationAlgorithmGaussNewton(std::unique_ptr<Solver> solver);
                ^
/usr/local/include/g2o/core/optimization_algorithm_gauss_newton.h:46:16: note:   no known conversion for argument 1 from ‘SlamBlockSolver* {aka g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> >*}’ to ‘std::unique_ptr<g2o::Solver>’
/home/paul/rtabmap/corelib/src/OptimizerG2O.cpp:234:78: error: no matching function for call to ‘g2o::OptimizationAlgorithmLevenberg::OptimizationAlgorithmLevenberg(SlamBlockSolver*&)’
    optimizer.setAlgorithm(new g2o::OptimizationAlgorithmLevenberg(blockSolver));
                                                                              ^
In file included from /home/paul/rtabmap/corelib/src/OptimizerG2O.cpp:46:0:
/usr/local/include/g2o/core/optimization_algorithm_levenberg.h:47:16: note: candidate: g2o::OptimizationAlgorithmLevenberg::OptimizationAlgorithmLevenberg(std::unique_ptr<g2o::Solver>)
       explicit OptimizationAlgorithmLevenberg(std::unique_ptr<Solver> solver);
                ^
/usr/local/include/g2o/core/optimization_algorithm_levenberg.h:47:16: note:   no known conversion for argument 1 from ‘SlamBlockSolver* {aka g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> >*}’ to ‘std::unique_ptr<g2o::Solver>’
/home/paul/rtabmap/corelib/src/OptimizerG2O.cpp: In member function ‘virtual std::map<int, rtabmap::Transform> rtabmap::OptimizerG2O::optimizeBA(int, const std::map<int, rtabmap::Transform>&, const std::multimap<int, rtabmap::Link>&, const std::map<int, rtabmap::CameraModel>&, std::map<int, cv::Point3_<float> >&, const std::map<int, std::map<int, cv::Point3_<float> > >&, std::set<int>*)’:
/home/paul/rtabmap/corelib/src/OptimizerG2O.cpp:699:76: error: no matching function for call to ‘g2o::BlockSolver<g2o::BlockSolverTraits<6, 3> >::BlockSolver(g2o::BlockSolver<g2o::BlockSolverTraits<6, 3> >::LinearSolverType*&)’
   g2o::BlockSolver_6_3 * solver_ptr = new g2o::BlockSolver_6_3(linearSolver);
                                                                            ^
In file included from /usr/local/include/g2o/core/block_solver.h:199:0,
                 from /home/paul/rtabmap/corelib/src/OptimizerG2O.cpp:42:
/usr/local/include/g2o/core/block_solver.hpp:40:1: note: candidate: g2o::BlockSolver<Traits>::BlockSolver(std::unique_ptr<typename Traits::LinearSolverType>) [with Traits = g2o::BlockSolverTraits<6, 3>; typename Traits::LinearSolverType = g2o::LinearSolver<Eigen::Matrix<double, 6, 6, 0> >]
 BlockSolver<Traits>::BlockSolver(std::unique_ptr<LinearSolverType> linearSolver)
 ^
/usr/local/include/g2o/core/block_solver.hpp:40:1: note:   no known conversion for argument 1 from ‘g2o::BlockSolver<g2o::BlockSolverTraits<6, 3> >::LinearSolverType* {aka g2o::LinearSolver<Eigen::Matrix<double, 6, 6, 0> >*}’ to ‘std::unique_ptr<g2o::LinearSolver<Eigen::Matrix<double, 6, 6, 0> >, std::default_delete<g2o::LinearSolver<Eigen::Matrix<doub [... cut ...]
/usr/local/include/g2o/core/optimization_algorithm_levenberg.h:47:16: note:   no known conversion for argument 1 from ‘g2o::BlockSolver_6_3* {aka g2o::BlockSolver<g2o::BlockSolverTraits<6, 3> >*}’ to ‘std::unique_ptr<g2o::Solver>’
`

I have installed g2o fresh from source and installed its dependencies before building:
libsuitesparse-dev - qtdeclarative5-dev - qt5-qmake - libqglviewer-dev -libeigen3-dev

Cmake finds g2o with the following: 
-- Found g2o: /usr/local/include;/usr/include/suitesparse;/usr/include/suitesparse

Thanks!

matlabbe, 2017-10-12:
I suggest to use the **c++03** branch of this g2o fork: https://github.com/felixendres/g2o/tree/c++03 or the libg2o built by ROS (e.g., on Ubuntu 16.04: `sudo apt-get install ros-kinetic-libg2o`).

The errors above are caused by a change in API on August 7 (https://github.com/RainerKuemmerle/g2o/commit/78280034b1983399cad3f93974b05b823ee20c25). I cannot find a way to detect this change on compilation (like a g2o version define), to not break backward compatibility with wide released g2o binaries in current and previous ROS versions. 

For your last question, no, g2o is not required.

cheers,
Mathieu

paulsammut, 2017-10-12:
Using the c++03 of that fork worked, thanks! I had to disable the examples (they complained about c++11 stuff) and the apps from the g2o build to get it to compile. I then recompiled rtabmap with the standard flags and it worked, thanks! 

Quick question, how could I "uninstall" the g2o I built from source, and installed with sudo make install, so I can switch to the ROS g2o binaries? Thanks!

matlabbe, 2017-10-13:
Unfortunately, g2o doesn't provide a "`$ sudo make uninstall`" . The best way is to do "sudo make install", then look at the base directories that are installed, remove them manually... like:
```bash
$ sudo rm -r /usr/local/lib/libg2o* /usr/local/include/g2o /usr/local/lib/g2o /usr/local/bin/g2o*
```

paulsammut, 2017-10-13:
Thanks @matlabbe , you're the best!!

closed 2017-10-13 by paulsammut

enwpoa332, 2018-08-08:
To piggy-back @paulsammut [here](https://github.com/introlab/rtabmap_ros/issues/199#issuecomment-336311838), the cli command is
```
cmake ../ -DBUILD_WITH_MARCH_NATIVE=OFF -DG2O_BUILD_EXAMPLES=OFF -DG2O_BUILD_APPS=OFF
```
Then `make` and `sudo make install`, and you are ready to work on compiling `rtabmap`.