introlab/rtabmap_ros #200: roslaunch rtabmap_ros demo_turtlebot_mapping.launch

vlon, 2017-10-13:
I am new learner to ros and rtabmap,and I did as the http://wiki.ros.org/rtabmap_ros/Tutorials/MappingAndNavigationOnTurtlebot told.However,I get the follow error:
[ INFO] [1507863746.475911419]: No matching device found.... waiting for devices. Reason: std::__cxx11::string openni2_wrapper::OpenNI2Driver::resolveDeviceURI(const string&) @ /tmp/binarydeb/ros-kinetic-openni2-camera-0.2.8/src/openni2_driver.cpp @ 642 : Invalid device number 1, there are 0 devices connected.
[ WARN] [1507863746.502727737]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmap subscribed to (approx sync):
   /camera/rgb/image_rect_color,
   /camera/depth_registered/image_raw,
   /camera/rgb/camera_info,
   /scan

How can I fix this? 
Thank you !

vlon, 2017-10-13:
I also tried to build from source,but when I run "make" at the rtabmap step,I got the follow error:

In file included from /usr/local/include/g2o/core/block_solver.h:199:0,
                 from /home/josh/rtabmap/corelib/src/OptimizerG2O.cpp:42:
/usr/local/include/g2o/core/block_solver.hpp:40:1: note: candidate: g2o::BlockSolver<Traits>::BlockSolver(std::unique_ptr<typename Traits::LinearSolverType>) [with Traits = g2o::BlockSolverTraits<-1, -1>; typename Traits::LinearSolverType = g2o::LinearSolver<Eigen::Matrix<double, -1, -1> >]
 BlockSolver<Traits>::BlockSolver(std::unique_ptr<LinearSolverType> line
 ^
/usr/local/include/g2o/core/block_solver.hpp:40:1: note:   no known conversion for argument 1 from ‘SlamLinearEigenSolver* {aka g2o::LinearSolverEigen<Eigen::Matrix<double, -1, -1> >*}’ to ‘std::unique_ptr<g2o::LinearSolver<Eigen::Matrix<double, -1, -1> >, std::default_delete<g2o::LinearSolver<Eigen::Matrix<double, -1, -1> > > >’
/home/josh/rtabmap/corelib/src/OptimizerG2O.cpp:208:50: error: no matching function for call to ‘g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> >::BlockSolver(SlamLinearCholmodSolver*&)’
    blockSolver = new SlamBlockSolver(linearSolver);
                                                  ^
In file included from /usr/local/include/g2o/core/block_solver.h:199:0,
                 from /home/josh/rtabmap/corelib/src/OptimizerG2O.cpp:42:
/usr/local/include/g2o/core/block_solver.hpp:40:1: note: candidate: g2o::BlockSolver<Traits>::BlockSolver(std::unique_ptr<typename Traits::LinearSolverType>) [with Traits = g2o::BlockSolverTraits<-1, -1>; typename Traits::LinearSolverType = g2o::LinearSolver<Eigen::Matrix<double, -1, -1> >]
 BlockSolver<Traits>::BlockSolver(std::unique_ptr<LinearSolverType> line
 ^
/usr/local/include/g2o/core/block_solver.hpp:40:1: note:   no known conversion for argument 1 from ‘SlamLinearCholmodSolver* {aka g2o::LinearSolverCholmod<Eigen::Matrix<double, -1, -1> >*}’ to ‘std::unique_ptr<g2o::LinearSolver<Eigen::Matrix<double, -1, -1> >, std::default_delete<g2o::LinearSolver<Eigen::Matrix<double, -1, -1> > > >’
/home/josh/rtabmap/corelib/src/OptimizerG2O.cpp:217:50: error: no matching function for call to ‘g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> >::BlockSolver(SlamLinearCSparseSolver*&)’
    blockSolver = new SlamBlockSolver(linearSolver);

I thought the new version g2o may incompatible with rtabmap,so downloaded the 2016 version,and I can get the "make" through,But it come out the first error.

matlabbe, 2017-10-13:
For g2o, the latest version is indeed incompatible: https://github.com/introlab/rtabmap_ros/issues/199

For your error: "_Invalid device number 1, there are 0 devices connected._", it means that openni cannot detect a camera. What is your camera? If no data is published, rtabmap will complain "_Did not receive data since 5 seconds!_".

cheers,
Mathieu

vlon, 2017-10-16:
@matlabbe thanks for your reply,I have fixed the problem.