introlab/rtabmap_ros #201: global mapping of rtabmap node by external odometry instead of visual odometry

Jinshingji, 2017-10-22:
hi, I have a bag file which contains stereo images and odometry. I want to get global point cloud from this bag file with rtabmap node instead of using visual odometry. I made republisher to publish them by receiving stereo_images and odometry from it when bag file is played.  And I have stereo_image_proc and rtabmap node to run for generating point cloud from this bag file with launch file. I have got a global point cloud, but the point cloud is messed. it is not recognizable. I do n't know why point cloud is messed. odometry is correct. I wish any one to solve this issue.
thank you.

matlabbe, 2017-10-22:
Do you have a screenshot? Maybe there is a missing optical rotation of the camera frame.

Jinshingji, 2017-10-23:
[image]
hi, this is point cloud messed.
how do I set optical rotation of the camera frame?

Jinshingji, 2017-10-23:
I have been trying to do it by static_trnaform_publisher. but the appearance of point cloud is not changed even when running static_transform_publisher. what should I do to rotate camera frame? rtabmap can receive tf_msgs when running static_transform_publisher?

Jinshingji, 2017-10-23:
rtabmap node can rotate camera frame?

matlabbe, 2017-10-23:
See #202