introlab/rtabmap_ros #203: Rtabmap Crashes after couple of minutes

mertmzk, 2017-10-25:
Hi all,

I have successfully run RTABMAP map on ROS with stereo camera and can get  results for a couple of minutes and then it crashes with this error:

> [rtabmap/rtabmap-2] process has died [pid 26721, exit code -11, cmd /opt/ros/kinetic/lib/rtabmap_ros/rtabmap --delete_db_on_start rgb/image:=/camera/rgb/image_rect_color depth/image:=/camera/depth_registered/image_raw rgb/camera_info:=/camera/rgb/camera_info left/image_rect:=/stereo_camera/left/image_rect_color right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info right/camera_info:=/stereo_camera/right/camera_info scan:=/scan scan_cloud:=/scan_cloud __name:=rtabmap __log:=/root/.ros/log/d5399fc8-b981-11e7-8cec-ec1127743a1c/rtabmap-rtabmap-2.log].
> log file: /root/.ros/log/d5399fc8-b981-11e7-8cec-ec1127743a1c/rtabmap-rtabmap-2*.log

I could not found the log file in specified path.  Odometry is keep working but map is not updating (octomap or projmap)

> [ INFO] [1508943891.653421853]: Odom: quality=215, std dev=0.011753m|0.022265rad, update time=0.406519s
> [ INFO] [1508943892.043820834]: Odom: quality=203, std dev=0.004086m|0.007182rad, update time=0.382750s

Here more details about my implementation. It is compiled from source as my SOM is Debian based.

Has anyone got any suggestion at all? I could not found any similar problem.

Thanks.

matlabbe, 2017-10-25:
Hi,  a log like that is difficult to debug. Is there a FATAL log before rtabmap died? Can you reproduce the error easily? 

You may give a try with rtabmap_ros built from source:https://github.com/introlab/rtabmap_ros#build-from-source

QinZiwen, 2018-03-26:
@matlabbe I have same problems, my log is:
```
$ roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start" rgb_topic:=/kinect2/qhd/image_color_rect depth_topic:=/kinect2/qhd/image_depth_rect camera_info_topic:=/kinect2/qhd/camera_info
... logging to /home/turtlebot/.ros/log/834461a8-30c8-11e8-8c8d-c8d3ffd835e0/roslaunch-turtlebot-OMEN-by-HP-Laptop-25056.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:50818/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /rtabmap/rgbd_odometry/approx_sync: True
 * /rtabmap/rgbd_odometry/config_path: 
 * /rtabmap/rgbd_odometry/frame_id: camera_link
 * /rtabmap/rgbd_odometry/ground_truth_base_frame_id: 
 * /rtabmap/rgbd_odometry/ground_truth_frame_id: 
 * /rtabmap/rgbd_odometry/guess_frame_id: 
 * /rtabmap/rgbd_odometry/guess_min_rotation: 0.0
 * /rtabmap/rgbd_odometry/guess_min_translation: 0.0
 * /rtabmap/rgbd_odometry/odom_frame_id: odom
 * /rtabmap/rgbd_odometry/queue_size: 10
 * /rtabmap/rgbd_odometry/subscribe_rgbd: False
 * /rtabmap/rgbd_odometry/wait_for_transform_duration: 0.2
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/approx_sync: True
 * /rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/database_path: ~/.ros/rtabmap.db
 * /rtabmap/rtabmap/frame_id: camera_link
 * /rtabmap/rtabmap/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/ground_truth_frame_id: 
 * /rtabmap/rtabmap/map_frame_id: map
 * /rtabmap/rtabmap/odom_frame_id: 
 * /rtabmap/rtabmap/odom_sensor_sync: False
 * /rtabmap/rtabmap/odom_tf_angular_variance: 1.0
 * /rtabmap/rtabmap/odom_tf_linear_variance: 1.0
 * /rtabmap/rtabmap/queue_size: 10
 * /rtabmap/rtabmap/scan_normal_k: 0
 * /rtabmap/rtabmap/subscribe_depth: True
 * /rtabmap/rtabmap/subscribe_odom_info: True
 * /rtabmap/rtabmap/subscribe_rgbd: False
 * /rtabmap/rtabmap/subscribe_scan: False
 * /rtabmap/rtabmap/subscribe_scan_cloud: False
 * /rtabmap/rtabmap/subscribe_stereo: False
 * /rtabmap/rtabmap/subscribe_user_data: False
 * /rtabmap/rtabmap/wait_for_transform_duration: 0.2
 * /rtabmap/rtabmapviz/approx_sync: True
 * /rtabmap/rtabmapviz/frame_id: camera_link
 * /rtabmap/rtabmapviz/odom_frame_id: 
 * /rtabmap/rtabmapviz/queue_size: 10
 * /rtabmap/rtabmapviz/subscribe_depth: True
 * /rtabmap/rtabmapviz/subscribe_odom_info: True
 * /rtabmap/rtabmapviz/subscribe_rgbd: False
 * /rtabmap/rtabmapviz/subscribe_scan: False
 * /rtabmap/rtabmapviz/subscribe_scan_cloud: False
 * /rtabmap/rtabmapviz/subscribe_stereo: False
 * /rtabmap/rtabmapviz/wait_for_transform_duration: 0.2

NODES
  /rtabmap/
    rgbd_odometry (rtabmap_ros/rgbd_odometry)
    rtabmap (rtabmap_ros/rtabmap)
    rtabmapviz (rtabmap_ros/rtabmapviz)

ROS_MASTER_URI=http://localhost:11311
[... 40 lines cut ...]
[ INFO] [1522066743.197483650]: RGBDOdometry: queue_size     = 10
[ INFO] [1522066743.197498431]: RGBDOdometry: subscribe_rgbd = false
[ INFO] [1522066743.197513820]: RGBDOdometry: rgbd_cameras   = 1
[ INFO] [1522066743.228072292]: 
/rtabmap/rgbd_odometry subscribed to (approx sync):
   /kinect2/qhd/image_color_rect,
   /kinect2/qhd/image_depth_rect,
   /kinect2/qhd/camera_info
[rtabmap/rtabmapviz-3] process has died [pid 25076, exit code -11, cmd /home/turtlebot/catkin_ws/devel/lib/rtabmap_ros/rtabmapviz -d ~/.ros/rtabmap_gui.ini rgb/image:=/kinect2/qhd/image_color_rect depth/image:=/kinect2/qhd/image_depth_rect rgb/camera_info:=/kinect2/qhd/camera_info rgbd_image:=rgbd_image left/image_rect:=/stereo_camera/left/image_rect_color right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info right/camera_info:=/stereo_camera/right/camera_info scan:=/scan scan_cloud:=/scan_cloud odom:=odom __name:=rtabmapviz __log:=/home/turtlebot/.ros/log/834461a8-30c8-11e8-8c8d-c8d3ffd835e0/rtabmap-rtabmapviz-3.log].
log file: /home/turtlebot/.ros/log/834461a8-30c8-11e8-8c8d-c8d3ffd835e0/rtabmap-rtabmapviz-3*.log
[ INFO] [1522066743.449113026]: Setting RTAB-Map parameter "Vis/InlierDistance"="0.02"
[ INFO] [1522066743.455802349]: Setting RTAB-Map parameter "Vis/MinInliers"="20"
[ INFO] [1522066743.757116067]: RTAB-Map detection rate = 1.000000 Hz
[ INFO] [1522066743.757297834]: rtabmap: Deleted database "/home/turtlebot/.ros/rtabmap.db" (--delete_db_on_start or -d are set).
[ INFO] [1522066743.757341936]: rtabmap: Using database from "/home/turtlebot/.ros/rtabmap.db" (0 MB).
[ INFO] [1522066745.035418606]: Odom: quality=0, std dev=99.995000m|99.995000rad, update time=0.023842s
[ INFO] [1522066745.143841476]: Odom: quality=779, std dev=0.003386m|0.000032rad, update time=0.068155s
[ INFO] [1522066745.191539573]: Odom: quality=807, std dev=0.002906m|0.000032rad, update time=0.045307s
[ INFO] [1522066745.252528133]: Odom: quality=798, std dev=0.003221m|0.000032rad, update time=0.045176s
[ INFO] [1522066745.322212699]: Odom: quality=788, std dev=0.003064m|0.000032rad, update time=0.045287s
[ INFO] [1522066745.388249199]: Odom: quality=788, std dev=0.003364m|0.000032rad, update time=0.046002s
[ INFO] [1522066745.456348273]: Odom: quality=786, std dev=0.003001m|0.000032rad, update time=0.045956s
[ INFO] [1522066745.519056630]: Odom: quality=787, std dev=0.003127m|0.000032rad, update time=0.045574s
[ INFO] [1522066745.587373162]: Odom: quality=800, std dev=0.002815m|0.000032rad, update time=0.045293s
```

matlabbe, 2018-03-26:
I can see this is on Indigo, but is it rtabmap binaries too? or latest from source? You could try to launch the rtabmapviz with gdb [launch-prefix](http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB), and do "bt" (backtrace) when it crashes.

cheers