introlab/rtabmap_ros #207: ZR300 -rtabmap  Did not receive data since 5 seconds!

ashikmh, 2017-11-18:
HI
I am new to rtabmap and realsense and trying to interface realsense zr300 with the rtabmap. Sometimes the camera senses images and I get the point cloud stream, But the 3d map remains red. The other times  I have tried the two approaches mentioned in earlier issues and the tutorial, I get the error as -  

`[ WARN] [1510993333.921116290]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmap subscribed to (approx sync):
   /rtabmap/odom,
   /camera/rgb/image_rect_color,
   /camera/depth_registered/image_raw,
   /camera/rgb/camera_info`

It would be nice if I could get some suggestions on the issue,
Thanks

matlabbe, 2017-11-18:
For ZR300, make sure you are using the legacy branch of realsense's github: https://github.com/IntelRealSense/librealsense/tree/legacy

For realsense cameras, to get a more dense depth image, the second approach of the [tutorial](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping) is better:

```bash
#ZR300
$ roslaunch realsense_camera zr300_nodelet_rgdb.launch

// To get registration without empty lines
$ rosrun rtabmap_ros pointcloud_to_depthimage cloud:=/camera/depth/points camera_info:=/camera/rgb/camera_info image_raw:=/camera/depth/points/image_raw image:=/camera/depth/points/image _approx:=false _fill_holes_size:=2

// Using depth registered from `rtabmap_ros/pointcloud_to_depthimage`
 $ roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start" depth_topic:=/camera/depth/points/image_raw
```
A more dense image will help to get more valid visual features to track. Then for red screens, see "Odometry Lost" section on this [page](https://github.com/introlab/rtabmap/wiki/Kinect-mapping#lostodometry). If you have still problems, post screenshots of the RGB image and the depth image.

cheers,
Mathieu

ashikmh, 2017-11-20:
Hi,
   I am now using the legacy branch as you said. In the tutorial you provided,When I try to run the commnad
`rosrun rtabmap_ros pointcloud_to_depthimage cloud:=/camera/depth/points camera_info:=/camera/rgb/camera_info image_raw:=/camera/depth/points/image_raw image:=/camera/depth/points/image _approx:=false _fill_holes_size:=2`

I get the error:
`[ERROR] [1511164878.678113962]: Failed to load nodelet [/pointcloud_to_depthimage] of type [rtabmap_ros/pointcloud_to_depthimage] even after refreshing the cache: MultiLibraryClassLoader: Could not create object of class type rtabmap_ros::PointCloudToDepthImage as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ERROR] [1511164878.678180641]: The error before refreshing the cache was: MultiLibraryClassLoader: Could not create object of class type rtabmap_ros::PointCloudToDepthImage as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()`

Thanks

ashikmh, 2017-11-20:
Hi,
    The problems are solved. I did not add the new workspace to the library path `$LD_LIBRARY_PATH`. Now it is working fine. Thanks for the help. :)

closed 2017-11-20 by ashikmh