introlab/rtabmap_ros #208: Does "rtabmap" node publish obstacle/ground point clouds?

seiya0412, 2017-11-24:
I found /rtabmap/cloud_obstacles and /rtabmap/cloud_ground topics were published from rtabmap node when I launched stereo_mapping.launch with RViz.
However, in the [ROS Wiki's rtabmap node description](http://wiki.ros.org/rtabmap_ros#rtabmap), these topics are not listed in the "4.1.3 Published Topics".

/rtabmap/obstacle_detection nodelet was not launched and not be found on `$rosnode list`.

I have two questions here.

- Does "rtabmap" node actually publish obstacle/ground point clouds?
- I didn't see any parameter setting for these topics. Can I modify parameters for them like /rtabmap/obstacle_detection nodelet?

[image]

Regards,

matlabbe, 2017-11-24:
Hi,

The documentation is outdated about these topics. For your first question, yes, if `RGBD/CreateOccupancyGrid` is true (default true in ros). You will get ground and obstacle clouds of the global map.

For your second question, the same algorithm is used in `rtabmap_ros/obstacles_detection` nodelet and `rtabmap` node (version >=Kinetic), and same parameters can be used for convenience:
```bash
$ rosrun rtabmap_ros rtabmap --params | grep Grid/
```
To reproduce same default behavior than the nodelet, set `Grid/3D` and `Grid/FromDepth` to true, to get a 3D segmentation from images (not laser scan).

cheers,
Mathieu

seiya0412, 2017-11-24:
It's very clear now.
Thank you very much for your help!

Seiya

closed 2017-11-24 by seiya0412