introlab/rtabmap_ros #209: the 2D /rtabmap/octomap_grid map disappear

harbouryi, 2017-11-30:
the stereo mapping , when the program has found loop closure ,the 2D map whose topic is  /rtabmap/octomap_grid in rviz has disappeared

matlabbe, 2017-11-30:
Make sure parameter "`RGBD/CreateOccupancyGrid`" is not set to false (default true for rtabmap_ros). I updated the outdoor stereo demo without setting this parameter to false so that ocotmap is not cleared on loop closures.

harbouryi, 2017-12-06:
I'm sorry I didn't respond to you in time， i 
 used this "https://github.com/introlab/rtabmap_ros/blob/master/launch/stereo_mapping.launch " to do stereo mapping ,but not with  Bumblebee2 or zed camera . The map disappeared when found loop closure.   should i set the parameter "RGBD/CreateOccupancyGrid " to false ?

matlabbe, 2017-12-06:
I cannot reproduce the problem with this small [example](https://github.com/introlab/rtabmap/wiki/Stereo-mapping#process-a-directory-of-stereo-images-in-ros):
```bash
$ cd Downloads/stereo_20Hz
$ roslaunch stereo_20Hz.launch rviz:-true rtabmapviz:=false align_with_ground:=true
```
(then choosing /rtabmap/octomap_grid in rviz). There were some loop closures and the grid was not cleared:
[image]

However, I tried live with a hand-held stereo camera and I reproduce the following.
Before loop:
[image]
After loop:
[image]

Some gray cells disappeared after loop closure detection, but obstacles (black cells) are still there. It is because empty cells created by ray tracing are not replaced back after the correction (when `GridGlobal/FullUpdate` is false), only the ground cells are corrected (which explains why there is still some gray cells in the map). Is it the problem you are seeing? If the whole map is cleared, which version of rtabmap are you using?

Note I just fixed bug where `GridGlobal/FullUpdate` was always false for OctoMap (making gray cells disappearing like above). Now the results should look like this:
Before loop:
[image]
After loop:
[image]

cheers,
Mathieu

harbouryi, 2017-12-08:
Thank you!
yes , that is the problem what i am seeing . i am using the newest version,it seems be 2.8.5.  I will try it with what you fixed. Thanks.

 i hava another question, why this rtabmap gui  does not have enough option 
this is my gui:
[image]
this is others gui:
[image]

matlabbe, 2017-12-08:
This is an old gui and it is the standalone version of rtabmapviz. The same options are in the menu Detection though. For the Play/Stop buttons, they are not in ros version of the ui, as the camera is controlled by an outside node.

harbouryi, 2017-12-11:
i have solved all my problem. thank you~~