introlab/rtabmap_ros #211: Losing part of the map in localization mode

smail1229, 2017-12-11:
Hello,
I have a problem about localization mode.
First, I used rtabmap to generate a grid map like this:
[image]
Then, I chose the localization mode, and the map it published look like this:
[image]

We can see that part of grid map on the left and bottom are missing.
Actually, I tried the localization mode several times and sometimes the gird map loses part of itself randomly.

I want to know how to get the complete map

Thank you

matlabbe, 2017-12-11:
See this answer: https://github.com/introlab/rtabmap_ros/issues/210#issuecomment-350729981

With these parameters, you should get always the global map.

smail1229, 2017-12-19:
Thank you! The problem is solved

closed 2017-12-19 by smail1229