introlab/rtabmap_ros #213: Feature request: Modify grid_map

paulsammut, 2017-12-13:
Hello! I am creating good maps with rtabmap, and using the grid_map output for planning use. My question is: Can I modify the grid_map published by rtab? My currently workflow is I first do a mapping run with settings that are good for mapping, then I save the database and run the refinements and more loop closures in rtabmap-database viewer. I then use the database in just localization mode for normal operation of my robot. 

It would be nice to add in some things into the 2d occupancy map that rtabmap serves up. For example, i'd like to create "lanes" for my robot to travel in, to make it easier for path planning with multiple waypoints. Like this:
[image]

Is there an easy way to do this? I was also thinking of maybe creating a separate map that has my stuff on it, and serve it with [ros' map server](http://wiki.ros.org/map_server) and publish a static tf link to the rtabmap map frame.

Thanks for all your work!

matlabbe, 2017-12-14:
Hi, 

The short answer: use the map_server workaround you described as we cannot modify the grid map in the database. 

The long answer: RTAB-Map doesn't keep the global occupancy grid map saved like this in the database, it keeps all local occupancy grid maps created for each node in the map, then reconstruct the global map on runtime. It is because on loop closure detection, the graph may change so that global map must be reconstructed. In localization mode, the map would be static as there are no map corrections, so the map could be saved as is in the database. However, it is not always the case because of the memory management (when used) that could transfer/retrieve some parts of the map even in localization mode, so that graph optimization should be done again. 

Well, for the special case of localization without memory management, we could think about saving the global map in database for convenience, instead of manually using the map_saver/map_server approach, to keep all data at the same place. That feature would require at least an export/import approach to get the occupancy grid map in an image editor, then re-save it back to database. The [Admin](https://github.com/introlab/rtabmap/blob/master/corelib/src/resources/DatabaseSchema.sql.in) table could be used to save the last optimized occupancy grid with corresponding optimized graph (similar to what is done for the optimized mesh saved by RTAB-Map Tango). 

If we address the lane preference, how would you edit the map to tell the planner that some cells should be used over others? Currently the cells are 0=empty, -1=unknown and 100=obstacle. While not used by rtabmap, values between 0 and 100 are probabilities of an obstacle, probably using values 101 to 128 and -2 to -128 may encode the lane preferences. Another approach to set lane preference, without the grid map, is to encode the preferences in the nodes of the graph (using [label](https://github.com/introlab/rtabmap/blob/71d9816f4bb6e47bb2a90fb3e82cf911f30a2b12/corelib/src/resources/DatabaseSchema.sql.in#L24) field for example). I already used this on a robot where I had to set zones where the robot could move faster or slower. The topic /rtabmap/mapGraph contains poses of the map with their label, so it is possible to know indirectly the position of the lanes (paths that you drove the robot during the mapping phase).

cheers,
Mathieu

paulsammut, 2017-12-15:
Thanks, I dug through the db hoping to find the grid, but couldn't as per your explanation :). 
I intend to use a static layer 2d costmap, and I will be able to set the values of the lanes accordingly. For instance, "suggested" lanes can have a lower value cost value, allowing the robot to enter or exit the area if need be, and stuff like bike lanes where I want my robot to stay in can have a higher costs. 

I think trying to modify the map rtabmap serves is not the right way to go as that would be stretching its intended purpose. I'll see how well it would be to integrate the map_saver/map_server approach. 

It seems there there is a philosophical approach to make rtabmap more of a "live mapper" type system, that maps and handles loop closures while normal robot operation, instead of a dedicate map maker and dedicated map localizer/server. I don't see how this works because if you intend to map while operating, you cannot operate in an environment with dynamic obstacles.

matlabbe, 2017-12-15:
Current research using rtabmap is about long-term continuous SLAM(see this [paper](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/8/87/LabbeAURO2017.pdf) for example): always updating the map while operating, so that the map can adapt to changes in environment and extend during exploration. Indeed, for many practical applications, it is simpler to go through a mapping phase, then switch to a localization phase. Beeing able to save the global grid map in the database as explained in my previous post would help then help to do this kind of application more easily and efficiently. 

Cheers,
Mathieu

paulsammut, 2017-12-15:
Cool, if you did that, I'd be able to go into the grid map and modify it to take out false obstacle cells where people walked in the front of the robot :)

paulsammut, 2018-03-01:
I've been working on this more and I have run into an issue with the map that rtab publishes changing over time. 

My approach is as follows: 
* Create a map of the environment with rtabmap
* Download the map using map_server
* Edit the map .pgm file, clean up the ghost obstacles and add things like lanes with variable cost
* Run move_base with the global map having one static 2d costmap layer that is seeded by map_server, that serves my edited map
* Run rtabmap in localization mode so it can publish the /map to /odom transform

This works but only for a few seconds. After that, rtabmap seems to serve a map that is slightly shifted or rotated. It seems like rtabmap isn't just sending a new transform, but modifying the map itself and shifting it to fit the transform. Any advice? Is there maybe a setting I could change that would leave rtab's served map unchanged? 

Here is a shot of me running rtabmap in localization mode, while serving up my custom map. I am also displaying the /grid_map publishes by rtabmap, which is localized properly, but shifts causing a discrepancy with the custom map i and using in move_base. 

[image]

Thanks!
-Paul

matlabbe, 2018-03-01:
If /map -> /odom stays at Identity transform and the map rotates/translates on localization, it looks like that `RGBD/OptimizeFromGraphEnd` parameter is set to true, is it?

paulsammut, 2018-03-03:
It is not unfortunately. Here's the parameter dump:

`<param name="Mem/IncrementalMemory" type="string"   value="false"/>
<param name="Rtabmap/DetectionRate" type="string"   value="5"/>
<param name="Grid/RangeMin"         type="double"   value="0.1"/>
<param name="Grid/RangeMax"         type="double"   value="0.2"/>
<param name="Grid/CellSize"             type="double"   value="0.05"/>
<param name="GridGlobal/MinSize"        type="double"   value="40"/>
<param name="Grid/MaxGroundHeight"      type="string"   value="0.50"/>       
<param name="Grid/MaxGroundAngle"       type="double"   value="80"/>
<param name="Grid/MaxObstacleHeight"    type="double"   value="2"/>
<param name="Grid/MinClusterSize"       type="int"      value="20"/>

<param name="Grid/FootprintLength"      type="double"   value="0.7"/>
<param name="Grid/FootprintWidth"       type="double"   value="0.9"/>
<param name="Grid/FootprintHeight"      type="double"   value="2"/>

<param name="Grid/NormalsSegmentation"  type="bool"     value="false"/>

<param name="Rtabmap/TimeThr"                   type="string"   value="1200"/>
<param name="Kp/MaxFeatures"                    type="string"   value="400"/>
<param name="Kp/RoiRatios"                      type="string"   value="0.03 0.03 0.04 0.04"/>
<param name="Kp/DetectorStrategy"               type="string"   value="0"/>

<param name="Kp/NNStrategy"                     type="string"   value="1"/>

<param name="SURF/HessianThreshold"             type="string"   value="1000"/>

<param name="RGBD/OptimizeMaxError"             type="double"   value="$(arg maxError)"  />

<param name="RGBD/OptimizeFromGraphEnd"         type="string"   value="false"/>
<param name="RGBD/LoopClosureReextractFeatures" type="string"   value="true"/>
<param name="Grid/MaxGroundHeight"              type="double"   value="0.18"/>

<param name="Vis/MinInliers"                    type="string"   value="10"/>
<param name="Vis/InlierDistance"                type="string"   value="0.1"/>
<param name="Viz/EstimationType"                type="int"      value="1"/>
<param name="Vis/MaxFeatures"                   type="int"      value="1000"/>

<param name="Reg/Force3DoF"                                     value="true" />
<param name="Optimizer/Slam2D"                                  value="true" />

<param name="frame_id"                  type="string"   value="base_link"               />
<param name="odom_frame_id"             type="string"   value="odom"    />
<param name="approx_sync"               type="bool"     value="true"                    />
<param name="queue_size"                type="int"      value="5"                       />
<param name="publish_tf"                type="bool"     value="$(arg publish_tf)"       />
<param name="publish_null_when_lost"    type="bool"     value="true"/>
<param name="guess_frame_id"            type="boolean"  value="true"                  />

<param name="Odom/Strategy"         type="string"   value="0"/> 
<param name="Odom/EstimationType"   type="string"   value="1"/> 
<param name="Odom/MinInliers"       type="string"   value="10"/>
<param name="Odom/RoiRatios"        type="string"   value="0.03 0.03 0.04 0.04"/>
<param name="Odom/MaxDepth"         type="string"   value="10"/>
<param name="OdomBow/NNDR"          type="string"   value="0.8"/>
<param name="Odom/MaxFeatures"      type="string"   value="1000"/>
<param name="Odom/FillInfoData"     type="string"   value="true"/> 
<param name="GFTT/MinDistance"      type="string"   value="10"/>
<param name="GFTT/QualityLevel"     type="string"   value="0.00001"/> 
<param name="Vis/InlierDistance"    type="string"   value="0.1"                 />
<param name="Vis/MinInliers"        type="string"   value="10"                  />
<param name="Vis/RoiRatios"         type="string"   value="0.03 0.03 0.04 0.04" />
<param name="Vis/MaxDepth"          type="string"   value="10"                  />
`

matlabbe, 2018-04-04:
Using commits above, the grid map is automatically saved to database on exit. We can use DatabaseViewer -> File- > Export saved 2D map... to export the map in pgm image file. That file can be edited using a image editor to remove/add obstacles. Then with File -> Import 2D map, the map can be re-imported and overwriting the one saved in the database. When relaunching the robot in **localization** mode, the edited map will be used.

paulsammut, 2018-04-19:
Hey @matlabbe thanks for the update! I just pulled the latest master commits for both rtabmap and rtabmap_ros and gave this a go. I did it using my sim gazebo setup and found that rtabmap was freezing after about a couple of minutes of mapping. By freezing i mean that new nodes were being added and rtabmapviz shoes no new information. I reverted back to the old version I was using and problem went away. Didn't have enough time to dig into it, just wanted to let you know. 

-Paul

matlabbe, 2018-04-26:
Thx to let me know, I am however not able to reproduce the problem when wandering a couple of minutes with [turtlebot gazebo demo](http://wiki.ros.org/rtabmap_ros/Tutorials/MappingAndNavigationOnTurtlebot#Simulation_.28Gazebo.29). A freezing without any errors may be related with some input topics not published anymore or that they are badly synchronized (some topics are lagging for example). If it is the GUI which is not responsive, maybe a GPU related problem.

Well, if you retry the latest version, get screenshot and the terminal log status. 

cheers,
Mathieu

tugbakara, 2022-10-19:
> Using commits above, the grid map is automatically saved to database on exit. We can use DatabaseViewer -> File- > Export saved 2D map... to export the map in pgm image file. That file can be edited [...]

Is there any other solution for ros side  only?

matlabbe, 2022-10-19:
The database should be closed before editing the map, so not sure how we could do that with ROS. Note that the grid map can be edited directly in rtabmap-databaseViewer without having to export/import back the map. See http://official-rtab-map-forum.206.s1.nabble.com/post-processing-manually-add-obstacles-to-map-tp6985p6993.html

tugbakara, 2022-10-19:
Thanks @matlabbe , we solved it from databaseviewer but we are facing one issue that is except octomap_grid topic others are recreates the mao even if we open in localization mode that's why we re trying to import modified octomap_grid map but there is not anything about it , databaseviewer only takes grid_map but in this map our obstacles are very rare compared to octomap_grid. Us there any solution to import changed octomap_grid map, or to stop mapping while opening grid_mp with localization?

matlabbe, 2022-10-19:
We would need to add an option in DatabaseViewer to select which map to edit and save. As a workaround, you could use `rosrun map_server map_saver map=/rtabmap/octomap_grid` with the `octomap_grid` topic, this will save a PGM of the octomap grid. You can then try to edit/import that PGM using DatabaseViewer as the new optimized grid map.

tugbakara, 2022-10-20:
I will try it but do you have an idea about, when I open the map for localization with rtabmap/grid_map topic or others (except octomap_grid) it starts to create map again on existing map, how can I stop it ?

matlabbe, 2022-10-20:
It feels you are not really in localization mode. `Mem/IncrementalMemory` parameter should be false, otherwise rtabmap will start a new session and re-create the grid. Can you share the database?

tugbakara, 2022-10-21:
I already set the param false, I will send the db when I am available.

tugbakara, 2022-10-22:
> You can then try to edit/import that PGM using DatabaseViewer as the new optimized grid map.

It didin't work.

matlabbe, 2022-10-23:
for some reason, the database doesn't have `RGBD/CreateOccupancyGrid` set to true. Fix it like this:
```
rtabmap-reprocess --RGBD/CreateOccupancyGrid true ofisstatic.db output.db
```

Then doing:
```
roscore
rosrun rtabmap_ros rtabmap _database_path:=~/Downloads/output.db
rosrun map_server map_saver map:=/octomap_grid
rosservice call /publish_map 0 1 0
```
we can get the octomap grid pgm file. However, unfortunately, I was wrong, we cannot import it back in databaseViewer because of this error:
[image]

I updated databaseViewer to add an option to regenerate optimized map from OctoMap. Here the new steps (from `output.db` above):
```
rtabmap-databaseViewer ~/Downloads/output.db
```
 * View->"Graph view"
 * File->"Regenerate optimized 2d map..."
 * At this point, the grid created from OctoMap will be saved in the database. You can preview it with File->"View optimized 2d map..."

Then on ROS, to verify that the same map is used under `grid_map` topic:
```
roscore
rosrun rtabmap_ros rtabmap _database_path:=~/Downloads/output.db
rviz
rosservice call /publish_map 0 1 0
```
[image]

tugbakara, 2022-10-24:
> rtabmap-reprocess --RGBD/CreateOccupancyGrid true ofisstatic.db output.db

when I run this command I got some erros like:
`Custom parameters:
  RGBD/CreateOccupancyGrid	= true
Set working directory to ".".
Target database version: "0.20.21" (set explicitly --Db/TargetVersion "" to output with latest version.
Reprocessing data of "ofisstatic.db"...
High variance detected, triggering a new map...
Processed 1/161 nodes [id=1 map=0 opt_graph=1]... 193ms
Processed 2/161 nodes [id=2 map=0 opt_graph=2]... 71ms
Processed 3/161 nodes [id=3 map=0 opt_graph=3]... 49ms
Processed 4/161 nodes [id=4 map=0 opt_graph=4]... 48ms
Processed 5/161 nodes [id=5 map=0 opt_graph=5]... 51ms
Processed 6/161 nodes [id=6 map=0 opt_graph=6]... 46ms
Processed 7/161 nodes [id=7 map=0 opt_graph=7]... 48ms
Processed 8/161 nodes [id=8 map=0 opt_graph=8]... 50ms
Processed 9/161 nodes [id=9 map=0 opt_graph=9]... 48ms
Processed 10/161 nodes [id=10 map=0 opt_graph=10]... 46ms
[ERROR] (2022-10-24 11:07:06.961) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 11/161 nodes [id=11 map=0 opt_graph=11]... 99ms Prox on 1 [0]
[ERROR] (2022-10-24 11:07:07.051) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 12/161 nodes [id=12 map=0 opt_graph=12]... 83ms Prox on 2 [0]
[ERROR] (2022-10-24 11:07:07.148) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 13/161 nodes [id=13 map=0 opt_graph=13]... 81ms Prox on 3 [0]
[ERROR] (2022-10-24 11:07:07.247) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 14/161 nodes [id=14 map=0 opt_graph=14]... 92ms Prox on 2 [0]
[ERROR] (2022-10-24 11:07:07.349) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 15/161 nodes [id=15 map=0 opt_graph=15]... 84ms Prox on 1 [0]
[ERROR] (2022-10-24 11:07:07.450) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 16/161 nodes [id=16 map=0 opt_graph=16]... 86ms Prox on 1 [0]
[ERROR] (2022-10-24 11:07:07.553) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 17/161 nodes [id=17 map=0 opt_graph=17]... 87ms Prox on 1 [0]
[ERROR] (2022-10-24 11:07:07.656) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 18/161 nodes [id=18 map=0 opt_graph=18]... 80ms Prox on 1 [0]
[ERROR] (2022-10-24 11:07:07.748) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 19/161 nodes [id=19 map=0 opt_graph=19]... 84ms Prox on 1 [0]
[ERROR] (2022-10-24 11:07:07.849) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 20/161 nodes [id=20 map=0 opt_graph=20]... 86ms Prox on 1 [0]
[ERROR] (2022-10-24 11:07:07.950) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 21/161 nodes [id=21 map=0 opt_graph=21]... 74ms Prox on 1 [0]
[ERROR] (2022-10-24 11:07:08.038) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 22/161 nodes [id=22 map=0 opt_graph=22]... 87ms Prox on 1 [0]
[ERROR] (2022-10-24 11:07:08.137) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 23/161 nodes [id=23 map=0 opt_graph=23]... 89ms Prox on 3 [0]
[ERROR] (2022-10-24 11:07:08.240) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 24/161 nodes [id=24 map=0 opt_graph=24]... 73ms Prox on 1 [0]
[ERROR] (2022-10-24 11:07:08.338) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 25/161 nodes [id=25 map=0 opt_graph=25]... 99ms Prox on 2 [0]
[ERROR] (2022-10-24 11:07:08.442) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 26/161 nodes [id=26 map=0 opt_graph=26]... 75ms Prox on 3 [0]
[ERROR] (2022-10-24 11:07:08.536) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 27/161 nodes [id=27 map=0 opt_graph=27]... 93ms Prox on 3 [0]
[ERROR] (2022-10-24 11:07:08.639) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 28/161 nodes [id=28 map=0 opt_graph=28]... 89ms Prox on 3 [0]
[ERROR] (2022-10-24 11:07:08.743) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 29/161 nodes [id=29 map=0 opt_graph=29]... 78ms Prox on 3 [0]
[ERROR] (2022-10-24 11:07:08.839) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 30/161 nodes [id=30 map=0 opt_graph=30]... 93ms Prox on 3 [0]
[ERROR] (2022-10-24 11:07:08.940) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 31/161 nodes [id=31 map=0 opt_graph=31]... 78ms Prox on 3 [0]
[ERROR] (2022-10-24 11:07:09.035) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 32/161 nodes [id=32 map=0 opt_graph=32]... 87ms Prox on 22 [0]
[ERROR] (2022-10-24 11:07:09.136) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 33/161 nodes [id=33 map=0 opt_graph=33]... 75ms Prox on 22 [0]
[ERROR] (2022-10-24 11:07:09.224) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 34/161 nodes [id=34 map=0 opt_graph=34]... 90ms Prox on 22 [0]
[... cut ...]
[ERROR] (2022-10-24 11:07:47.755) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
[ERROR] (2022-10-24 11:07:47.814) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 161/161 nodes [id=161 map=0 opt_graph=161]... 120ms Prox on 126 [0]
Total loop closures = 111 (Loop=0, Prox=111, In Motion=111/161)
Closing database "output.db"...
Closing database "output.db"... done!
`

If it will any other things happen, I will write them here.

matlabbe, 2022-11-26:
The computer you are reprocessing the database doesn't have rtabmap with g2o dependency. TORO optimizer is used as backup but it doesn't support BA optimization.

tugbakara, 2022-11-28:
Thanks @matlabbe !

dimitryn, 2022-12-07:
Hi @matlabbe , can you please summarize steps to modify grid_map. thank you .

matlabbe, 2022-12-11:
You won't see the changes in Graph View, you can make sure your changes are there by doing File->Edit optimized 2d map..., then you will see the actual optimized/modified occupancy grid that rtabmap will use when relaunching in localization mode. See also http://official-rtab-map-forum.206.s1.nabble.com/post-processing-manually-add-obstacles-to-map-tp6985p6993.html: 
[image]

dimitryn, 2022-12-12:
Hi @matlabbe  thanks for reply
1.My steps. Open database -> Export optimized 2d map -> edit image ->Import optimized 2d map -> Close optimized 2d map.
2.No dbDriver_->save2DMap called on closed (I set breakpoint on close). When I reopen databse and call ->Edit optimzed 2d map , it show 2d image without obstacle !
3.Should regenerate 2d map  be called on some step ?
thanks

matlabbe, 2022-12-13:
Can you try editing the map with the embedded editor (just for a test)? Maybe there is a bug in the ui that makes it not detecting the map has been updated when re-importing the map.

matlabbe, 2022-12-13:
The Export/Import workflow seems working. Here the output after closing and reloading rtabmap-databaseViewer (I put a "T" in the middle with GIMP editor):
[image]

closed 2022-12-13 by matlabbe