introlab/rtabmap_ros #214: I found it hard to localized

harbouryi, 2017-12-19:
Hello,
I  ran the tutorial--"RGB-D Hand-Held Mapping" , when it is on localization mode ,  i found it hard to localized ,after  cost three to seven seconds ，it can successfully localized . Isthis normal ? And can i decrease the time what it cost?

matlabbe, 2017-12-19:
You can try to map the environment in different directions/camera orientations so that localization is more point of view invariant. Rtabmap uses visual clues to localize, and depending on the feature detector used, some require to have almost the same point of view so that features extracted match with the ones in the map. Most robust features for localization are those which are scale/rotation/shear invariant like SIFT and SURF (Kp/DetectorStrategy=0 or 1). See also "anchors" description on this [page](https://github.com/introlab/rtabmap/wiki/Multi-Session-Mapping-with-RTAB-Map-Tango):

> Anchors: to be able to merge automatically maps, loop closures should be detected between them. For example, try to start the new scan with the same point of view in front of a location mapped in previous scan. Use a location that have a lot of discriminative visual features under 4 meters, which we will refer as an "anchor location". For example, paintings or billboards on the walls are good anchors. A least one anchor should be revisited in new sessions to be able to merge them with a previous one.

If your goal is to get a first fast localization, looking at an anchor when starting would "instantly" localize at the first image or two.

harbouryi, 2017-12-21:
Thanks!
I found it hard to localized then "ORB_SLAM" , What's the difference between it and "ORB_SLAM" , why do not you use the "ORB_SLAM" to localized?

matlabbe, 2017-12-21:
Do you mean ORB_SLAM as standalone or ORB_SLAM2 as visual odometry option inside rtabmap?

harbouryi, 2017-12-21:
ORB_SLAM as visual odometry option, use orb_slam to do mapping and localization

matlabbe, 2017-12-21:
When using ORB_SLAM2 as odometry in RTAB-Map, the difference is that localization/loop closure detection is done by RTAB-Map instead of DBoW, as the map management is different between the two approaches (graph-based map vs feature-based map, for rtabmap and orb_slam2 respectively). In localization mode, you can increase Rtabmap/DetectionRate to 0, to try to localize more often. During mapping, you can increase Rtabmap/DetectionRate to 2 or 5 to get more images in the map to localize to.

cheers,
Mathieu