introlab/rtabmap_ros #215: Save map cloud

plnegre, 2018-01-15:
Hi,

Is there an option to save the RGB pointcloud that I can see with the rviz on the rtabmap/mapData topic?

I see there is a checkbox called "Download Map" in the rviz which is supposed to save the pointcloud but I cannot see I what location and with what format...

Thank you!

matlabbe, 2018-01-15:
Hi,

"Download Map" function is to refresh MapCloud plugin by re-downloading all map data. It is useful to get the map when RVIZ has been started after the map has already been created.

To save the cloud, I suggest to open the database afterward with rtabmap standalone app (`$ rtabmap ~/.ros/rtabmap.db`), then do "File->Export 3D clouds...", you will have many options to export the point cloud (PLY or PCD, or even OBJ if meshing is enabled). If you don't care about cloud density, /rtabmap/cloud_map topic can be a good choice too, which is a PointCloud2 topic that can be easily save to a PCD using [pcl_ros](http://wiki.ros.org/pcl_ros#pointcloud_to_pcd).

cheers,
Mathieu

karishmathumu, 2021-06-28:
Hello. I am having similar issue, I need to save a entire 3d pointcloud. I am using rtabmap_ros and a turtlebot2 with a Kinect camera. I am saving a rosbag file and playing it with the goal to save the pointcloud with pcl_ros package. But all I could save is a set of multiple pcd files (kind of image files), which when viewed with pcl_viewer, looks like an image  outline ( made of red dots) of a map , where as the actual pointcloud saving and viewing is the goal. Kindly please help me.

karishmathumu, 2021-06-29:
all the options on RTABMAPVIZ seem to be disabled, though I have first selected "download all the clouds" option under Edit menu on thr RTabmapviz application.

matlabbe, 2021-07-13:
crossref: http://official-rtab-map-forum.67519.x6.nabble.com/CATKIN-MAKE-error-while-cloning-rtabmap-ros-package-tp7502p8291.html