introlab/rtabmap_ros #217: Compile errors Ubuntu 14.04

aaravrav142, 2018-02-07:
Hi, 

I am trying to build rtabmap from source. Followed all the instructions carefully. 
Was able to build rtabmap but when trying to catkin_make on the catkin_ws frequently getting the build error as follows: 
```
/usr/bin/ld: warning: libpointmatcher.so.1, needed by /usr/local/lib/librtabmap_core.so, not found (try using -rpath or -rpath-link)
/usr/local/lib/librtabmap_core.so: undefined reference to `PointMatcher<float>::DataPoints::getDescriptorRowViewByName(std::string const&, unsigned int) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `PointMatcher<float>::DataPoints::DataPoints(PointMatcher<float>::DataPoints::Labels const&, PointMatcher<float>::DataPoints::Labels const&, unsigned long)'
/usr/local/lib/librtabmap_core.so: undefined reference to `PointMatcher<float>::ICPChainBase::setDefault()'
/usr/local/lib/librtabmap_core.so: undefined reference to `PointMatcher<float>::ErrorMinimizer::getWeightedPointUsedRatio() const'
/usr/local/lib/librtabmap_core.so: undefined reference to `PointMatcher<float>::DataPoints::getDescriptorRowViewByName(std::string const&, unsigned int)'
/usr/local/lib/librtabmap_core.so: undefined reference to `PointMatcher<float>::ICPChainBase::ICPChainBase()'
/usr/local/lib/librtabmap_core.so: undefined reference to `PointMatcher<float>::DataPoints::featureExists(std::string const&) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `PointMatcher<float>::ICPChainBase::loadAdditionalYAMLContent(YAML_PM::Node&)'
/usr/local/lib/librtabmap_core.so: undefined reference to `PointMatcher<float>::DataPoints::DataPoints()'
/usr/local/lib/librtabmap_core.so: undefined reference to `PointMatcher<float>::DataPoints::Label::Label(std::string const&, unsigned long)'
/usr/local/lib/libopencv_nonfree.so.2.4.11: undefined reference to `cv::ocl::integral(cv::ocl::oclMat const&, cv::ocl::oclMat&)'
/usr/local/lib/librtabmap_core.so: undefined reference to `PointMatcher<float>::DataPoints::getFeatureViewByName(std::string const&) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `PointMatcher<float>::ICP::operator()(PointMatcher<float>::DataPoints const&, PointMatcher<float>::DataPoints const&)'
/usr/local/lib/librtabmap_core.so: undefined reference to `PointMatcher<float>::ICPChainBase::~ICPChainBase()'
/usr/local/lib/librtabmap_core.so: undefined reference to `PointMatcher<float>::ICPChainBase::loadFromYaml(std::istream&)'
/usr/local/lib/librtabmap_core.so: undefined reference to `PointMatcher<float>::DataPoints::Labels::Labels()'
/usr/local/lib/librtabmap_core.so: undefined reference to `PointMatcher<float>::get()'
/usr/local/lib/librtabmap_core.so: undefined reference to `PointMatcher<float>::DataPoints::getFeatureViewByName(std::string const&)'
/usr/local/lib/librtabmap_core.so: undefined reference to `PointMatcherSupport::InvalidElement::InvalidElement(std::string const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/alex/catkin_ws/devel/lib/rtabmap_ros/icp_odometry] Error 1
make[1]: *** [rtabmap_ros/CMakeFiles/icp_odometry.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

Can you suggest something. 
Thanks

matlabbe, 2018-02-09:
Can you do "ldd /usr/local/lib/librtabmap_core.so" to see if ldd can actually find libpointmatcher.so.1 library?

aaravrav142, 2018-02-13:
Hi, 
This solved the problem. I had to edit the LD_LIBRARY_PATH. But then again I am facing another error: 

```
/usr/local/lib/libopencv_nonfree.so.2.4.11: undefined reference to `cv::ocl::integral(cv::ocl::oclMat const&, cv::ocl::oclMat&)'
collect2: error: ld returned 1 exit status
```

Although OpenCV was installed earlier and rtabmap build did not complain about OpenCL, i am not sure why rtabmap_ros cannot find it.

matlabbe, 2018-02-13:
That problem should have been fixed since 2015, see http://official-rtab-map-forum.67519.x6.nabble.com/catkin-make-ros-pkg-failed-td22.html

You can try a clean catkin build (delete catkin_ws/build folder and redo catkin_make)

aaravrav142, 2018-02-14:
Hi, 
I know this should have been solved. Do not know why it is coming up now. 
I build openCV 2 with openCL = off and everything works fine now. I will dig into this later. 
Thanks a lot Mathieu for the help.

closed 2018-02-14 by aaravrav142