introlab/rtabmap_ros #220: Global localization guess when initializing the robot

matlabbe, 2018-03-04:
Sometimes, on initialization, the robot cannot localize on the previous map because of lack of visual features. The robot has to be driven manually until it can localize. For convenience, we could force the robot to be localized using some guess. That guess could be provided by the user thought RVIZ like AMCL, a ros parameter, or the robot automatically uses the last pose in the graph after the mapping session (the robot would assume it is starting from where it has been shutdown). When running in localization mode, it won't save the last pose as the graph is not updated. Database schema should be modified to save this information. 

This can be done along this other issue: https://github.com/introlab/rtabmap_ros/issues/213#issuecomment-351807068

Cross-ref: http://official-rtab-map-forum.67519.x6.nabble.com/Initial-localization-guess-in-ROS-td4318.html

matlabbe, 2020-11-07:
Note sure when it was implemented, but we can know use `2d pose estimate` button to set an initial estimate in localization mode (see rtabmap's `initialpose` topic). This will close this issue.

closed 2020-11-07 by matlabbe

rakshith95, 2023-03-01:
@matlabbe how exactly does the `initialpose` topic work? Can we publish to the topic even after initialization, i.e. if some pose is already found, but maybe incorrect?

matlabbe, 2023-03-03:
When rtabmap is in localization mode (`Mem/IncrementalMemory=false`), you can send this pose anytime, even if it was localized.

Tf between /map and /odom will be updated so that the pose of /base_link is moved to where you set the pose (/map -> /odom -> /base_link).