introlab/rtabmap_ros #221: Rtabmapviz immediate segfault using known good launch file after complete ros rebuild

eric-schleicher, 2018-03-08:
TL;DR
rtabmapviz segfaults on start of known good launch file after complete rebuild of rtabmap (pcl 1.7) & rtabmap_ros (1.7) in a clean new workspace.

========================================
I've been working to get an update version of rtabmap and rtabmap_ros running.   I originally had a few other packages which forces the use of PCL 1.8 in the workspace, which triggered this process (as it preventing rtabmap_ros from compiling/linking correctly to an rtabmap version compiles with 1.7.x.

I was successful in getting both rtabmap and rtabmap_ros to compile with 1.8 howevevr was not able to get a known good launch file to work; with both the them crashing: rtabmapviz (immediate seg fault) **and** rtabmap (when messages started to flow).

I struggled in vain to get it to work, and so made the decision to separate the workspaces so that i could verify that it wasn't pcl 1.8 causing any issues.  I rebuilt rtabmap 0.14.0 with pcl 1.7 (ros kinetic binary) and create a clean new workspace just for rtabmap.  everything compiles successfully.

##### Fig 1. Launch File
```xml
<launch>

   <!-- Choose visualization -->
   <arg name="rviz" default="false" />
   <arg name="rtabmapviz" default="true" /> 

   <!-- Corresponding config files -->
   <arg name="rtabmapviz_cfg"          default="-d $(find rtabmap_ros)/launch/config/rgbd_gui.ini" />
   <arg name="rviz_cfg"                default="-d $(find rtabmap_ros)/launch/config/rgbd.rviz" />

<!-- Which image resolution to process in rtabmap: sd, qhd, hd -->
    <arg name="resolution" default="qhd"/>
    <arg name="frame_id" default="kinect2_base"/>
   
   <!-- Synchronize kinect data before rtabmap node, generating single /rgbd_image topic -->
   <node pkg="nodelet" type="nodelet" name="rgbd_sync" args="load rtabmap_ros/rgbd_sync kinect2">
      <remap from="rgb/image"       to="/kinect2/$(arg resolution)/image_color_rect"/>
      <remap from="depth/image"     to="/kinect2/$(arg resolution)/image_depth_rect"/>
      <remap from="rgb/camera_info" to="/kinect2/$(arg resolution)/camera_info"/>
      <param name="approx_sync"     value="false"/> <!-- exact synchronization -->
   </node> 

   <node pkg="rtabmap_ros" type="rtabmap" name="rtabmap" output="screen" args="--delete_db_on_start">

      <!-- Remap topics -->
      <param name="subscribe_depth"  type="bool"   value="false"/>
      <param name="subscribe_rgbd"   type="bool"   value="true"/>
      <param name="rgbd_cameras"     type="int"    value="1"/>
      <remap from="rgbd_image0"      to="/rgbd_image"/>

      <!-- Frames: fixed frame is /map, base frame of lidar and kinect is /kinect2_ir_optical_frame -->
       <!--<param name="odom_frame_id"  value="map"/>-->
       <param name="odom_frame_id" value="world"/>
      <param name="frame_id"       value="$(arg frame_id)"/>
      <param name="publish_tf"     value="false"/>
   
      <!-- Disable loop closure detection -->
      <param name="Kp/MaxFeatures" type="string" value="-1"/>

       <!-- For laser assembler integration, use rate at which scans are assembled --> <!-- not sure if thie comment still valid -->
      <param name="Rtabmap/DetectionRate"          type="string" value="10"/>
   </node>

    <!-- Visualisation RTAB-Map -->
    <node if="$(arg rtabmapviz)" pkg="rtabmap_ros" type="rtabmapviz" name="rtabmapviz" args="$(arg rtabmapviz_cfg)" output="screen">
      <param name="subscribe_depth" type="bool" value="true"/>
      <param name="subscribe_odom_info" type="bool" value="true"/>
      <param name="approx_sync" type="bool" value="true"/>
      <param name="frame_id" type="string" value="$(arg frame_id)"/>
    
      <remap from="rgb/image" to="/kinect2/$(arg resolution)/image_color_rect"/>
      <remap from="depth/image" to="/kinect2/$(arg resolution)/image_depth_rect"/>
      <remap from="rgb/camera_info" to="/kinect2/$(arg resolution)/camera_info"/>
    </node>

</launch>

```
[... 118 lines cut ...]
[New Thread 0x7fff87fff700 (LWP 26694)]
[New Thread 0x7fff877fe700 (LWP 26695)]
[New Thread 0x7fff86ffd700 (LWP 26700)]
[New Thread 0x7fff867fc700 (LWP 26713)]
[ INFO] [1520490769.005292080]: rtabmapviz: Using configuration from "/home/rosuser/catkin_ws/src/rtabmap_ros/launch/config/rgbd_gui.ini"
[Thread 0x7fff961cc700 (LWP 26689) exited]

Thread 1 "rtabmapviz" received signal SIGSEGV, Segmentation fault.
0x00007ffff001e657 in rtabmap::Transform::Transform() () from /usr/local/lib/librtabmap_core.so.0.14
(gdb) backtrace
#0  0x00007ffff001e657 in rtabmap::Transform::Transform() () from /usr/local/lib/librtabmap_core.so.0.14
#1  0x00007fffeff9f20d in rtabmap::CameraModel::CameraModel() () from /usr/local/lib/librtabmap_core.so.0.14
#2  0x000000000053064e in rtabmap::StereoCameraModel::StereoCameraModel() ()
#3  0x00007fffef7c75f6 in rtabmap::CalibrationDialog::CalibrationDialog(bool, QString const&, bool, QWidget*) () from /usr/local/lib/librtabmap_gui.so.0.14
#4  0x00007fffef69055a in rtabmap::PreferencesDialog::PreferencesDialog(QWidget*) () from /usr/local/lib/librtabmap_gui.so.0.14
#5  0x0000000000599e60 in PreferencesDialogROS::PreferencesDialogROS(QString const&) ()
#6  0x0000000000523974 in rtabmap_ros::GuiWrapper::GuiWrapper(int&, char**) ()
#7  0x000000000051db19 in main ()
(gdb) 

```
this is the limit of my diagnostic ability.   buy my suspicion is that perhaps this is something to do with the 
 configuration?   the rgb_gui.ini I'm using is the default one in the repository.  

Any help very much appreciated, as I'm in a pinch to get this working again.

matlabbe, 2018-03-08:
rtabmap/rtabmap_ros won't do seg fault if the launch file has bad parameters, it would assert instead pointing where is the problem. Seg faults happen more often when: 

1. a slightly different library is taken than the one that is built against (when there are multiple versions of the same libraries)
2. rtabmap_ros links on different version of third party libraries, like PCL 1.7 vs PCL 1.8, Opencv 2.4 vs Opencv 3
3. rtabmap and thirdparty libraries have been built against different Eigen versions. With gdb, it would crash in functions using Eigen. Note also if some libraries are built with -march=native and other not, this could generate crashes too
4. not using compatible rtabmap and rtabmap_ros together. Same tag or branch should be used for both.
4. bug in rtabmap/rtabmap_ros

From the gdb log above, it is crashing in the constructor of Transform, maybe related to 1). I tried on my side with rtabmap 0.14.0 + rtabmap_ros (master) and everything works fine (rtabmap, rtabmapviz, rviz). Here is my cmake status of rtabmap with include directories used:
```bash
~/workspace/rtabmap/build $ cmake ..
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- Could NOT find PCL_APPS (missing:  PCL_APPS_LIBRARY) 
-- Found OpenMP
-- Found OpenCV: /opt/ros/kinetic/include/opencv-3.3.1;/opt/ros/kinetic/include/opencv-3.3.1/opencv
-- Found PCL: /usr/include/pcl-1.7;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/openni2;/usr/include/vtk-6.2;/usr/include/jsoncpp;/usr/include/freetype2;/usr/include/x86_64-linux-gnu/freetype2;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent/include;/usr/lib/openmpi/include;/usr/lib/openmpi/include/openmpi;/usr/include/hdf5/openmpi;/usr/include/x86_64-linux-gnu;/usr/include/libxml2;/usr/include/python2.7;/usr/include/tcl
-- Found ZLIB: /usr/include
-- Found Freenect: /opt/ros/kinetic/include/libfreenect
-- Found freenect2: /usr/local/include
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
-- Found g2o: /opt/ros/kinetic/include;/usr/include/suitesparse;/usr/include/suitesparse
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include;/usr/include;/usr/include/eigen3
-- Found libpointmatcher: /usr/local/include;/usr/include;/usr/include/eigen3;/usr/local/include;/home/mathieu/workspace/libpointmatcher/pointmatcher
-- Found CUDA: /usr/local/cuda (found suitable exact version "8.0") 
-- Found ZED sdk: /usr/local/zed/include
-- Found CUDA: /usr/local/cuda (found version "8.0") 
-- Found CUDA: /usr/local/cuda/include
-- Found RealSense: /opt/ros/kinetic/include
-- Found RealSenseSlam: /opt/ros/kinetic/include
-- Found octomap: /opt/ros/kinetic/include
-- Found CPUTSDF: /usr/local/include/cpu_tsdf
-- Found Pthreads
-- --------------------------------------------
-- Info :
--   Version : 0.14.0
--   CMAKE_INSTALL_PREFIX = /home/mathieu/catkin_tmp/devel
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp -std=c++11
--   With OpenCV 3 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = YES (License: Non commercial)
--   With Freenect             = YES (License: Apache v2 and/or GPLv2)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = YES (License: Apache v2 and/or GPLv2)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--   With g2o                  = YES (License: BSD)
--   With GTSAM                = YES (License: BSD)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = NO (cvsba not found)
--   With libpointmatcher      = YES (License: BSD)
--   With ZED                  = YES (With CUDA)
--   With RealSense            = YES (License: Apache-2)
--   With RealSenseSlam        = YES
--   With OCTOMAP              = YES (License: BSD)
--   With CPUTSDF              = YES (License: BSD)
--   With libfovis             = NO (libfovis not found)
--   With libviso2             = NO (libviso2 not found)
--   With dvo_core             = NO (dvo_core not found)
--   With ORB_SLAM2            = NO (WITH_G2O should be OFF as ORB_SLAM2 uses its own g2o version)
--   With Qt5                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mathieu/workspace/rtabmap_master/build
```

matlabbe, 2018-03-08:
You can try bring up only rtabmap (`$ rtabmap`) to see if at least the rtabmap library is fine, and the problem is when compiling on rtabmap_ros side (in that case maybe catkin_make is using different versions of libraries than those used by rtabmap).

eric-schleicher, 2018-03-08:
Yes rtabmap application works just fine. and in the current state PCL 1.8 is ***not*** installed, it's just compiled in my `~`

Should i try catkin_make clean -DCMAKE_BUILD_TYPE=RELEASE VERBOSE=1 or is the answer in a CMAKE cache file?

eric-schleicher, 2018-03-08:
@matlabbe  thanks for the helpful diagnostics.   the straggling opencv libraries were the issue.

by removing 
- libopencv-core-dev
- libopencv-dev

after recompiling i was able to get it all working again.

Thanks!

closed 2018-03-08 by eric-schleicher

matlabbe, 2018-03-08:
(I was writing an answer when you closed the issue! glad it worked, I still put here more info for reference...)

When sourcing ros setup.bash, it should put the kinetic paths before system paths in PATH. CMake should then find OpenCV in kinetic before the one in system. As long as all other ros packages are linked to opencv in kinetic and rtabmap, everything should be fine. 

So in your case, "Found OpenCV" should point in kinetic folder. Here is my ldd of installed rtabmap (it should be similar if installed in /usr/local). However, even on my system I have two OpenCV 3 installed, one in kinetic and one in system path. It seems to use "`libopencv_core3.so.3.3 => /opt/ros/kinetic/lib/libopencv_core3.so.3.3`" and "`libopencv_core.so.3.1 => /usr/lib/x86_64-linux-gnu/libopencv_core.so.3.1`"!? at the same time and no problems (maybe because the one in Kinetic is first in ldd)... I think mixing OpenCV 3 versions are fine but I know that you cannot mix OpenCV 2 and OpenCV 3 libraries.
```
$ ldd ~/catkin_ws/devel/bin/rtabmap
	linux-vdso.so.1 =>  (0x00007ffefc1ec000)
	librtabmap_gui.so.0.16 => /home/mathieu/catkin_ws/devel/lib/librtabmap_gui.so.0.16 (0x00007fcc24870000)
	librtabmap_core.so.0.16 => /home/mathieu/catkin_ws/devel/lib/librtabmap_core.so.0.16 (0x00007fcc23e89000)
	libSP_Core.so.2 => /usr/lib/libSP_Core.so.2 (0x00007fcc1f8f8000)
	librtabmap_utilite.so.0.16 => /home/mathieu/catkin_ws/devel/lib/librtabmap_utilite.so.0.16 (0x00007fcc1f6d8000)
	libQt5Widgets.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007fcc1f04b000)
	libQt5Core.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007fcc1eb75000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fcc1e7f3000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fcc1e5dd000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fcc1e213000)
	libboost_system.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.58.0 (0x00007fcc1e00f000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fcc1ddf2000)
	libpcl_common.so.1.7 => /usr/lib/x86_64-linux-gnu/libpcl_common.so.1.7 (0x00007fcc1db38000)
	libpcl_io.so.1.7 => /usr/lib/x86_64-linux-gnu/libpcl_io.so.1.7 (0x00007fcc1d6c9000)
	libpcl_search.so.1.7 => /usr/lib/x86_64-linux-gnu/libpcl_search.so.1.7 (0x00007fcc1d2d9000)
	libpcl_surface.so.1.7 => /usr/lib/x86_64-linux-gnu/libpcl_surface.so.1.7 (0x00007fcc1cc93000)
	libpcl_visualization.so.1.7 => /usr/lib/x86_64-linux-gnu/libpcl_visualization.so.1.7 (0x00007fcc1c91a000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fcc1c611000)
	libvtkRenderingVolumeOpenGL-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkRenderingVolumeOpenGL-6.2.so.6.2 (0x00007fcc1c36a000)
	libvtkRenderingFreeTypeOpenGL-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkRenderingFreeTypeOpenGL-6.2.so.6.2 (0x00007fcc1c158000)
	libvtkRenderingMatplotlib-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkRenderingMatplotlib-6.2.so.6.2 (0x00007fcc1bf47000)
	libvtkRenderingFreeTypeFontConfig-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkRenderingFreeTypeFontConfig-6.2.so.6.2 (0x00007fcc1bd3d000)
	libvtkIOMPIImage-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkIOMPIImage-6.2.so.6.2 (0x00007fcc1bb14000)
	libcpu_tsdf.so => /usr/local/lib/libcpu_tsdf.so (0x00007fcc1b8d3000)
	libQt5Svg.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Svg.so.5 (0x00007fcc25098000)
	libQt5PrintSupport.so.5 => /usr/lib/x86_64-linux-gnu/libQt5PrintSupport.so.5 (0x00007fcc25025000)
	libvtkGUISupportQt-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkGUISupportQt-6.2.so.6.2 (0x00007fcc1b688000)
	libvtkInteractionStyle-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkInteractionStyle-6.2.so.6.2 (0x00007fcc1b3fc000)
	libvtkRenderingLOD-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkRenderingLOD-6.2.so.6.2 (0x00007fcc1b1e2000)
	libvtkRenderingAnnotation-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkRenderingAnnotation-6.2.so.6.2 (0x00007fcc1ae8a000)
	libvtkRenderingOpenGL-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkRenderingOpenGL-6.2.so.6.2 (0x00007fcc1a9e2000)
	libvtkRenderingFreeType-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkRenderingFreeType-6.2.so.6.2 (0x00007fcc1a71f000)
	libvtkRenderingCore-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkRenderingCore-6.2.so.6.2 (0x00007fcc1a28b000)
	libvtkFiltersSources-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkFiltersSources-6.2.so.6.2 (0x00007fcc19fc5000)
	libvtkCommonExecutionModel-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkCommonExecutionModel-6.2.so.6.2 (0x00007fcc19d17000)
	libvtkCommonDataModel-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkCommonDataModel-6.2.so.6.2 (0x00007fcc197a8000)
	libvtkCommonMath-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkCommonMath-6.2.so.6.2 (0x00007fcc19586000)
	libvtkCommonCore-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkCommonCore-6.2.so.6.2 (0x00007fcc19058000)
	libopencv_calib3d3.so.3.3 => /opt/ros/kinetic/lib/libopencv_calib3d3.so.3.3 (0x00007fcc18cd4000)
	libopencv_imgcodecs3.so.3.3 => /opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.3 (0x00007fcc186bf000)
	libopencv_imgproc3.so.3.3 => /opt/ros/kinetic/lib/libopencv_imgproc3.so.3.3 (0x00007fcc15a1e000)
	libopencv_core3.so.3.3 => /opt/ros/kinetic/lib/libopencv_core3.so.3.3 (0x00007fcc14bb8000)
	libgomp.so.1 => /usr/lib/x86_64-linux-gnu/libgomp.so.1 (0x00007fcc14996000)
	libQt5Gui.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007fcc1444e000)
[... cut ...]
	libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007fcbd8e6f000)
	libkrb5.so.3 => /usr/lib/x86_64-linux-gnu/libkrb5.so.3 (0x00007fcbd8b9d000)
	libk5crypto.so.3 => /usr/lib/x86_64-linux-gnu/libk5crypto.so.3 (0x00007fcbd896e000)
	libcom_err.so.2 => /lib/x86_64-linux-gnu/libcom_err.so.2 (0x00007fcbd876a000)
	libkrb5support.so.0 => /usr/lib/x86_64-linux-gnu/libkrb5support.so.0 (0x00007fcbd855f000)
	libkeyutils.so.1 => /lib/x86_64-linux-gnu/libkeyutils.so.1 (0x00007fcbd835b000)
	libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x00007fcbd8140000)
```

eric-schleicher, 2018-03-08:
whats interesting was that with the other libs installed even if i sourced a workspace, the value wouldn't change.