introlab/rtabmap_ros #222: librtabmap_gui.so.0.15: cannot open shared object file

QinZiwen, 2018-03-11:
For view rtabmap.db, I entered the following script into the terminal and I got an error：
```
~/.ros$ rtabmap-databaseViewer rtabmap.db 
rtabmap-databaseViewer: error while loading shared libraries: librtabmap_gui.so.0.15: cannot open shared object file: No such file or directory
```
But, This file exists in /usr/local/lib
```
/usr/local/lib$ ll librtabmap_*
lrwxrwxrwx 1 root root      23  3月 10 22:56 librtabmap_core.so -> librtabmap_core.so.0.15
lrwxrwxrwx 1 root root      25  3月 10 22:56 librtabmap_core.so.0.15 -> librtabmap_core.so.0.15.0
-rw-r--r-- 1 root root 8170138  3月 10 22:46 librtabmap_core.so.0.15.0
lrwxrwxrwx 1 root root      22  3月 10 22:56 librtabmap_gui.so -> librtabmap_gui.so.0.15
lrwxrwxrwx 1 root root      24  3月 10 22:56 librtabmap_gui.so.0.15 -> librtabmap_gui.so.0.15.0
-rw-r--r-- 1 root root 4470408  3月 10 22:47 librtabmap_gui.so.0.15.0
lrwxrwxrwx 1 root root      26  3月 10 22:56 librtabmap_utilite.so -> librtabmap_utilite.so.0.15
lrwxrwxrwx 1 root root      28  3月 10 22:56 librtabmap_utilite.so.0.15 -> librtabmap_utilite.so.0.15.0
-rw-r--r-- 1 root root  155797  3月 10 22:45 librtabmap_utilite.so.0.15.0
```
ldd as follow
```
/usr/local/bin$ ldd rtabmap-databaseViewer 
	linux-vdso.so.1 =>  (0x00007fff8b970000)
	librtabmap_gui.so.0.15 => not found
	librtabmap_utilite.so.0.15 => not found
	libQtGui.so.4 => /usr/lib/x86_64-linux-gnu/libQtGui.so.4 (0x00007f687db7f000)
	libQtCore.so.4 => /usr/lib/x86_64-linux-gnu/libQtCore.so.4 (0x00007f687d69b000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f687d397000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f687d181000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f687cdb8000)
	libfontconfig.so.1 => /usr/lib/x86_64-linux-gnu/libfontconfig.so.1 (0x00007f687cb7c000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f687c95e000)
	libaudio.so.2 => /usr/lib/x86_64-linux-gnu/libaudio.so.2 (0x00007f687c746000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f687c43e000)
	libpng12.so.0 => /lib/x86_64-linux-gnu/libpng12.so.0 (0x00007f687c219000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f687c000000)
	libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f687bd5d000)
	libgobject-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007f687bb0c000)
	libSM.so.6 => /usr/lib/x86_64-linux-gnu/libSM.so.6 (0x00007f687b904000)
	libICE.so.6 => /usr/lib/x86_64-linux-gnu/libICE.so.6 (0x00007f687b6e8000)
	libXi.so.6 => /usr/lib/x86_64-linux-gnu/libXi.so.6 (0x00007f687b4d8000)
	libXrender.so.1 => /usr/lib/x86_64-linux-gnu/libXrender.so.1 (0x00007f687b2ce000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007f687b0bc000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f687ad87000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f687aa81000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f687a87d000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f687a675000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f687e831000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f687a44b000)
	libXt.so.6 => /usr/lib/x86_64-linux-gnu/libXt.so.6 (0x00007f687a1e5000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f6879fe1000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f6879da3000)
	libffi.so.6 => /usr/lib/x86_64-linux-gnu/libffi.so.6 (0x00007f6879b9b000)
	libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007f6879996000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f6879777000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f6879571000)
```
ldd show:
```
librtabmap_gui.so.0.15 => not found
librtabmap_utilite.so.0.15 => not found
```

QinZiwen, 2018-03-11:
I found that the default installation path for rtabmap is "/usr/local/lib" and "/usr/local/bin"
So, I append "/usr/local/lib" in .bashrc, as follow:
```
export LD_LIBRARY_PATH="${LD_LIBRARY_PATH}:/usr/local/lib"
```
But, when I run "rtabmap", new error emerge:
```
$ ./rtabmap
ERROR: In /build/buildd/vtk-5.8.0/Rendering/vtkXOpenGLRenderWindow.cxx, line 404
vtkXOpenGLRenderWindow (0x31f4370): Could not find a decent visual

ERROR: In /build/buildd/vtk-5.8.0/Rendering/vtkXOpenGLRenderWindow.cxx, line 404
vtkXOpenGLRenderWindow (0x31f4370): Could not find a decent visual

ERROR: In /build/buildd/vtk-5.8.0/Rendering/vtkXOpenGLRenderWindow.cxx, line 631
vtkXOpenGLRenderWindow (0x31f4370): Cannot create GLX context.  Aborting.

Aborted (core dumped)

$ ./rtabmap-databaseViewer ~/.ros/rtabmap.db 
ERROR: In /build/buildd/vtk-5.8.0/Rendering/vtkXOpenGLRenderWindow.cxx, line 404
vtkXOpenGLRenderWindow (0x3312560): Could not find a decent visual

ERROR: In /build/buildd/vtk-5.8.0/Rendering/vtkXOpenGLRenderWindow.cxx, line 404
vtkXOpenGLRenderWindow (0x3312560): Could not find a decent visual

ERROR: In /build/buildd/vtk-5.8.0/Rendering/vtkXOpenGLRenderWindow.cxx, line 631
vtkXOpenGLRenderWindow (0x3312560): Cannot create GLX context.  Aborting.

Aborted (core dumped)
```

matlabbe, 2018-03-11:
What is your machine? Is it virtual machine? See https://github.com/UV-CDAT/uvcdat/issues/1134#issuecomment-93031935
Ankther post when connecting via ssh: www.cfd-online.com/Forums/openfoam-installation/57218-parafoam-error-could-not-find-

QinZiwen, 2018-03-12:
Not a virtual machine. It is Ubuntu 14.04, cuda 8.0, nvidia-gtx-1070.
I still don't know how to solve this problem by see [UV-CDAT/uvcdat#1134 (comment)](https://github.com/UV-CDAT/uvcdat/issues/1134#issuecomment-93031935)
This link,https://www.cfd-online.com/Forums/openfoam-installation/57218-parafoam-error-could-not-find- , cannot be opened

matlabbe, 2018-03-12:
Sorry, copy-paste failed: https://www.cfd-online.com/Forums/openfoam-installation/57218-parafoam-error-could-not-find-decent-visual.html

matlabbe, 2018-03-12:
Another post based on PCL using VTK: http://www.pcl-users.org/pcd-viewer-failure-td1684243.html

QinZiwen, 2018-03-12:
pcl_viewer runs without problems, my pcl version is 1.7

matlabbe, 2018-03-13:
RTAB-Map depends on VTK by PCL and Qt. The difference between pcl_viewer and RTAB-Map is that the OpenGL context is created using QVTKWidget in RTAB-Map (which is VTK + Qt). There is maybe a problem with QVTKWidget, that may need to be rebuilt from source on your computer.

HaowenLai, 2020-03-23:
@QinZiwen It seems that you have encountered a similar problem like the issue in here: https://github.com/introlab/rtabmap/issues/459  
Maybe you can try the method I mentioned in it to see whether it works.