introlab/rtabmap_ros #224: Rtabmap_ros /  Orbbec Astra / ROS indigo

MariRob, 2018-03-22:
I use **rtabmap_ros** on ROS indigo (Ubuntu 14.04) , trying to get it working with **Orbbec Astra** 
is't possible ??

> roslaunch rtabmap_ros rtabmap.launch 

`[ WARN] [1521735878.626707188]: /rtabmap/rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. 
/rtabmap/rgbd_odometry subscribed to (approx sync):
   /camera/rgb/image_rect_color,
   /camera/depth_registered/image_raw,
   /camera/rgb/camera_info
[ WARN] [1521735879.160572040]: /rtabmap/rtabmapviz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmapviz subscribed to (approx sync):
   /rtabmap/odom,
   /camera/rgb/image_rect_color,
   /camera/depth_registered/image_raw,
   /camera/rgb/camera_info,
   /rtabmap/odom_info
`
how can i solve this problem ? 
I appreciate your help...
Thanks.

matlabbe, 2018-03-22:
Yes it is possible. You may have to remap some topics though. See this [tutorial](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping) for some examples. The idea is to remap correctly the depth, rgb and camera_info topics:
```bash
$ roslaunch rtabmap_ros rtabmap.launch \
   rtabmap_args:="--delete_db_on_start" \
   rgb_topic:=/camera/rgb/image_rect_color \
   depth_topic:=/camera/depth_registered/image_raw \
   camera_info_topic:=/camera/rgb/camera_info \
   frame_id:=camera_link
```
`depth_topic` should be a depth image topic that is **registered** to RGB camera! (each RGB pixel matches exactly its corresponding depth pixel)

You can make sure they are published by your camera driver with `$ rostopic hz /camera/rgb/image_rect_color` for example. The nodes rgbd_odometry/rtabmap/rtabmapviz will complain like the warnings you have if the topics are not published.

cheers

MariRob, 2018-03-23:
Hello ^^
Thank you so much for your answer
I'm trying to apply your proposal ..

cheers

MariRob, 2018-03-23:
[image]
So the error disappears , but nothing happens ...even if I disconnect the camera, it gives the same result seen on the picture.
can you help me ? 
`$ rostopic hz /camera/rgb/camera_info`  gives : 

>   subscribed to [/camera/rgb/camera_info]

`~$ rostopic hz /camera/depth_registered/image_raw`  gives: 

> subscribed to [/camera/depth_registered/image_raw]

`~$ rostopic hz /camera/rgb/image_rect_color`   gives:

> subscribed to [/camera/rgb/image_rect_color]

matlabbe, 2018-03-23:
"rostopic hz" should show the rate of the subscribed topic if it is actually sent, like:
```bash
$ rostopic hz /camera/rgb/image_rect_color
subscribed to [/camera/rgb/image_rect_color]
no new messages
average rate: 30.323
	min: 0.024s max: 0.034s std dev: 0.00193s window: 31
average rate: 30.125
	min: 0.024s max: 0.037s std dev: 0.00164s window: 61
average rate: 30.084
	min: 0.024s max: 0.037s std dev: 0.00146s window: 91
average rate: 30.079
	min: 0.024s max: 0.037s std dev: 0.00141s window: 105
```

You can also use `rqt_image_view` to see if the images are published.

cheers