introlab/rtabmap_ros #225: Process died at startup, exit code -11

RajathSwaroop, 2018-03-24:
I am using rtabmap ros with the realsense D415 camera and when I launch the file I get this error mentioned below. The process works fine on one of my laptops and fails with this error on the other. Both of them have same version of rtab map installed. Can someone please explain the error.

[rtabmap/rtabmapviz-3] process has died [pid 12102, exit code -11, cmd /home/rajath/catkin_ws/devel/lib/rtabmap_ros/rtabmapviz -d ~/.ros/rtabmap_gui.ini rgb/image:=/camera/color/image_raw depth/image:=/camera/depth/image_rect_raw rgb/camera_info:=/camera/color/camera_info rgbd_image:=rgbd_image left/image_rect:=/stereo_camera/left/image_rect_color right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info right/camera_info:=/stereo_camera/right/camera_info scan:=/scan scan_cloud:=/scan_cloud odom:=odom __name:=rtabmapviz __log:=/home/rajath/.ros/log/a8aa5aca-2f05-11e8-bf78-00e18c8e5af3/rtabmap-rtabmapviz-3.log].
log file: /home/rajath/.ros/log/a8aa5aca-2f05-11e8-bf78-00e18c8e5af3/rtabmap-rtabmapviz-3*.log
[rtabmap/rgbd_odometry-1] process has died [pid 12100, exit code -11, cmd /home/rajath/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry --delete_db_on_start rgb/image:=/camera/color/image_raw depth/image:=/camera/depth/image_rect_raw rgb/camera_info:=/camera/color/camera_info rgbd_image:=rgbd_image odom:=odom __name:=rgbd_odometry __log:=/home/rajath/.ros/log/a8aa5aca-2f05-11e8-bf78-00e18c8e5af3/rtabmap-rgbd_odometry-1.log].
log file: /home/rajath/.ros/log/a8aa5aca-2f05-11e8-bf78-00e18c8e5af3/rtabmap-rgbd_odometry-1*.log
[ INFO] [1521856773.348328495]: RTAB-Map detection rate = 1.000000 Hz
[ INFO] [1521856773.348415900]: rtabmap: Deleted database "/home/rajath/.ros/rtabmap.db" (--delete_db_on_start or -d are set).

matlabbe, 2018-03-24:
Sometimes this kind of error is a mismatch of libraries. What is the output of cmake for rtabmap? Is rtabmap standalone working (`$ rtabmap`)? See also #28 and #226

RajathSwaroop, 2018-03-25:
Yes the rtabmap standalone is working, I was able to get the GUI

matlabbe, 2018-03-25:
To compare between the laptops, can you do "$ ldd librtabmap_ros.so" on where the ros lib is built (e.g., catkin_ws/devel/lib/librtabmap_ros.so)? This could help to know if different third party library versions are used.

Cheers

RajathSwaroop, 2018-03-25:
Sure will do that and post the output here. Thanks

QinZiwen, 2018-03-26:
@RajathSwaroop  I have the same problem. How did you solve it?

RajathSwaroop, 2018-03-27:
@matlabbe For both of the laptops use of "ldd librtabmap.so" says "ldd: ./librtabmap_ros.so: no such file or directory"

but I could find the librtabmap_ros.so in the  catkin_ws/devel/lib folder in both the laptops.

@QinZiwen I did not solve the issue yet. Trying to fix it

matlabbe, 2018-03-27:
Strange, try rtabmap instead like in your log:

ldd /home/rajath/catkin_ws/devel/lib/rtabmap_ros/rtabmap

RajathSwaroop, 2018-03-27:
It gave the same output "no such file or directory" but i was able to find the file at the same location that you mentioned "/home/rajath/catkin_ws/devel/lib/rtabmap_ros"

QinZiwen, 2018-03-27:
@matlabbe I try ldd rtabmap and I use opencv2.4.13, pcl 1.7, vtk 6.3
```
~/catkin_ws/devel/lib/rtabmap_ros$ ldd rtabmap
	linux-vdso.so.1 =>  (0x00007ffdba159000)
	libroscpp.so => /opt/ros/indigo/lib/libroscpp.so (0x00007f07d4d52000)
	librosconsole.so => /opt/ros/indigo/lib/librosconsole.so (0x00007f07d4b26000)
	libboost_system.so.1.54.0 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.54.0 (0x00007f07d4922000)
	libnodeletlib.so => /opt/ros/indigo/lib/libnodeletlib.so (0x00007f07d46c6000)
	librtabmap_core.so.0.16 => /usr/local/lib/librtabmap_core.so.0.16 (0x00007f07d3e4d000)
	librtabmap_utilite.so.0.16 => /usr/local/lib/librtabmap_utilite.so.0.16 (0x00007f07d3c2e000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f07d392a000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f07d3714000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f07d334b000)
	libroscpp_serialization.so => /opt/ros/indigo/lib/libroscpp_serialization.so (0x00007f07d3148000)
	librostime.so => /opt/ros/indigo/lib/librostime.so (0x00007f07d2f1e000)
	libxmlrpcpp.so => /opt/ros/indigo/lib/libxmlrpcpp.so (0x00007f07d2d00000)
	libcpp_common.so => /opt/ros/indigo/lib/libcpp_common.so (0x00007f07d2af8000)
	libboost_thread.so.1.54.0 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.54.0 (0x00007f07d28e2000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f07d26c4000)
	libboost_filesystem.so.1.54.0 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.54.0 (0x00007f07d24ae000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f07d21a8000)
	librosconsole_log4cxx.so => /opt/ros/indigo/lib/librosconsole_log4cxx.so (0x00007f07d1f94000)
	librosconsole_backend_interface.so => /opt/ros/indigo/lib/librosconsole_backend_interface.so (0x00007f07d1d92000)
	liblog4cxx.so.10 => /usr/lib/liblog4cxx.so.10 (0x00007f07d19aa000)
	libboost_regex.so.1.54.0 => /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.54.0 (0x00007f07d16a3000)
	libbondcpp.so => /opt/ros/indigo/lib/libbondcpp.so (0x00007f07d1473000)
	libclass_loader.so => /opt/ros/indigo/lib/libclass_loader.so (0x00007f07d1254000)
	libroslib.so => /opt/ros/indigo/lib/libroslib.so (0x00007f07d1042000)
	libtinyxml.so.2.6.2 => /usr/lib/x86_64-linux-gnu/libtinyxml.so.2.6.2 (0x00007f07d0e2d000)
	libconsole_bridge.so.0.2 => /usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.2 (0x00007f07d0c23000)
	libpcl_common.so.1.7 => /usr/local/lib/libpcl_common.so.1.7 (0x00007f07d0980000)
	libOpenNI2.so.0 => /usr/lib/libOpenNI2.so.0 (0x00007f07d0725000)
	libpcl_io.so.1.7 => /usr/local/lib/libpcl_io.so.1.7 (0x00007f07d0332000)
	libpcl_kdtree.so.1.7 => /usr/local/lib/libpcl_kdtree.so.1.7 (0x00007f07d0020000)
	libpcl_search.so.1.7 => /usr/local/lib/libpcl_search.so.1.7 (0x00007f07cfca4000)
	libpcl_surface.so.1.7 => /usr/local/lib/libpcl_surface.so.1.7 (0x00007f07cf6ae000)
	libpcl_filters.so.1.7 => /usr/local/lib/libpcl_filters.so.1.7 (0x00007f07cee53000)
	libpcl_features.so.1.7 => /usr/local/lib/libpcl_features.so.1.7 (0x00007f07cd47d000)
	libpcl_sample_consensus.so.1.7 => /usr/local/lib/libpcl_sample_consensus.so.1.7 (0x00007f07cc9ae000)
	libpcl_segmentation.so.1.7 => /usr/local/lib/libpcl_segmentation.so.1.7 (0x00007f07cbea9000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f07cbc90000)
	libfreenect.so.0.5 => /opt/ros/indigo/lib/libfreenect.so.0.5 (0x00007f07cba81000)
	librealsense.so.1 => /opt/ros/indigo/lib/librealsense.so.1 (0x00007f07cb796000)
	libdc1394.so.22 => /usr/lib/x86_64-linux-gnu/libdc1394.so.22 (0x00007f07cb522000)
	libomp.so => /usr/local/lib/libomp.so (0x00007f07cb283000)
	libcvsba.so => /usr/local/lib/libcvsba.so (0x00007f07cb050000)
	liboctomap.so.1.6 => /opt/ros/indigo/lib/liboctomap.so.1.6 (0x00007f07cae08000)
	liboctomath.so.1.6 => /opt/ros/indigo/lib/liboctomath.so.1.6 (0x00007f07cac02000)
	libopencv_nonfree.so.2.4 => /usr/local/lib/libopencv_nonfree.so.2.4 (0x00007f07ca9e3000)
	libopencv_gpu.so.2.4 => /usr/local/lib/libopencv_gpu.so.2.4 (0x00007f07ca798000)
	libopencv_video.so.2.4 => /usr/local/lib/libopencv_video.so.2.4 (0x00007f07ca53b000)
	libopencv_calib3d.so.2.4 => /usr/local/lib/libopencv_calib3d.so.2.4 (0x00007f07ca25e000)
	libopencv_features2d.so.2.4 => /usr/local/lib/libopencv_features2d.so.2.4 (0x00007f07c9fae000)
	libopencv_highgui.so.2.4 => /usr/local/lib/libopencv_highgui.so.2.4 (0x00007f07c9d5c000)
	libopencv_imgproc.so.2.4 => /usr/local/lib/libopencv_imgproc.so.2.4 (0x00007f07c986c000)
	libopencv_core.so.2.4 => /usr/local/lib/libopencv_core.so.2.4 (0x00007f07c93c2000)
	libvtkCommonCore-6.3.so.1 => /usr/local/lib/libvtkCommonCore-6.3.so.1 (0x00007f07c8b99000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f07c8995000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f07d50b7000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f07c878d000)
	libaprutil-1.so.0 => /usr/lib/x86_64-linux-gnu/libaprutil-1.so.0 (0x00007f07c8566000)
	libapr-1.so.0 => /usr/lib/x86_64-linux-gnu/libapr-1.so.0 (0x00007f07c8335000)
	libicuuc.so.52 => /usr/lib/x86_64-linux-gnu/libicuuc.so.52 (0x00007f07c7fbc000)
	libicui18n.so.52 => /usr/lib/x86_64-linux-gnu/libicui18n.so.52 (0x00007f07c7bb5000)
	libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007f07c79b0000)
	libPocoFoundation.so.9 => /usr/lib/libPocoFoundation.so.9 (0x00007f07c7660000)
	librospack.so => /opt/ros/indigo/lib/librospack.so (0x00007f07c7420000)
	libgomp.so.1 => /usr/lib/x86_64-linux-gnu/libgomp.so.1 (0x00007f07c7211000)
[... cut ...]
~/catkin_ws/devel/lib/rtabmap_ros/rtabmapviz seems to have a problem，I run rtabmapviz by oneself:
```
~/catkin_ws/devel/lib/rtabmap_ros$ ./rtabmapviz 
[ INFO] [1522130332.550452826]: Starting node...
[ INFO] [1522130332.638681098]: rtabmapviz: Using configuration from "/home/turtlebot/.ros/rtabmapGUI.ini"
Segmentation fault (core dumped)
```
so rtabmap.launch, I close rtabmapviz and open rviz, like:
```
 <arg name="rtabmapviz"              default="false" />
 <arg name="rviz"                    default="true" />
```
I can see the picture in real time

matlabbe, 2018-03-27:
For rtabmapviz, it can be a VTK/PCL/Qt compatibility problem. I don't really recommend to build all dependencies from source. There should not be any problems when building rtabmap against default PCL/VTK/OpenCV binaries. If you want to use newer versions of PCL/VTK/OpenCV, I suggest to upgrade Ubuntu to 16.04 at least and ROS Kinetic, you will then get more recent versions by default.

For the original post, "rgbd_odometry" is also crashing, which means that it may be that not the right version of third party libraries are found. Comparing the output of cmake of rtabmap and "ldd" of librtabmap_ros.so can help to see if rtabmap is built with some libraries in /usr/local for example, while librtabmap_ros.so is linked un runtime with some other version of the libraries in /opt/indigo/lib or /usr/lib instead. For example, rtabmap can be built against OpenCV 3 built locally in /usr/local, but librtabmap_ros.so is linked to librtabmap_core.so and cv_bridge at the same time. As cv_bridge is linked to OpenCV2 on Indigo, rtabmap node would seg fault on start because two versions of OpenCV is linked at the same time.

It is then recommend to have only one version of a library installed to avoid these kind of problems.

RajathSwaroop, 2018-03-27:
@matlabbe  I have just one librtabmap_ros.so and it is located at "/home/rajath/catkin_ws/devel/lib". It was a fresh installation with not many things installed on the laptop so I dont believe I have some different versions of libraries conflicting. I had one rtabmap_ros 0.11 version folder in /opt/ros/kinetic/include but I deleted that to check if error gets fixed but I still get the same error.

If I launch the correct camera program and rtabmap launch file I get two errors:

[rtabmap/rtabmapviz-3] process has died [pid 9900, exit code -11, 

and [rtabmap/rgbd_odometry-1] process has died [pid 9898, exit code -11,

matlabbe, 2018-03-27:
When building rtabmap from source, you should uninstall rtabmap and rtabmap_ros binaries (as explained in [build from source](https://github.com/introlab/rtabmap_ros#build-from-source) instructions):
```
$ sudo apt-get remove ros-indigo-rtabmap ros-indigo-rtabmap-ros
```
then rebuild the projects.

RajathSwaroop, 2018-03-27:
Yes I did that. However, I still get the same error.

RajathSwaroop, 2018-04-03:
Got the issue resolved. I did "$ sudo apt-get remove ros-indigo-rtabmap ros-indigo-rtabmap-ros" and also removed every file related to rtabmap standalone and rtabmap. Installed fresh installation of rtabmap from source and rtabmap ros. Finally worked!!!

closed 2018-04-03 by RajathSwaroop

matlabbe, 2020-04-08:
cross-ref, see debugging instructions here: https://github.com/introlab/rtabmap_ros/issues/28