introlab/rtabmap_ros #227: rtabmap_ros 0.16 build error

QinZiwen, 2018-03-26:
when I try to build rtabmap_ros after successfully compiled rtabmap, termial show errors like:
```
/usr/local/lib/librtabmap_core.so: undefined reference to `pcl::search::KdTree
```
I have install ros-indigo, pcl-1.7, qt 4.8.6 and vtk 6.3.0

matlabbe, 2018-03-26:
IS PCL built from source? which full version? Here is a similar post: http://official-rtab-map-forum.67519.x6.nabble.com/Compiling-in-ROS-Indigo-with-PCL-1-7-2-Undefined-reference-to-tp386p397.html

QinZiwen, 2018-03-27:
@matlabbe Thank you. I fix the problem by add find_package(PCL REQUIRED) in CMakeLists.txt
Terminal show error for me, but rtabmap_ros can run
```
CMake Warning at rtabmap_ros/CMakeLists.txt:339 (add_executable):
  Cannot generate a safe runtime search path for target
  pointcloud_to_depthimage because files in some directories may conflict
  with libraries in implicit directories:

    runtime library [libpcl_common.so.1.7] in /usr/lib may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_octree.so.1.7] in /usr/lib may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_io.so.1.7] in /usr/lib may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_kdtree.so.1.7] in /usr/lib may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_search.so.1.7] in /usr/lib may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_sample_consensus.so.1.7] in /usr/lib may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_filters.so.1.7] in /usr/lib may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_features.so.1.7] in /usr/lib may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_keypoints.so.1.7] in /usr/lib may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_segmentation.so.1.7] in /usr/lib may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_visualization.so.1.7] in /usr/lib may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_outofcore.so.1.7] in /usr/lib may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_registration.so.1.7] in /usr/lib may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_recognition.so.1.7] in /usr/lib may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_surface.so.1.7] in /usr/lib may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_people.so.1.7] in /usr/lib may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_tracking.so.1.7] in /usr/lib may be hidden by files in:
      /usr/local/lib
```

matlabbe, 2018-03-27:
You have two versions of PCL, depending on the order of paths in PATH or LD_LIBRARY_PATH, the wrong library could be used on runtime and cause a seg fault. In your case, the right libraries seem taken on runtime, so rtabmap is not crashing. 

Normally, you should not have to add "find_package(PCL REQUIRED)" to rtabmap_ros CMakeLists.txt. rtabmap_ros depends already on pcl_ros, which will set the PCL link libraries to rtabmap_ros. As you have two PCL installed, you have to build pcl_ros from source so that it is linked to PCL in /usr/local instead of the default binaries. **Uninstall PCL binaries**. This will remove a couple of ROS packages depending on it, so you will have to rebuild them from source in your catkin workspace so that they are correctly linked to PCL built from source.