introlab/rtabmap_ros #228: Rtab_map &Mapping...  where can I find the map?

MariRob, 2018-03-26:
Hello ! 
Getting this when I am trying to do the mapping , using **Rtab_map** 
[image]
so my questions are:

1. Does Rtabmap automatically save the map?
-if yes ,where can I find this map?
-if no ,how can I save the map ? 

2. Whenever I start the program, can I start with the last recording or I have to start from the beginning?

matlabbe, 2018-03-26:
Hi,

The map is saved in `~/.ros/rtabmap.db` by default. When you bring-up again rtabmap_ros, you have the choice to start from a past session or create a new one. Start rtabmap node with argument `--delete_db_on_start` to delete `~/.ros/rtabmap.db` and start from a new database. If you don't set `--delete_db_on_start`, rtabmap will load the previous `~/.ros/rtabmap.db` and start a new session in this database. When the camera is relocalized with the previous sessions contained in the database, they should re-appear (the old map is merged with the new one).

A summary:
1. Continue from past database contained in `~/.ros/rtabmap.db`:
    ```bash
    $ roslaunch rtabmap_ros rtabmap.launch
    ```
2. Continue from past database contained in the specified path:
    ```bash
    $ roslaunch rtabmap_ros rtabmap.launch database_path:="my/path/to/database"
    ```
3. Create a new default database `~/.ros/rtabmap.db`:
    ```bash
    $ roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start"
    ```
4. Create a new database at the the specified path:
    ```bash
    $ roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start" database_path:="my/path/to/database"
    ```

By default when reusing a past database, rtabmap will create a new session wherever the camera is localized with a previous map or not. To force rtabmap to start a new session only when it is localized to previous map, set parameter `Rtabmap/StartNewMapOnLoopClosure` to true:

```bash
$ roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--Rtabmap/StartNewMapOnLoopClosure true"
```

cheers,
Mathieu

MariRob, 2018-03-26:
Thanks a lot **Mathieu** 
I will try this..
I really appreciate your help  <3

MariRob, 2018-03-28:
Hi ! i bother you again
In fact Mathieu, I would like to imitate this video
[](https://www.youtube.com/watch?v=Bh8WZsU4YC8)

1. So first of all i create a new database 
> $ roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start"

2. Mapping 

3. ctrl +c 

4. then
>  roslaunch rtabmap_ros rtabmap.launch 

In fact **it doesn't detect the previous map**, but it starts from the beginning and rather shows the previous steps /location of the Camera : 
[image]

any proposals ? 

In addition  the camera detects  objects in motion in it's field of view, but it consider them as a part of the fixed map, normally it should consider them as passing objects during a transient duration and never map them. how can i fix this pb ?? 

[image]
(someone has passed)

matlabbe, 2018-03-28:
Hi,

In the first screenshot, there are multiple paths colored in different colors, that means the current map is linked to previous maps. However, to see the actual data of the previous maps, you should update the cache of the UI (Edit->Download all clouds). This will download the node data from previous nodes. In the video, the cache is updated at 1:57 when the clouds are reappearing. The difference is that with ROS, rtabmap is started automatically and a new map is created, thus updating the cache at the beginning won't load previous nodes. A way to do like the video is to not start the camera yet when starting rtabmap.launch, then download the cache, then start the camera.

For dynamic obstacles, the clouds shown in 3D Map view won't clear them. RTAB-Map doesn't do ray tracing on this 3D cloud visualization. To clear them, the OctoMap can be used instead. In rtabmapviz, OctoMap visualization can be enabled in Preferences->Grid Map Assembling. In rviz, you can show the octomap generated by rtabmap by subscribing to /rtabmap/octomap_full. Build from source [octomap_rviz_plugins](https://github.com/OctoMap/octomap_rviz_plugins) so that the octomap display in rviz is compatible with the octomap sent from rtabmap.

Another approach would be to filter dynamic objects from the depth images before sending them to rtabmap.

cheers,
Mathieu

MariRob, 2018-03-29:
Thank u (y)

ayush-j9, 2019-06-27:
Hi, 

I am trying to map the place around me and then using it again for localization. I am able to do the things that have been shown in the video using the rtabmapviz GUI. But I would like to do it using the launch file.

I made a map and then ran the algorithm again without deleting the previous data base, what happens is it starts mapping again but does not load the previous map. Any solution?

matlabbe, 2019-07-02:
Hi, it should load the previous map (make sure `--delete_db_on_start` or `-d` argument is not set and parameter `Mem/IncrementalMemory` is false). On `rtabmapviz` side, you should refresh the map (Edit->Download all clouds) to download it from `rtabmap` node.

matlabbe, 2022-10-09:
Look at turtlebot tutorial:
http://wiki.ros.org/rtabmap_ros/Tutorials/MappingAndNavigationOnTurtlebot

AngelS18, 2022-11-23:
> Hi,
> 
> The map is saved in `~/.ros/rtabmap.db` by default. When you bring-up again rtabmap_ros, you have the choice to start from a past session or create a new one. Start rtabmap node with argument `--delete_db_on_start` to delete `~/.ros/rtabmap.db` and start from a new database. If you don't set `--delete_db_on_start`, rtabmap will load the previous `~/.ros/rtabmap.db` and start a new session in this database. When the camera is relocalized with the previous sessions contained in the database, they should re-appear (the old map is merged with the new one).
> 
> A summary:
> 
>     1. Continue from past database contained in `~/.ros/rtabmap.db`:
>        ```shell
>        $ roslaunch rtabmap_ros rtabmap.launch
>        ```
> 
>     2. Continue from past database contained in the specified path:
>        ```shell
>        $ roslaunch rtabmap_ros rtabmap.launch database_path:="my/path/to/database"
>        ```
> 
>     3. Create a new default database `~/.ros/rtabmap.db`:
>        ```shell
>        $ roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start"
>        ```
> 
>     4. Create a new database at the the specified path:
>        ```shell
>        $ roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start" database_path:="my/path/to/database"
>        ```
> 
> 
> By default when reusing a past database, rtabmap will create a new session wherever the camera is localized with a previous map or not. To force rtabmap to start a new session only when it is localized to previous map, set parameter `Rtabmap/StartNewMapOnLoopClosure` to true:
> 
> ```shell
> $ roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--Rtabmap/StartNewMapOnLoopClosure true"
> ```
> 
> cheers, Mathieu

Hi, I'm having trouble opening the map, the process I do is once it's saved to .ros/ rtabmap.db, I make a copy of rtabmap, rename it, and open it in rtabmap >> file >> open data base

but the map doesn't load in fact I can't see anything but it doesn't throw me any error either

I hope you can help me, thank you

matlabbe, 2022-11-27:
Can you share your database?

Aliffahmi123, 2022-12-14:
hai, I'm student in Indonesia
i wanna ask abt my project
my project about rtab map use sensor kinect
when i launch
- roslaunch rplidar_ros rplidar.launch
- roslaunch freenect_launch freenect.launch depth_registration:=true
- roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start" rviz:=true rtabmapviz:=false

actually it works, but when i start my robot, it can't draw side way mapping, just one way, can you help me?

matlabbe, 2022-12-15:
Please open a new issue. For help, see:
http://wiki.ros.org/rtabmap_ros/Tutorials/MappingAndNavigationOnTurtlebot
http://wiki.ros.org/rtabmap_ros/Tutorials/SetupOnYourRobot

Aliffahmi123, 2022-12-15:
haloo matlabbe
i have a problem when i was launch kinect and rtab map,
map can open, but it was draw only one way, but if turn left or right, it
didn't draw map, what should i do?:(

Aliffahmi123, 2022-12-15:
i men like this, the robot only describes something that is in front, but
if it turns left and right, it doesn't describe anything anymore