introlab/rtabmap_ros #229: Mapping and Navigation with Turtlebot with Kinect2

rui-cao, 2018-04-03:
Hi,

I am following your tutorial on Mapping and Navigation with Turtlebot ( http://wiki.ros.org/rtabmap_ros/Tutorials/MappingAndNavigationOnTurtlebot).

I have problem to make the tutrtlebot to work with the Kinect2. The reason we found out is that turtlebot does not have 3d_sensor launch file to Kinect2 driver (Freenect2 and iai_kinect/kinect2_bridge). I tried to modify the original Freenect (driver to the older Kinect) launch file which is already in the turtlebot, to make it compatible with the Kinect2, by pointing to  kinect2_bridge.launch, and modified the arguments to make it compatible. However it does not work. If you are interested, more detail can be viewed from my wiki page (https://github.com/rui-cao/SLAM_Project/wiki/Simulation-with-Turtlebot).

While not using the turtlebot package, your package was working perfectly with Kinect2 on my computer, both in standalone and the ros version. I am using ROS Kinetic.

Thank you for your time!

matlabbe, 2018-04-03:
You will have to modify [3dsensor.launch](https://github.com/turtlebot/turtlebot/blob/kinetic/turtlebot_bringup/launch/3dsensor.launch) too. It calls kinect.launch.xml with some arguments that it doesn't need anymore in your modified version, it is why you have `unused args` error. You would also need to remap the depth image topic sent to `depthimage_to_laserscan`. 

You could also make a copy of 3dsensor.launch called kinect2sensor.launch for example, launching directly kinect2 inside and remap the correct topics/frames of depthimage_to_laserscan. Then replace 3dsensor.launch in [demo_turtlebot_mapping.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/demo/demo_turtlebot_mapping.launch) to kinect2sensor.launch. You may have to remap camera topics in [demo_turtlebot_mapping.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/demo/demo_turtlebot_mapping.launch) too.

You may have to verify if in the included move_base configuration, it uses topic from the camera directly. I think it uses only the fake scan generated but just to verify.

cheers,
Mathieu

rui-cao, 2018-04-05:
Hi,

thank you for the really helpful comment. It solves the problem that I have previously. Now the Kinect2 can communicate with the Turtlebot. However new problem comes....

I get this error after excuting `roslaunch rtabmap_ros demo_turtlebot_mapping.launch`
[image]

I don't know why the the dimension of the RGB image has to be integer multiple of the IR image ([rgb=960x540 depth=512x424]). With the same image resolution ratio, I could make the RTABmap standalone package to work. However, here it does not work.

While ignoring this error, I get the following warning:
[image]

I did checked the topics that are published, by runing `rostopic list`:

/camera/bond
/camera/hd/camera_info
/camera/hd/image_color
/camera/hd/image_color/compressed
/camera/hd/image_color_rect
/camera/hd/image_color_rect/compressed
/camera/hd/image_depth_rect
/camera/hd/image_depth_rect/compressed
/camera/hd/image_mono
/camera/hd/image_mono/compressed
/camera/hd/image_mono_rect
/camera/hd/image_mono_rect/compressed
/camera/hd/points
/camera/qhd/camera_info
/camera/qhd/image_color
/camera/qhd/image_color/compressed
/camera/qhd/image_color_rect
/camera/qhd/image_color_rect/compressed
/camera/qhd/image_depth_rect
/camera/qhd/image_depth_rect/compressed
/camera/qhd/image_mono
/camera/qhd/image_mono/compressed
/camera/qhd/image_mono_rect
/camera/qhd/image_mono_rect/compressed
/camera/qhd/points
/camera/sd/camera_info
/camera/sd/image_color_rect
/camera/sd/image_color_rect/compressed
/camera/sd/image_depth
/camera/sd/image_depth/compressed
/camera/sd/image_depth_rect
/camera/sd/image_depth_rect/compressed
/camera/sd/image_ir
/camera/sd/image_ir/compressed
/camera/sd/image_ir_rect
/camera/sd/image_ir_rect/compressed
/camera/sd/points
/map
/move_base/current_goal
/move_base/goal
/move_base_simple/goal
/navigation_velocity_smoother/raw_cmd_vel
/rosout
/rosout_agg
/rtabmap/cloud_ground
/rtabmap/cloud_map
/rtabmap/cloud_obstacles
/rtabmap/global_path
/rtabmap/global_pose
/rtabmap/goal
/rtabmap/goal_node
/rtabmap/goal_out
/rtabmap/goal_reached
/rtabmap/info
/rtabmap/labels
/rtabmap/local_path
/rtabmap/mapData
/rtabmap/mapGraph
/rtabmap/mapPath
/rtabmap/move_base/cancel
/rtabmap/move_base/feedback
/rtabmap/move_base/goal
/rtabmap/move_base/result
/rtabmap/move_base/status
/rtabmap/octomap_binary
/rtabmap/octomap_empty_space
/rtabmap/octomap_full
/rtabmap/octomap_grid
/rtabmap/octomap_occupied_space
/rtabmap/odom
/rtabmap/odom_info
/rtabmap/odom_last_frame
/rtabmap/odom_local_map
/rtabmap/odom_local_scan_map
/rtabmap/proj_map
/rtabmap/scan_map
/rtabmap/user_data_async
/scan
/tf
/tf_static

So the required topics are indeed published. I don't know what the error is...

This is the result in the visualisation in RVIZ (basically blank):
[image]

I also attached my modified launch files for the `demo_turtlebot_mapping.launch` and `kinect2sensor.launch` in the zip file, in case you want to check the settings:
[launchFiles.zip](https://github.com/introlab/rtabmap_ros/files/1878278/launchFiles.zip)

Again, thank you very much for your time!

matlabbe, 2018-04-05:
The assert makes rtabmap node crashing, so rtabmap is not running anymore. rtabmap needs registered depth image to RGB camera, which in general RGB and depth has the same size. You should use all "qhd" topics or all "sd" topics, not a mix of both. With kinect v2, it is preferred to use all "qhd" topics to have better 3D point cloud reconstruction without black borders.

cheers,
Mathieu

rui-cao, 2018-04-05:
Hi,

thank you again for your generous help. Originally I thought the IR image is maximum only up to sd quality, so I made the RGB image to qhd and IR iamge to sd. Anyway, now after I change every image to qhd, the problem goes. However, I got new problem:

[image]

For the first warning from the image, I checked each topic with `rostopic hz ...` and found out that, only the `/scan` topic is not publishing any data, all other topics are publishing at normal rate. I did map the `kinect2/qhd/image_depth_rect` to `depthimage_to_laserscan` in my `kinect2sensor.launch` file. However, either I am doing it incorrectly or there are some error elsewhere that prohibits the publish of the /scan topic.

I did some changes to my last launch files, so I attach it here in the zip file for your reference.
[launch.zip](https://github.com/introlab/rtabmap_ros/files/1881851/launch.zip)

The second warning that I get is that the transform from base_footprint to map is not available. I have no clue on what to do for this warning...

Here I attach the TF frames:
[frames.pdf](https://github.com/introlab/rtabmap_ros/files/1881841/frames.pdf)

Thank you for your time and help!

matlabbe, 2018-04-06:
From this file:
```xml
<launch>
  <!-- "camera" should uniquely identify the device. All topics are pushed down
       into the "camera" namespace, and it is prepended to tf frame ids. -->
  <arg name="camera"      default="kinect2"/>
  <arg name="publish_tf"  default="true"/>
 <!-- <arg name="jpeg_quality"  default="50"/>-->
 <arg name="depth_method"  default="opengl"/>
  <arg name="scan_processing"  default="true"/>
  <arg name="depth_registration"              default="true"/>
  <arg     if="$(arg depth_registration)" name="depth" value="depth_registered" />
  <arg unless="$(arg depth_registration)" name="depth" value="depth" />

  <!-- Laserscan topic -->
  <arg name="scan_topic" default="scan"/>

<include file="/home/rui/catkin_ws/src/iai_kinect2/kinect2_bridge/launch/kinect2_bridge.launch">
<arg name="base_name"                      value="$(arg camera)"/>                         
    <arg name="publish_tf"                      value="$(arg publish_tf)"/>
  <!-- <arg name="jpeg_quality"                      value="$(arg jpeg_quality)"/>-->
<arg name="depth_method"                      value="$(arg depth_method)"/>

</include>

  <remap from="qhd/image_depth_rect" to="depthimage_to_laserscan"/>

   <!--                        Laserscan 
     This uses lazy subscribing, so will not activate until scan is requested.
   -->
  <group if="$(arg scan_processing)">
    <node pkg="nodelet" type="nodelet" name="depthimage_to_laserscan" args="load depthimage_to_laserscan/DepthImageToLaserScanNodelet $(arg camera)/$(arg camera)_nodelet_manager">
      <!-- Pixel rows to use to generate the laserscan. For each column, the scan will
           return the minimum value for those pixels centered vertically in the image. -->
      <param name="scan_height" value="10"/>
      <param name="output_frame_id" value="/$(arg camera)_depth_frame"/>
      <param name="range_min" value="0.45"/>
      <remap from="image" to="$(arg camera)/$(arg depth)/image_raw"/>
      <remap from="scan" to="$(arg scan_topic)"/>

      <!-- Somehow topics here get prefixed by "$(arg camera)" when not inside an app namespace,
           so in this case "$(arg scan_topic)" must provide an absolute topic name (issue #88).
           Probably is a bug in the nodelet manager: https://github.com/ros/nodelet_core/issues/7 -->
      <remap from="$(arg camera)/image" to="$(arg camera)/$(arg depth)/image_raw"/>
      <remap from="$(arg camera)/scan" to="$(arg scan_topic)"/>
    </node>
  </group>
</launch>
```
Review the topics and parameters set to depthimage_to_laserscan for kinect2 topics and frames. For example, it may look like this:
```xml
<node pkg="nodelet" type="nodelet" name="depthimage_to_laserscan" args="standalone depthimage_to_laserscan/DepthImageToLaserScanNodelet">
  <param name="scan_height" value="10"/>
  <param name="output_frame_id" value="kinect2_rgb_optical_frame"/>
  <param name="range_min" value="0.45"/>
  <remap from="image" to="/kinect2/qhd/image_depth_rect"/>
  <remap from="scan" to="/scan"/>
</node>
```
I set also a standlaone nodelet as I am not sure what was the name of kinect2 nodelet manager. You may load the nodelet on the kinect2 nodelet manager to save some serialization overhead.

cheers,
Mathieu

rui-cao, 2018-04-06:
Hi again,

thank you very much, the problem is solved!

However, a new problem comes up. Now I can do the mapping and see the point cloud and costmap in Rviz. However, the point cloud and the cost map and the laser scan are not aligned, as seen in this picture:
[image]
The yellow circled thing is the fake 2d laser scan (I believe). Notice it is vertical to the cost map. On the other hand, the point cloud is also about 90° in the yaw axis from the cost map. How can I align the three?

Very thank you for your time!

matlabbe, 2018-04-06:
For the scan, maybe it is related to output_frame_id. You may want to show TF in RVIZ to debug TF.

What is your fixed frame in global options of RVIZ? it should be /map to correctly see the data. Maybe there is a TF rotation missing somewhere.

rui-cao, 2018-04-07:
Hi,

thank you for the reply.
 
I did check the TF in Rviz. In the image I applied a rotation to the kinect2link such that the point cloud is on the ground, but the scan is vertical to the ground, like this.
[image]

This is the rotation that I defined for the kinect2link.
```xml
  <arg name="pi/2" value="0"/>
   <arg name="optical_rotate" value="0 0 0 -$(arg pi/2) 0 -$(arg pi/2)" />
   <node pkg="tf" type="static_transform_publisher" name="base_link1"
args="$(arg optical_rotate) base_link kinect2_link 100" /> 
```

When I do not apply the rotation, the point cloud will be floating in the air above the robot, but the scan will be on the ground and looks normal:
[image]

Since the scan is generated from the kinect2_rgb_optical_frame, it is thus coupled with the kinect2 frame. If I rotate the kinect2, then the scan will also be rotated. Thus, if I make one to be in the normal orientation, the other will be in wrong orientation. If I make one on the ground, the other one will be vertical to the ground or above the robot.

I tried to rotate the scan topic by
```xml
   <node pkg="tf" type="static_transform_publisher" name="base_link1"
args="$(arg optical_rotate) base_link kinect2_link 100" />
```
but the scan did not get rotated in the Rviz.

This is my current TF tree:
[frames.pdf](https://github.com/introlab/rtabmap_ros/files/1886809/frames.pdf)

Maybe you have more insight in how to solve such problem?

Thank you for your time!

matlabbe, 2018-04-09:
Adjust the `<param name="output_frame_id" value="kinect2_rgb_optical_frame"/>` of the scan so that it matches the point cloud orientation. Maybe it should be base_link or kinect2_link.

rui-cao, 2018-04-13:
Hi,

thank you for the useful comment. It is working! Now I have a problem is the navigation mode where the costmap cannot be constructed. The warning is like this:
[image]
[image]

We tried to set `wait_for_transform_duration ` to 0.2 on the command line as `roslaunch rtabmap_ros demo_turtlebot_mapping.launch localization:=true wait_for_transform_duration:=0.2
`but we get the same warning as the `wait_for_transform_duration ` parameter is still 0.1. So how can I change the `wait_for_transform_duration `?

And is the timeout the cause of the problem for the costmap not able to be constructed?

We also find out that our `odom` topic is empty:
[image]

Is that also a problem? And how to solve it?

Thank you for your time and help!

matlabbe, 2018-04-13:
You may use [rqt_console](http://wiki.ros.org/rqt_console) to know exactly from which node the warning message is coming. I think it is not from rtabmap, but from move_base's costmap update. For move_base, odom should be published at least on TF (/odom -> /base_footprint). rtabmap won't publish /map->/odom if it is not receiving odom either.

matlabbe, 2018-04-25:
**/mobile_base/commands/velocity** is the twist sent by move_base:
```
$ rostopic type /mobile_base/commands/velocity
geometry_msgs/Twist
```

din1369, 2018-06-14:
Hi, Thank you guys for all the help. i also able to fix lots of issues from this thread. 
@rui-cao can you upload latest launch files. Im still having an error 

[image]

din1369, 2018-06-15:
Hi I'm having this error now. I did All the changes mentioned here.
[image]

matlabbe, 2018-06-15:
Can you show the TF tree?
```bash
$ rosrun tf view_frames
$ evince frames.pdf
```
It seems there is a missing static_transform_publisher between base_footprint and kinect frame.

Cheers,
Mathieu

din1369, 2018-06-18:
Hi, thank you very much for respond. Here is my launch files and frames.pdf
[frames.pdf](https://github.com/introlab/rtabmap_ros/files/2109857/frames.pdf)
[my.tar.gz](https://github.com/introlab/rtabmap_ros/files/2109858/my.tar.gz)

matlabbe, 2018-06-18:
You don't have `base_footprint` frame. Set `base_link` instead as `frame_id` in the demo_turtlebot_mapping.launch file. For the `depthimage_to_laserscan` error, you may ask on ROS answers.

cheers,
Mathieu

luohuiwu, 2018-11-14:
> Hi,
> 
> thank you for the useful comment. It is working! Now I have a problem in the navigation mode where the costmap cannot be constructed. The warning is like this:
> [image]
> [image]
> 
> We tried to set `wait_for_transform_duration ` to 0.2 on the command line as `roslaunch rtabmap_ros demo_turtlebot_mapping.launch localization:=true wait_for_transform_duration:=0.2 `but we get the same warning as the `timeout `is still 0.1. So how can I change the `timeout `?
> 
> Do you think solving timeout will fix the costmap problem?
> 
> We also find out that our `odom` topic is empty:
> [image]
> 
> Is that also a problem? And how to solve it?
> 
> Thank you for your time and help!

Hi, rui-cao @rui-cao, I met the same problem. Can you share your launch file for the community? Many thanks. For convenience, I open an issue [Autonomous Navigation with Turtlebot with Kinect2](https://github.com/introlab/rtabmap_ros/issues/286) on the autonomous task with RTAB-Map using Kinect 2. @rui-cao @matlabbe

ZzhKlaus, 2019-04-17:
@matlabbe Sorry to interrupt you but I am new to ROS and got into a big problem. I am running the rtabmap_ros package and trying to add the "depthimage_to_laserscan" node, the published topic /kinect_scan could be shown in rqt_graph and everthing seems ok, but the /kinect_scan of message type sensor_msgs/LaserScan has all nan values in ranges[] and empty intensities[], can you help me?

ZzhKlaus, 2019-04-17:
@matlabbe  And here is my rtabmap.launch file.

<launch>
  

  
           
  
     
        
  <arg name="stereo"          default="false"/>
 
  
  <arg name="rtabmapviz"              default="true" /> 
  <arg name="rviz"                    default="false" />
  
  
  <arg name="localization"            default="false"/>
  
  
  <arg name="use_sim_time"            default="false"/>
  <param if="$(arg use_sim_time)" name="use_sim_time" value="true"/>
  
  
  <arg name="cfg"                     default="" /> 
  <arg name="gui_cfg"                 default="~/.ros/rtabmap_gui.ini" />
  <arg name="rviz_cfg"                default="$(find rtabmap_ros)/launch/config/rgbd.rviz" />
  
  <arg name="frame_id"                default="camera_link"/>     
  <arg name="odom_frame_id"           default=""/>                
  <arg name="map_frame_id"            default="map"/>
  <arg name="ground_truth_frame_id"   default=""/>     
  <arg name="ground_truth_base_frame_id" default=""/>  
  <arg name="namespace"               default="rtabmap"/>
  <arg name="database_path"           default="~/.ros/rtabmap.db"/>
  <arg name="queue_size"              default="10"/>
  <arg name="wait_for_transform"      default="0.2"/>
  <arg name="args"                    default=""/>              
  <arg name="rtabmap_args"            default="$(arg args)"/>   
  <arg name="launch_prefix"           default=""/>              
  <arg name="output"                  default="screen"/>        
  <arg name="publish_tf_map"          default="true"/>

  
  <arg     if="$(arg stereo)" name="approx_sync"  default="false"/>
  <arg unless="$(arg stereo)" name="approx_sync"  default="true"/>         
   
  
  <arg name="rgb_topic"               default="/camera/rgb/image_rect_color" />
  <arg name="depth_topic"             default="/camera/depth_registered/image_raw" />
  <arg name="camera_info_topic"       default="/camera/rgb/camera_info" />
  <arg name="depth_camera_info_topic" default="$(arg camera_info_topic)" />
  
  
  <arg name="stereo_namespace"        default="/stereo_camera"/>
  <arg name="left_image_topic"        default="$(arg stereo_namespace)/left/image_rect_color" />
  <arg name="right_image_topic"       default="$(arg stereo_namespace)/right/image_rect" />      
  <arg name="left_camera_info_topic"  default="$(arg stereo_namespace)/left/camera_info" />
  <arg name="right_camera_info_topic" default="$(arg stereo_namespace)/right/camera_info" />
  
  
  <arg name="rgbd_sync"               default="false"/>         
  <arg name="approx_rgbd_sync"        default="true"/>          
  <arg name="subscribe_rgbd"          default="$(arg rgbd_sync)"/>
  <arg name="rgbd_topic"              default="rgbd_image" />
  <arg name="depth_scale"             default="1.0" />
  
  <arg name="compressed"              default="false"/>         
  <arg name="rgb_image_transport"     default="compressed"/>    
  <arg name="depth_image_transport"   default="compressedDepth"/>  
[... 262 lines cut ...]
      <remap from="scan_cloud"             to="$(arg scan_cloud_topic)"/>
      <remap from="odom"                   to="$(arg odom_topic)"/>
    </node>
  
  </group>
  
  
  <node if="$(arg rviz)" pkg="rviz" type="rviz" name="rviz" args="-d $(arg rviz_cfg)"/>
  <node if="$(arg rviz)" pkg="nodelet" type="nodelet" name="points_xyzrgb" args="standalone rtabmap_ros/point_cloud_xyzrgb" output="$(arg output)">
    <remap if="$(arg stereo)" from="left/image"        to="$(arg left_image_topic_relay)"/>
    <remap if="$(arg stereo)" from="right/image"       to="$(arg right_image_topic_relay)"/>
    <remap if="$(arg stereo)" from="left/camera_info"  to="$(arg left_camera_info_topic)"/>
    <remap if="$(arg stereo)" from="right/camera_info" to="$(arg right_camera_info_topic)"/>
    <remap unless="$(arg subscribe_rgbd)" from="rgb/image"         to="$(arg rgb_topic_relay)"/>
    <remap unless="$(arg subscribe_rgbd)" from="depth/image"       to="$(arg depth_topic_relay)"/>
    <remap unless="$(arg subscribe_rgbd)" from="rgb/camera_info"   to="$(arg camera_info_topic)"/>
    <remap from="rgbd_image"        to="$(arg rgbd_topic_relay)"/>
    <remap from="cloud"             to="voxel_cloud" />

    <param name="decimation"  type="double" value="4"/>
    <param name="voxel_size"  type="double" value="0.0"/>
    <param name="approx_sync" type="bool"   value="$(arg approx_sync)"/>
  </node>

</launch>

matlabbe, 2019-04-24:
> but the /kinect_scan of message type sensor_msgs/LaserScan has all nan values in ranges[] and empty intensities[]

Are there obstacles in from of the camera? Can you post a screenshot of the depth image?