introlab/rtabmap_ros #234: Mapping using rtabmap viz and localization using demo_turtlebot_mapping.launch, rviz

RajathSwaroop, 2018-04-06:
Hi,

I am developing a robot that need the camera to have 3DOF. The demo_turtlebot_mapping was developed for a robot that has camera fixed to its body. I wish to move the camera using a gimbal while mapping and thus prefer rtabmap viz for mapping.  However, when I recorded the map using rtabmapviz or rviz directly without using the demo_turtlebot_mapping launch file, I am not able to get the map in localization model. Is it not possible to perform handheld mapping and use demo_turtlebot_mapping to drive or navigate the robot in localization mode?

matlabbe, 2018-04-06:
Yes, it is possible, but TF of your gimbal should be updated to make it work. For example, the turtlebot demo uses wheel odometry, the frame_id is the base_link or base_footprint of the robot. The images from camera are converted to frame_id using TF. If the camera rotates on the robot **and** if TF between frame_id and the camera link is updated accordingly, there should not be any problems. 

To use visual odometry while the camera can rotate on the robot, the frame_id of the odometry should still be the base_link or base_footprint frame, and if TF between the base and camera is also updated accordingly to movements of the camera, it should work too.

cheers,
Mathieu

RajathSwaroop, 2018-04-06:
When I tried recording a map using rtabmap and use demo_turtlebot_mapping in localization mode to navigate, I was not able to get the 2-D gird map even after 10 minutes of trying to localize. I did provide any roll or pitch to the camera while recording the rtabmap or localization. I used the same db file recorded using rtabmap to localize the robot system using demo_turtlebot_mapping launch file.

matlabbe, 2018-04-06:
If the environment has visual features, localization would be almost instantaneous.

Are you trying to create a map with hand-held device, then copy the database on the robot so it can localize in it? Is the same sensor used? If the point of view of the robot is not the same than when the map was created, it will be difficult to localize.

RajathSwaroop, 2018-04-06:
My environment is a room with a lot of features. When I mapped using demo_turtlebot_mapping launch file it localizes instantaneously or worst case 20 seconds. 

I tried mapping using rtabmapviz using hand held mapping example while the camera was on a gimbal and I did not provide any roll or pitch. The same sensor Intel D415 camera was then used to localize after I copied the database file to the required location and robot, camera's point of view was same. position or orientation of the camera was not changed.

matlabbe, 2018-04-06:
The only differences that could influence localization between rtabmap.launch and demo_turtlebot_mapping.launch is that in the later, these parameters are set:
```xml
<param name="Vis/MinInliers"               type="string" value="15"/>      <!-- 3D visual words minimum inliers to accept loop closure -->
<param name="Vis/InlierDistance"           type="string" value="0.1"/>    <!-- 3D visual words correspondence distance -->
```

Why not doing both mapping and localization with demo_turtlebot_mapping.launch then?

matlabbe, 2018-04-09:
Why not having two cameras? one for SLAM on the robot and other on your arm for other stuff you need to do.

If TF between the base and camera is always updated, you may always use the 6DoF configuration of the held tutorial for both mapping and localization.

RajathSwaroop, 2018-04-11:
I need one camera on the arm with 3DOF to perform mapping while the camera has tilt, pan motions (similar to handheld mapping) and the same camera to use the recorded map to perform localization and run visual slam based navigation. For this I would have to use RTABmap hand held example and map with a 3DOF arm and use turtlebot_demo_mapping launch file to localize and navigate.

matlabbe, 2018-04-12:
[turtlebot_demo_mapping](http://wiki.ros.org/rtabmap_ros/Tutorials/MappingAndNavigationOnTurtlebot) uses wheel odometry and does mapping in 2D. In your case, re-uses the handheld configuration to navigate. You may relaunch the handheld mode in 2D if you want.

Mapping (assuming default openni camera topics):
`
$ roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start" frame_id:=base_footprint
`

Relaunch for localization/navigation in 2D:
`
$ roslaunch navigation.launch
`
where `navigation.launch`:
```xml
<launch>
   <include file="$(find turtlebot_navigation)/launch/includes/move_base.launch.xml"/>
   <remap from="/rtabmap/grid_map" to="/map"/>
   <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
      <arg name="rtabmap_args" value="--delete_db_on_start --Reg/Force3DoF true"/>
      <arg name="frame_id" value="base_footprint"/>
      <arg name="localization" value="true"/>
   </include>
</launch>
```

If you would want to use your robot odometry, that could be look to this instead (disable visual odometry and subscribe to robot /odom):
```xml
<launch>
   <include file="$(find turtlebot_navigation)/launch/includes/move_base.launch.xml"/>
   <remap from="/rtabmap/grid_map" to="/map"/>
   <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
      <arg name="rtabmap_args" value="--delete_db_on_start --Reg/Force3DoF true"/>
      <arg name="frame_id" value="base_footprint"/>
      <arg name="localization" value="true"/>
      <arg name="visual_odometry" value="false"/>
      <arg name="odom_topic" value="/odom"/>
   </include>
</launch>
```

RajathSwaroop, 2018-04-12:
I tried "roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start" frame_id:=base_footprint" and then used it to localize the robot, camera in rviz using the above mentioned launch file and it worked. Thanks.

Also when I record a map using rtabmaprviz and perform postprocessing, where does it save the database if I save using "download all cloud"? I tried performing post processing and saving the cloud but I could notice that db file in the .ros directory was not updated. Even when I moved the file from .ros to a different location and downloading the cloud It did not generate a db file in the .ros directory.

matlabbe, 2018-04-12:
Post-processing in the rtabmapviz doesn't affect the database, just the visualization on rtabmapviz side. They are not very useful for ROS, mainly used with mapping with the standalone app. However, if you open the resulting database in rtabmap-databaseViewer, the changes using actions under Edit menu will be saved when closing the database. For example, like post-processing, you can click on Edit-> Detect more loop closures to add more constraints to graph, then when closing DatabaseViewer, they will be saved in the database. The robot can then use that database to have the changes.

RajathSwaroop, 2018-04-13:
Is it possible to do multi-session mapping using rtabmap-ros. Also I do not get access to edit the default settings in windows->preferences->source when I try to do it in rtabmap ros but I have access to it in the rtabmap standalone. Is it not allowed in ROS?

matlabbe, 2018-04-13:
Yes, you can do multi-session mapping. When relaunching in mapping mode, remove "--delete_db_on_start" argument so that the default database (~/.ros/rtabmap.db) will not be cleared. When a loop closure or localization will happen, the previous map will be merged with the new one.

In ROS, the source panel is disabled because the camera is started from its own ROS node, external to rtabmap. 

cheers,
Mathieu

RajathSwaroop, 2018-04-16:
Yes that worked for me. Great. Thanks

closed 2018-04-16 by matlabbe