introlab/rtabmap_ros #235: How do I convert a recorded database file into a mesh with tsdf setting

RajathSwaroop, 2018-04-16:
Hi, I have been using rtabmap to develop 3-D reconstruction of my room and found it to be great. However I want to visualize it as a mesh and I found that the mesh reconstruction of the map loses resolution. I would like to know if it is possible to use tsdf setting in meshing to improve the mesh file resolution. 

When I try selecting tsdf in the list it does not show me the option.

matlabbe, 2018-04-16:
For Poisson reconstruction, increase the depth to get more polygons (try 9 or 10). Note however it will be more sensible to noise. 

TSDF options are only available if rtabmap is built with [cpu-tsdf](https://github.com/sdmiller/cpu_tsdf) library or [open_chisel](https://github.com/personalrobotics/OpenChisel).

cheers,
Mathieu

RajathSwaroop, 2018-04-16:
How do I build the rtabmap with cpu-tsdf? the cmakelist.txt has cpu-tsdf "on". will that be enough or any other flags to be set?

matlabbe, 2018-04-16:
You have to download [cpu-tsdf](https://github.com/sdmiller/cpu_tsdf), build/install it so that cmake can find it. In the cmake status of rtabmap it will be shown if cpu has been found. Example:
```
-- --------------------------------------------
-- Info :
--   Version : 0.17.0
--   CMAKE_INSTALL_PREFIX = /home/mathieu/catkin_ws/devel
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp -std=c++11
--   PCL_DEFINITIONS =  -march=native -msse4.2 -mfpmath=sse ;-DDISABLE_OPENNI2;-DDISABLE_PCAP;-DDISABLE_PNG;-DDISABLE_LIBUSB_1_0
--   With OpenCV 3 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = YES (License: Non commercial)
--   With Freenect             = YES (License: Apache v2 and/or GPLv2)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = YES (License: Apache v2 and/or GPLv2)
--   With Kinect for Windows 2 = NO (Kinect for Windows 2 SDK not found)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--   With g2o                  = YES (License: BSD)
--   With GTSAM                = YES (License: BSD)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = NO (cvsba not found)
--   With libpointmatcher      = YES (License: BSD)
--   With ZED                  = YES (With CUDA)
--   With RealSense            = YES (License: Apache-2)
--   With RealSenseSlam        = YES
--   With OCTOMAP              = YES (License: BSD)
--   With CPUTSDF              = YES (License: BSD)
--   With OpenChisel           = YES (License: ???)
--   With libfovis             = YES (License: GPLv2)
--   With libviso2             = YES (License: GPLv3)
--   With dvo_core             = YES (License: GPLv3)
--   With okvis                = YES (License: BSD)
--   With ORB_SLAM2            = NO (WITH_G2O should be OFF as ORB_SLAM2 uses its own g2o version)
--   With Qt5                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
-- Configuring done

```

RajathSwaroop, 2018-04-18:
I will try that. Also here is a preview of my mapping session. The RTABmap fails to do a loop closure some times and I get a very bad disoriented map.. Can you please tell me if there is a setting to optimize it

matlabbe, 2018-04-19:
Which version are you using? Is it with default settings? Is it built with g2o?

You may check with help of the Constraints View in rtabmap-databaseViewer if there is a neighbor link (i.e., odometry link) that has a very gross error. Then look what could have happened, like was there a fast movement of the camera, there were not so many features to track...

If loop closures that should be good are rejected, you could increase RGBD/OptimizeMaxError parameter to accept them.

RajathSwaroop, 2018-04-19:
@matlabbe  I am using the latest 0.17 version of rtabmap. I have g2o installed and in the CMakelists.txt I have set it to install with g2o.  There are enough features to track. I face this issue when I try to record a large room of size 8m x 6m and come back to the same point after recording other parts of the room. I tried changing the threshold setting for similarity threshold and loop closure threshold but nothing changed. I will try to increase the RGBD/OptimizeMaxError

RajathSwaroop, 2018-04-20:
@matlabbe I want to perform a sensor fusion for getting accurate Odometry. If I use a kalman filter algorithm on my sensor data that include (visual odometry), IMU data and feed this back under odom topic to the RTABMAP will it perform better. Will RTABmap accept external odometry while computing its own visual odometry at the same time.

RajathSwaroop, 2018-04-25:
Also when I record a map and run the rtabmap in localization mode after completion of mapping session, I am able to download the point cloud of the map into the rtabmapviz and see the odometry traces of previous sessions. However, the current /rtabmap/odom gives 0,0 odometry for the reference point where I started localization mode and tends to assume that point as origin instead of finding its co-ordinates from the built map. Can you please let me know if there is any setting to get the current location of camera from the previous map??

matlabbe, 2018-04-27:
Yes, rtabmap can use external odometry. Make sure that if rgbd_odometry (or stereo_odometry) is used, it doesn't publish TF (`publish_tf`=false) to avoid conflicts in the TF tree. See [these](https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/sensor_fusion.launch#L28-L36) visual odometry parameters in the sensor fusion example. Then make rtabmap subscribe on your odometry topic (or set `odom_frame_id` to use TF of your odometry) instead of the one published by visual odometry. 

cheers,
Mathieu

RajathSwaroop, 2018-04-27:
The sensor fusion example takes the IMU data and outputs visual odometry. Instead if I want to provide my own odometry to the rtabmap while also getting its visual odometry is it possible? I want to fuse my odometry and visual odometry so the rtabmap visual odometry does not get lost when the camera encounters featureless space. Is that possible??

matlabbe, 2018-04-28:
Yes it is possible. Just connect your odometry to input odometry of rtabmap. This is what is done in this [example](http://wiki.ros.org/rtabmap_ros/Tutorials/SetupOnYourRobot#Kinect_.2B-_Odometry_.2B-_2D_laser), using odometry from the robot instead of visual odometry:
```xml
<remap from="odom" to="/base_controller/odom"/>
```

"dynamic obstacle avoidance" is done by move_base, see this [page](http://wiki.ros.org/navigation/Tutorials/RobotSetup#Costmap_Configuration_.28local_costmap.29_.26_.28global_costmap.29). If you means how obstacles are added to occupancy grid map generated by rtabmap, see `Grid/` parameters:
```
$ rtabmap --params | grep Grid/
```

matlabbe, 2018-04-30:
If 0.17 libraries are still installed, rtabmap_ros may link into them instead of 0.16.3. Make sure to remove all 0.17 libraries and headers. That parameter is in 0.16.3 [Parameters.h](https://github.com/introlab/rtabmap/blob/0.16.3/corelib/include/rtabmap/core/Parameters.h#L625).

RajathSwaroop, 2018-05-02:
I tried that and It worked. Thanx.

When I record a map setting transforms between the basefootprint and the camera frame for navigation and try to localize it in the rviz visualizer, I noticed that the origin (0,0) is changing everytime I localize. 

Example first time if I localize near table 1 I get space around table 1 to be 0,0 and next time when I localize around table 2 I get that as 0,0.  Does rtabmap mapping and localizing algorithm assume a world co-ordinate system?? How do I rectify it

matlabbe, 2018-05-03:
/map -> /odom is published [here](https://github.com/introlab/rtabmap_ros/blob/2c64849807256111c77abfe0ce712da4008a6f1a/src/CoreWrapper.cpp#L724-L730). You should not have to modify the code, as you can use simply a [static_transform_publisher](http://wiki.ros.org/tf#static_transform_publisher) to publish TF /world -> /map. Then if you would want robot pose in world frame, you would get /world -> /base_link.

cheers,
Mathieu

RajathSwaroop, 2018-05-03:
Ok that would be an easier workaround. Thanx. Is there any topic published by rtabmap that indicates if the robot has successfully localized in the given map??

matlabbe, 2018-05-03:
[rtabmap_ros/Info](http://docs.ros.org/api/rtabmap_ros/html/msg/Info.html) msg: 
```
$ rostopic echo /rtabmap/info/loopClosureId
```
localized if `loopClosureId` is not 0.

RajathSwaroop, 2018-05-03:
Got it. Thanx

RajathSwaroop, 2018-05-07:
When I try localizing the robot in a known map recorded using base footprint and frame id and providing the transform between camera frame and base footprint, the /rtabmap/info/loopClosureId topic sends a 0 after first 5 or 10 seconds of start but I do not get the map in rviz for quite long, sometimes I dont get it forever. Are these two things inter-related? a success in localizing should load the map in rviz or it is a different sequence??

RajathSwaroop, 2018-05-07:
Also I recorded a map using robot 1 providing the transform between the robot base footprint and the camera frame, then I use that map to localize robot 2 while providing the correct transform between base footprint of the robot 2 and camera frame and the camera used in both the cases is same intel realsense. Robot 2 is not able to localize even after too many attempts. Is there a reason behind this?

matlabbe, 2018-05-08:
If the robots are not identical and the camera is not placed at the same position (angle, height) than on the other robot creating the map, the point of view may cause some challenge for localization. To be more robust to point of view changes, use SURF or SIFT features for the loop closure vocabulary (Kp/DetectorStrategy=0 or 1), and even for visual motion estimation (Vis/FeatureType=0 or 1). Look at the terminal for warnings about if a localization is rejected after valid loop closure detection.

For the previous question, the map 2D occupancy grid map should be republished **after the second robot is localized**.

cheers,
Mathieu

RajathSwaroop, 2018-05-08:
Ok I will try those options mentioned above. 

What are loop closure threshold and similarity threshold in rtabmap. Changing the value of loop closure improves loop closure in any way?

what is the effect of T_time Max time on the same

RajathSwaroop, 2018-05-08:
When I use the rtabmap hand held mapping demo I get this error in the rtabmapviz command terminal: "[ERROR] (2018-05-08 14:02:28.999) MainWindow.cpp:2085::updateMapCloud() map id of node -1 not found!" What does this mean??

The rtabmap does not map anything after this error and just hangs

matlabbe, 2018-05-09:
For the error, this is fixed in https://github.com/introlab/rtabmap/commit/5e93803eef1b4643f8a499a571ecb1a36e5738fb
Otherwise, just disable Node filtering in Preferences->Node filtering.

For the thresholds, see this paper: [Appearance-Based Loop Closure Detection for Online Large-Scale and Long-Term Operation](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/b/bc/TRO2013.pdf). For similarity threshold (T_similarity), see section III-B. For loop closure threshold (T_loop), see section III-D. For max time (T_time), see section III-F. 

For T_loop, setting higher will decrease the chance of bad loop closures (false positives), but will decrease the number of good loop closures as well. In metric SLAM mode, it can be relatively low as loop closures are accepted only if motion estimation can be computed.

RajathSwaroop, 2018-05-09:
Thanks, the node filtering solution seems to work and I could get rid of the error. However I am facing a new issue after that. When I map a region and I move the camera in vertical direction (pitch) the rtabmap constructs multiple surfaces instead of building a nice map. I tried reinstalling the entire rtabmap standalone + ros after completely removing rtabmap from the computer but still have this issue.

Picture here describes the error.
[image]

RajathSwaroop, 2018-05-09:
If I wish to give an initial position to the rtabmap before it starts mapping so it maps starting from my given co-ordinates x,y,z and ypr is that possible?? When I run the rtabmap I see initial co-ordinates x,y,z and ypr as 0. Where do I change these?

matlabbe, 2018-05-10:
For the map errors, it depends on the sensor you are using, some will have more noise than others at different distance.

It is possible to set an initial pose, for odometry node, set "initial_pose" string parameter with format "x y z roll pitch yaw".

You can subscribe to images directly if you have enough bandwidth. See this [answer](https://answers.ros.org/question/289845/rtabmap_ros-bandwith/?answer=290495#post-id-290495) for some comparisons.

cheers,
Mathieu

RajathSwaroop, 2018-05-10:
I am getting this error when I run the rtabmap launch file on ROS.

[rtabmap/rtabmap-2] process has died [pid 18694, exit code -11, cmd /home/evodyne/catkin_ws/devel/lib/rtabmap_ros/rtabmap rgb/image:=/camera/color/image_raw depth/image:=/camera/depth/image_rect_raw rgb/camera_info:=/camera/color/camera_info rgbd_image:=rgbd_image left/image_rect:=/stereo_camera/left/image_rect_color right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info right/camera_info:=/stereo_camera/right/camera_info scan:=/scan scan_cloud:=/scan_cloud user_data:=/user_data user_data_async:=/user_data_async odom:=odom __name:=rtabmap __log:=/home/evodyne/.ros/log/4760f7fe-521b-11e8-ac34-b827ebf080fe/rtabmap-rtabmap-2.log].
log file: /home/evodyne/.ros/log/4760f7fe-521b-11e8-ac34-b827ebf080fe/rtabmap-rtabmap-2*.log

matlabbe, 2018-05-10:
rtabmap node has died, you won't get any maps. Try [starting the node in gdb](http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB) to have more info.

RajathSwaroop, 2018-05-10:
I just did that, Nothing changed. I did not get any prompt or new kind of error. The same error repeats

matlabbe, 2018-05-10:
Yes, gdb won't fix the error, it is just to see more information about the error. You can use "bt" (backtrace) when it crash to get more logs.

RajathSwaroop, 2018-05-10:
Hi I solved that error. Some installation that I did just a while ago had removed the ros image transport and the ros compressed image transport. I had to install those back and the rtabmap is working fine now.

However I get this waning " Costmap2DROS transform timeout. Current time: 1525992706.0815, global_pose stamp: 1525992702.8539, tolerance: 0.5000
[ WARN] [1525992707.081549222]: Costmap2DROS transform timeout. Current time: 1525992707.0815, global_pose stamp: 1525992702.8539, tolerance: 0.5000" is it because of my transform or the rtabmap?

RajathSwaroop, 2018-05-11:
Is there a way of forcing it to not subscribe to the compressed topics of the camera apart from setting compressed false in launch file or the argument?

matlabbe, 2018-05-11:
By default with rtabmap.launch, compressed is [false](https://github.com/introlab/rtabmap_ros/blob/master/launch/rtabmap.launch#L74). It should not subscribe to compressed topics. With "compressed:=false rgbd_sync:=true", you should have something like this:
```
/rtabmap/rtabmap subscribed to (approx sync):
  /rtabmap/rgbd_image,
  /rtabmap/odom_info
```
Do you restart the roscore between trials? to make sure all rosparam are cleared? 

The Costmap2DROS warnings are coming from move_base, you may look at move_base's code to know why.

Can you please list the commands you are using to start all launch files? It will be easier to test them here. Show also "rqt_graph" to better see how nodes are connected.

cheers,
Mathieu

RajathSwaroop, 2018-05-11:
I got the issue resolved when I restarted the roscore. Probably the rosparams were not cleared. Thanks

closed 2018-05-11 by RajathSwaroop

RajathSwaroop, 2018-05-12:
Is there a way to clear all rosparams without restarting the roscore?

matlabbe, 2018-05-13:
See https://answers.ros.org/question/31711/delete-multiple-ros-parameters-at-once/, in particular the clear_params approach.

0dmf0, 2020-04-09:
> For Poisson reconstruction, increase the depth to get more polygons (try 9 or 10). Note however it will be more sensible to noise.
>
> TSDF options are only available if rtabmap is built with [...]

how to use cpu-tsdf ? Thx

matlabbe, 2020-04-17:
If you install [cpu-tsdf](https://github.com/sdmiller/cpu_tsdf), cmake would be able to [find it](https://github.com/introlab/rtabmap/blob/6074e84653ea544450da38d6fb213d1e6175944b/CMakeLists.txt#L480). If not installed in default paths, cmake could find CPUTSDFConfig.cmake with:
```bash
$ cd rtabmap/build
$ cmake -DCPUTSDF_DIR=/path/to/CPUTSDFConfig.cmake/directory ..
```
Make sure you see this in cmake status before compiling:
```
With CPUTSDF              = YES (License: BSD)
```