introlab/rtabmap_ros #239: rtabmap/rtabmapviz-4 died when run roslaunch rtabmap_ros rgbd_mapping

LZDSJTU, 2018-04-26:
I have complied rtabmap and rtabmap_ros successfully. However, when I run 
roslaunch rtabmap_ros rgbd_mapping_kinect2.launch resolution:=qhd
It cannot work.
The error is:
tabmap/rtabmapviz-4] process has died [pid 1203, exit code -11, cmd /home/liang/project_ws/devel/lib/rtabmap_ros/rtabmapviz -d /home/liang/project_ws/src/rtabmap_ros/launch/config/rgbd_gui.ini rgb/image:=/kinect2/qhd/image_color_rect depth/image:=/kinect2/qhd/image_depth_rect rgb/camera_info:=/kinect2/qhd/camera_info __name:=rtabmapviz __log:=/home/liang/.ros/log/643bb2b2-491d-11e8-8d5e-00dbdf68d8eb/rtabmap-rtabmapviz-4.log].
log file: /home/liang/.ros/log/643bb2b2-491d-11e8-8d5e-00dbdf68d8eb/rtabmap-rtabmapviz-4*.log

LZDSJTU, 2018-04-26:
My environment is:
ubuntu16.04
ROS kinetic

LZDSJTU, 2018-04-26:
I change PCL1.8 to PCL1.7. And when I run:
roslaunch rtabmap_ros rgbd_mapping_kinect2.launch resolution:=qhd
The rtabmap can start(when I used PCL1.8, rtabmap cannot start)
But now the new error occurs:
[FATAL] (2018-04-26 17:00:16.477) OptimizerG2O.cpp:1092::optimizeBA() Condition (optimizer.verifyInformationMatrices()) not met!
[rtabmap/visual_odometry-2] process has died [pid 28058, exit code -6, cmd /home/liang/project_ws/devel/lib/rtabmap_ros/rgbd_odometry rgb/image:=/kinect2/qhd/image_color_rect depth/image:=/kinect2/qhd/image_depth_rect rgb/camera_info:=/kinect2/qhd/camera_info __name:=visual_odometry __log:=/home/liang/.ros/log/354c9864-4930-11e8-8d5e-00dbdf68d8eb/rtabmap-visual_odometry-2.log].

matlabbe, 2018-04-27:
For the first error, it is because when building from source PCL 1.8, they automatically compile with "-march=native" flag. When detecting this, rtabmap now builds also with "**-march=native**" to avoid Eigen problems with PCL. However, by doing so, other rtabmap's dependencies like GTSAM or g2o using also Eigen must be also rebuilt from source with "**-march=native**" (in both of these projects there are CMake options to activate this). It would be nicer if PCL could build by default without "**-march=native**" to avoid rebuilding everything else (depending on Eigen) with "**-march=native**". The PCL 1.7 binaries are not built with "**-march=native**", which avoids this issue.

Is g2o built from source or is it the binaries ros-kinetic-libg2o? Note sure why you get this error (optimizer.verifyInformationMatrices()), is this happen systematically or randomly? I have difficulty to reproduce the problem. To disable SBA from odometry, add `<param name="OdomF2M/BundleAdjustment" type="string" value="0"/>`  to launch file under visual odometry node.

cheers,
Mathieu

ElderWanng, 2019-04-26:
> For the first error, it is because when building from source PCL 1.8, they automatically compile with "-march=native" flag. When detecting this, rtabmap now builds also with "**-march=native**" to [...]

Hi, 
I want to use pcl 1.8 ( because 1.7 is too old to use with Ubuntu16 or higher ). How can I disable the    `-march=native` when cmake? I didn't find the related option in ccmake.

THX!

matlabbe, 2019-04-27:
It should be disabled on PCL side, maybe by commenting this line: https://github.com/PointCloudLibrary/pcl/blob/73f6af1642752267818d2280d35d57a7316275a0/cmake/pcl_find_sse.cmake#L13