introlab/rtabmap_ros #241: ROS melodic release

matlabbe, 2018-05-01:
Currently blocked by some repos: http://repositories.ros.org/status_page/blocked_releases_melodic.html?q=id%3D%22rtabmap_ros%22&s=6&r=1

nathangeorge1, 2019-08-09:
Hi @matlabbe,

Is there an estimate on when Melodic might be released?

matlabbe, 2019-08-16:
Yes it is released, thx to point out. Ialso [updated](https://github.com/introlab/rtabmap_ros/commit/5b3fe2fcf16fade46629ea2209721501b863e987) the README. 

cheers,
Mathieu

closed 2019-08-16 by matlabbe

W4li8, 2022-12-06:
Hi @matlabbe, I cannot find both rtabmap packages for my jetson nano, any developments on this issue?

E: Unable to locate package ros-melodic-rtabmap
E: Unable to locate package ros-melodic-rtabmap-ros

Thx

W4li8, 2022-12-06:
Here is a compiler output, it's either this or segfault (swap memory is already max)
```
nvidia@jetson:~/Developer/Libraries/rtabmap/build$ make -j6
[  4%] Built target rtabmap_utilite
Scanning dependencies of target imagesJoiner
Scanning dependencies of target extractObject
[  5%] Built target res_tool
[  5%] Building CXX object tools/ImagesJoiner/CMakeFiles/imagesJoiner.dir/main.cpp.o
[  5%] Building CXX object tools/ExtractObject/CMakeFiles/extractObject.dir/main.cpp.o
[  6%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Memory.cpp.o
[  7%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Rtabmap.cpp.o
[  7%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Statistics.cpp.o
[  7%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/RtabmapThread.cpp.o
[  8%] Linking CXX executable ../../bin/rtabmap-imagesJoiner
[  8%] Built target imagesJoiner
[  8%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/DBDriver.cpp.o
^Ccorelib/src/CMakeFiles/rtabmap_core.dir/build.make:186: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/DBDriver.cpp.o' failed
tools/ExtractObject/CMakeFiles/extractObject.dir/build.make:62: recipe for target 'tools/ExtractObject/CMakeFiles/extractObject.dir/main.cpp.o' failed
make[2]: *** [tools/ExtractObject/CMakeFiles/extractObject.dir/main.cpp.o] Interrupt
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/DBDriver.cpp.o] Interrupt
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:162: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/Memory.cpp.o' failed
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/Memory.cpp.o] Interrupt
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:90: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/Rtabmap.cpp.o' failed
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/Rtabmap.cpp.o] Interrupt
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:114: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/RtabmapThread.cpp.o' failed
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/RtabmapThread.cpp.o] Interrupt
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:138: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/Statistics.cpp.o' failed
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/Statistics.cpp.o] Interrupt
CMakeFiles/Makefile2:666: recipe for target 'tools/ExtractObject/CMakeFiles/extractObject.dir/all' failed
make[1]: *** [tools/ExtractObject/CMakeFiles/extractObject.dir/all] Interrupt
CMakeFiles/Makefile2:267: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Interrupt
Makefile:151: recipe for target 'all' failed
make: *** [all] Interrupt
```

matlabbe, 2022-12-07:
arm64 binaries exist: http://repositories.ros.org/status_page/melodic_default.html?q=rtabmap

You are building to many files at the same time, do:
```
make --j1
```

Then for catkin:
```
catkin_make -j1
```

W4li8, 2022-12-07:
Hi, I didn't mention I tried the -j1 flag.
```
nvidia@jetson:~/Developer/Libraries/rtabmap/build$ make -j1
[  6%] Built target rtabmap_utilite
[  6%] Built target res_tool
[  6%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Rtabmap.cpp.o
c++: internal compiler error: Segmentation fault (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:90: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/Rtabmap.cpp.o' failed
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/Rtabmap.cpp.o] Error 4
CMakeFiles/Makefile2:267: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
```
Now, I see the binaries exist indeed but how do I get them? 
```
nvidia@jetson:~$ sudo apt install ros-melodic-rtabmap ros-melodic-rtabmap-ros 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-melodic-rtabmap
E: Unable to locate package ros-melodic-rtabmap-ros
```
I can see other packages, just nothing related to ros-melodic although my sources list states "deb http://packages.ros.org/ros/ubuntu bionic main". This goes beyond rtabmap now but do you have ideas of what to look for?

W4li8, 2022-12-07:
Hi, I didn't mention I tried the -j1 flag.
```
nvidia@jetson:~/Developer/Libraries/rtabmap/build$ make -j1
[  6%] Built target rtabmap_utilite
[  6%] Built target res_tool
[  6%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Rtabmap.cpp.o
c++: internal compiler error: Segmentation fault (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
corelib/src/CMakeFiles/rtabmap_corhttps://packages.ubuntu.com/search?keywords=ros&searchon=names&suite=bionic&section=alle.dir/build.make:90: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/Rtabmap.cpp.o' failed
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/Rtabmap.cpp.o] Error 4
CMakeFiles/Makefile2:267: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
```
Now, I see the binaries exist indeed but how do I get them? 
```
nvidia@jetson:~$ sudo apt install ros-melodic-rtabmap ros-melodic-rtabmap-ros 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-melodic-rtabmap
E: Unable to locate package ros-melodic-rtabmap-ros
```
I can see other packages, just nothing related to ros-melodic although my sources list states "deb http://packages.ros.org/ros/ubuntu bionic main". This goes beyond rtabmap now but do you have ideas of what to look for? Anything ros-melodic related does not list in [ubuntu packages](https://packages.ubuntu.com/search?keywords=ros&searchon=names&suite=bionic&section=all)

matlabbe, 2022-12-07:
Is your jetson on armhf (32 bits) or arm64? the binaries are not created for 32 bits, only for 64 bits. rtabmap is in package file: http://packages.ros.org/ros/ubuntu/dists/bionic/main/binary-arm64/Packages

Maybe your "sudo apt update" failed for some reasons?

W4li8, 2022-12-07:
Jetson is 64bits, I see the packages are indeed all there... "sudo apt update" doesn't indicate any failure
```
nvidia@jetson:~$ sudo apt update
Hit:1 http://packages.microsoft.com/repos/code stable InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports bionic InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease
Hit:5 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease
Hit:6 https://librealsense.intel.com/Debian/apt-repo bionic InRelease
Hit:7 https://repo.download.nvidia.com/jetson/common r32.6 InRelease
Get:8 https://repo.download.nvidia.com/jetson/t210 r32.6 InRelease [2,547 B]   
Hit:9 http://packages.ros.org/ros/ubuntu bionic InRelease                      
Fetched 2,547 B in 5s (484 B/s)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
```

matlabbe, 2022-12-11:
Can you install other ros packages?

W4li8, 2022-12-12:
Hi, problem solved (for now at least) by a repetitive combination of `sudo apt clean`, `sudo apt update` and updating the `/etc/apt/sources.list` (from the terminal instead of through the Software and Updates app seemed to help). 

It's been a bit strange... Thanks for helping me eliminate possible sources of error @matlabbe

W4li8, 2022-12-12:
> Can you install other ros packages?

Before I decided to delete ROS for a fresh install thinking that could help, I could see and install a few... but I think those were core packages only