introlab/rtabmap_ros #243: rtabmap failing to build on Indigo Trusty i386 due to openni abstraction layer

tfoote, 2018-05-14:
Apparently i386 has been declared unsupported by OpenNI

The failing job is here: http://build.ros.org/view/Ibin_uT32/job/Ibin_uT32__rtabmap__ubuntu_trusty_i386__binary/

From: http://build.ros.org/view/Ibin_uT32/job/Ibin_uT32__rtabmap__ubuntu_trusty_i386__binary/42/console
```
01:24:11 [ 10%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/CameraThread.cpp.o
01:24:11 cd /tmp/binarydeb/ros-indigo-rtabmap-0.17.0/obj-i686-linux-gnu/corelib/src && /usr/bin/i686-linux-gnu-g++   -DEIGEN_USE_NEW_STDVECTOR -DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET -DFLANN_STATIC -DFREENECT_DASH_INCLUDES -DOCTOMAP_PRE_18 -DPCL_OMP -Dqh_QHpointer -Drtabmap_core_EXPORTS -g -O2 -fstack-protector --param=ssp-buffer-size=4 -Wformat -Werror=format-security -DNDEBUG -D_FORTIFY_SOURCE=2  -fmessage-length=0  -Wno-deprecated -fopenmp -std=c++11 -fPIC -I/usr/include/opencv -I/usr/include/vtk-5.8 -I/tmp/binarydeb/ros-indigo-rtabmap-0.17.0/utilite/include -I/tmp/binarydeb/ros-indigo-rtabmap-0.17.0/corelib/src/../include -I/tmp/binarydeb/ros-indigo-rtabmap-0.17.0/corelib/src -I/tmp/binarydeb/ros-indigo-rtabmap-0.17.0/obj-i686-linux-gnu/corelib/src -I/usr/include/pcl-1.7 -I/usr/include/eigen3 -I/usr/include/ni -I/usr/include/dc1394 -I/opt/ros/indigo/include -I/usr/include/suitesparse    -Wall -Wno-unknown-pragmas -o CMakeFiles/rtabmap_core.dir/CameraThread.cpp.o -c /tmp/binarydeb/ros-indigo-rtabmap-0.17.0/corelib/src/CameraThread.cpp
01:24:12 In file included from /usr/include/ni/XnOS.h:28:0,
01:24:12                  from /usr/include/pcl-1.7/pcl/io/openni_camera/openni.h:47,
01:24:12                  from /usr/include/pcl-1.7/pcl/io/openni_camera/openni_depth_image.h:44,
01:24:12                  from /tmp/binarydeb/ros-indigo-rtabmap-0.17.0/corelib/src/../include/rtabmap/core/CameraRGBD.h:42,
01:24:12                  from /tmp/binarydeb/ros-indigo-rtabmap-0.17.0/corelib/src/CameraThread.cpp:31:
01:24:12 /usr/include/ni/XnPlatform.h:74:3: error: #error OpenNI Platform Abstraction Layer - Unsupported Platform!
01:24:12   #error OpenNI Platform Abstraction Layer - Unsupported Platform!
01:24:12    ^
01:24:12 In file included from /usr/include/pcl-1.7/pcl/io/openni_camera/openni.h:47:0,
01:24:12                  from /usr/include/pcl-1.7/pcl/io/openni_camera/openni_depth_image.h:44,
01:24:12                  from /tmp/binarydeb/ros-indigo-rtabmap-0.17.0/corelib/src/../include/rtabmap/core/CameraRGBD.h:42,
01:24:12                  from /tmp/binarydeb/ros-indigo-rtabmap-0.17.0/corelib/src/CameraThread.cpp:31:
01:24:12 /usr/include/ni/XnOS.h:52:3: error: #error OpenNI OS Abstraction Layer - Unsupported Platform!
01:24:12   #error OpenNI OS Abstraction Layer - Unsupported Platform!
01:24:12    ^
01:24:12 In file included from /usr/include/pcl-1.7/pcl/io/openni_camera/openni.h:50:0,
01:24:12                  from /usr/include/pcl-1.7/pcl/io/openni_camera/openni_depth_image.h:44,
01:24:12                  from /tmp/binarydeb/ros-indigo-rtabmap-0.17.0/corelib/src/../include/rtabmap/core/CameraRGBD.h:42,
01:24:12                  from /tmp/binarydeb/ros-indigo-rtabmap-0.17.0/corelib/src/CameraThread.cpp:31:
01:24:12 /usr/include/ni/XnCppWrapper.h:10019:76: error: macro "XN_VALIDATE_NEW" passed 4 arguments, but takes just 2
```
Since this is declared unsupported I'd usually suggest that we blacklist this package but unfortunately we  don't support blacklisting i386 without blocking amd64 because this is an unusual differentiation.

matlabbe, 2018-05-14:
I observed the same problem, and reported it in https://github.com/introlab/rtabmap/issues/282

It is more clear what could be the problem when looking at the Kinetic i386 build: http://build.ros.org/job/Kbin_uX32__rtabmap__ubuntu_xenial_i386__binary/

The only "big" change between May 1 (success) and May 2 (fail) is described in https://github.com/introlab/rtabmap/issues/282 (added libg2o dependency)

I may remove g2o dependency to not have a c++11 build, or see if we can build with the indigo/kinetic version of g2o without c++11.

cheers,
Mathieu

matlabbe, 2018-05-14:
I just pushed a new version on rosdistro including the fix above.

matlabbe, 2018-05-15:
Kinetic build has passed: http://build.ros.org/job/Kbin_uX32__rtabmap__ubuntu_xenial_i386__binary/

For indigo (http://build.ros.org/view/Ibin_uT32/job/Ibin_uT32__rtabmap__ubuntu_trusty_i386__binary/), it fails when reaching the 2 hours timeout. Looking at the console, there is one hour lost just after getting libproj-dev package:
```
15:40:59 Get:725 http://archive.ubuntu.com/ubuntu/ trusty/universe proj-data i386 4.8.0-2ubuntu2 [2,067 kB]
15:40:59 Get:726 http://archive.ubuntu.com/ubuntu/ trusty/universe libproj0 i386 4.8.0-2ubuntu2 [111 kB]
15:40:59 Get:727 http://archive.ubuntu.com/ubuntu/ trusty/universe libproj-dev i386 4.8.0-2ubuntu2 [133 kB]
16:49:32 
Extracting templates from packages: 4%
Extracting templates from packages: 8%
Extracting templates from packages: 12%
Extracting templates from packages: 16%
Extracting templates from packages: 20%
Extracting templates from packages: 24%
Extracting templates from packages: 28%
Extracting templates from packages: 33%
Extracting templates from packages: 37%
Extracting templates from packages: 41%
Extracting templates from packages: 45%
Extracting templates from packages: 49%
Extracting templates from packages: 53%
Extracting templates from packages: 57%
Extracting templates from packages: 61%
Extracting templates from packages: 66%
Extracting templates from packages: 70%
Extracting templates from packages: 74%
Extracting templates from packages: 78%
Extracting templates from packages: 82%
Extracting templates from packages: 86%
Extracting templates from packages: 90%
Extracting templates from packages: 94%
Extracting templates from packages: 99%
Extracting templates from packages: 100%
16:49:32 Preconfiguring packages ...
16:49:32 Fetched 248 MB in 1min 48s (2,294 kB/s)
16:49:37 (Reading database ... 15814 files and directories currently installed.)
16:49:37 Preparing to unpack .../libcomerr2_1.42.9-3ubuntu1.3_i386.deb ...
16:49:37 Unpacking libcomerr2:i386 (1.42.9-3ubuntu1.3) over (1.42.9-3ubuntu1) ...
```
Not sure what happened.

tfoote, 2018-05-15:
That's due to https://github.com/ros-infrastructure/ros_buildfarm/issues/535 The apt-get install between download and extracting seems to significantly hit the slowed down functions. I retriggered it and it went onto the same executor so hopefully it can take advantage of the docker layer caching and skip a bunch of this bootstrapping: http://build.ros.org/view/Ibin_uT32/job/Ibin_uT32__rtabmap__ubuntu_trusty_i386__binary/44/ it will be nip/tuck on if it can pass. We've been seeing up to a 4x slow down due to that issue.

matlabbe, 2018-05-15:
19 min success! Thx

closed 2018-05-15 by matlabbe