introlab/rtabmap_ros #244: Can't locate node rtabmap

scanf30, 2018-05-16:
I'm new in ROS and have installed rtabmap and rtamap-ros via apt in a jetsonTx2. i did a launch file with the zed wrapper, my package and the rtabmap-ros and it can not locate the node.

This is the launch file:

`<?xml version="1.0"?>
<launch>
	<env name="ROS_NAMESPACE" value="camera"/>
	<include file="$(find zed_wrapper)/launch/zed_camera.launch"/>
 	<node name="foo" pkg="sleep_launch" type="itanimulli.sh"/> 
	<include file="/opt/ros/kinetic/share/rtabmap_ros/launch/rtabmap.launch">
		<arg name="rtabmap_args" value="--delete_db_on_start"/>
		<arg name="depth_topic" value="/camera/depth/depth_registered"/>
		<arg name="frame_id" value="zed_center"/>
		<arg name="approx_sync" value="false"/>
		<arg name="visual_odometry" value="false"/>
		<arg name="odom_topic" value="/camera/odom"/>
	</include>
</launch>`

Then i ran the launch file without the zed and my package and this is my output: 

... logging to /home/nvidia/.ros/log/9a7c832e-58b8-11e8-ab5b-00044b8d7c46/roslaunch-tegra-ubuntu-10577.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tegra-ubuntu:38392/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/approx_sync: False
 * /rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/database_path: ~/.ros/rtabmap.db
 * /rtabmap/rtabmap/frame_id: zed_center
 * /rtabmap/rtabmap/odom_frame_id: 
 * /rtabmap/rtabmap/queue_size: 10
 * /rtabmap/rtabmap/subscribe_depth: True
 * /rtabmap/rtabmap/subscribe_scan: False
 * /rtabmap/rtabmap/subscribe_scan_cloud: False
 * /rtabmap/rtabmap/subscribe_stereo: False
 * /rtabmap/rtabmap/wait_for_transform_duration: 0.2
 * /rtabmap/rtabmapviz/approx_sync: False
 * /rtabmap/rtabmapviz/frame_id: zed_center
 * /rtabmap/rtabmapviz/odom_frame_id: 
 * /rtabmap/rtabmapviz/queue_size: 10
 * /rtabmap/rtabmapviz/subscribe_depth: True
 * /rtabmap/rtabmapviz/subscribe_odom_info: False
 * /rtabmap/rtabmapviz/subscribe_scan: False
 * /rtabmap/rtabmapviz/subscribe_scan_cloud: False
 * /rtabmap/rtabmapviz/subscribe_stereo: False
 * /rtabmap/rtabmapviz/wait_for_transform_duration: 0.2

NODES
  /rtabmap/
    rtabmap (rtabmap_ros/rtabmap)
    rtabmapviz (rtabmap_ros/rtabmapviz)

ROS_MASTER_URI=http://localhost:11311

ERROR: cannot launch node of type [rtabmap_ros/rtabmap]: can't locate node [rtabmap] in package [rtabmap_ros]
ERROR: cannot launch node of type [rtabmap_ros/rtabmapviz]: can't locate node [rtabmapviz] in package [rtabmap_ros]

matlabbe, 2018-05-16:
Did you [source](http://wiki.ros.org/kinetic/Installation/Ubuntu#kinetic.2BAC8-Installation.2BAC8-DebEnvironment.Environment_setup) `setup.bash`?

matlabbe, 2018-05-18:
It seems working. Try again your launch file, maybe the ros path was not updated yet.

scanf30, 2018-05-20:
still not working. I tried uninstall ROS and delete my ws and didn't work.

matlabbe, 2018-05-20:
In your previous post, rtabmap node test was working, then you also said the zed example worked too (which uses rtabmap.launch file). If you create a launch file like this (trying to break down your initial post):

```xml
<launch>
	<include file="$(find rtabmap_ros)/launch/rtabmap.launch">
		<arg name="rtabmap_args" value="--delete_db_on_start"/>
	</include>
</launch>
```

what errors do you have?

hariharan382, 2022-08-12:
same error for me too

umbnich, 2023-04-01:
Same error here, any solutions?

shrinivas96, 2023-04-30:
Not exactly sure when the change occurred (but surely at least somewhere after Feb 2023, I assume), but for people now not able to find the `rtabmap` node under `rtabmap_ros` package, that is because this node has now moved to `rtabmap_slam` package. The new changes are highlighted in the migration guide on the [rtabmap_ros wiki](http://wiki.ros.org/rtabmap_ros) page.