introlab/rtabmap_ros #247: Mapping deviates from GPS mapping

aswingururaj, 2018-05-23:
Hi,
     I am using rtabmap_ros with a ZED camera for a project on driverless car. The mapping deviates drastically from that of the GPS at speeds greater than 10 km/h. I tried changing the parameters that you have listed in issue #102 . But still the map deviates very much after it loses odometry for the first time after launching.What can be possible reasons for this? Is there any other parameters that I need to change to get better results?
Aswin

matlabbe, 2018-05-25:
Can you record zed data and share it? What is the framerate of the odometry? Which rtabmap version do you use?

aswingururaj, 2018-05-28:
https://drive.google.com/file/d/1diho8FUIiswRrAB5QsjozSEl3GN33w7M/view 
It starts at 60 Hz and slows down as we increase the speed of the car. we are using rtabmap_ros. The red screens don't appear that often but still the mapping is largely deviated from the GPS odometry. 
[image]

aswingururaj, 2018-05-31:
We will try to rotate that. We are using 0.11 version. We changed the following parameters:
Odom/Strategy=1       
Odom/GuessMotion=true 
Vis/EstimationType=1
Vis/CorType=1
Odom/ResetCountdown=10
We tried using stereo_odometry but there was an error(Didnot receive data for 5 seconds).As of now we are using only the resetcountdown parameter. After the odometry gets lost the plotting is full of errors.Since we are using it in a car it is not possible to come back to the frame at which the odometry was lost. Is there any way by which we can account for the frames that are lost in that time?

matlabbe, 2018-05-31:
I suggest to update to 0.17 (binaries are available on Indigo/Kinetic) to get better odometry first. Then for odometry lost, the new version (of rgbd_odometry or stereo_odometry) can publish odometry only if there is a valid velocity with parameter `publish_null_when_lost` set to false. You can use [robot_localization](http://wiki.ros.org/robot_localization) package to integrate the odometry velocity to get a continuous pose estimation like:
```xml
<rosparam param="odom0_config">[false, false, false,
                                false, false, false,
                                true, true, true, <!-- Linear velocity -->
                                true, true, true, <!-- Angular velocity -->
                                false, false, false]</rosparam>
```
When using `robot_localization`, don't forget to set `publish_tf` of rgbd_odometry or stereo_odometry at false to avoid TF problems ('robot_localization' would be publish also odom->base_link).

cheers,
Mathieu