introlab/rtabmap_ros #248: After update to version 0.17.2 cannot open rtabmap node

ruifanwu, 2018-05-31:
Hi there,

I just updated rtabmap_ros to 0.17.2. But I cannot open rtabmap node.

I am using kinect v2, and when I do `roslaunch rgbd_mapping_kinect2.launch`. A warning popped out, there is no image in loop closure detection window. And there is no transform between map->odom. I checked the rosnode in rqt, there is no rtabmap node. Before the update, rtabmap_ros worked just fine. And I did not change anything before or after the update. The output of `roslaunch rgbd_mapping_kinect2.launch` is as follows:

`administrator@jackal:~/catkin_ws/src/rtabmap_ros/launch$ roslaunch rgbd_mapping_kinect2.launch 
... logging to /home/administrator/.ros/log/6e9b35b4-648b-11e8-804c-9689319299bc/roslaunch-jackal-17725.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jackal:54827/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /rtabmap/rtabmap/GFTT/BlockSize: 5
 * /rtabmap/rtabmap/GFTT/MinDistance: 5
 * /rtabmap/rtabmap/approx_sync: False
 * /rtabmap/rtabmap/frame_id: kinect2_base_link
 * /rtabmap/rtabmap/subscribe_depth: True
 * /rtabmap/rtabmapviz/approx_sync: False
 * /rtabmap/rtabmapviz/frame_id: kinect2_base_link
 * /rtabmap/rtabmapviz/subscribe_depth: True
 * /rtabmap/rtabmapviz/subscribe_odom_info: True
 * /rtabmap/visual_odometry/GFTT/BlockSize: 5
 * /rtabmap/visual_odometry/GFTT/MinDistance: 5
 * /rtabmap/visual_odometry/approx_sync: False
 * /rtabmap/visual_odometry/frame_id: kinect2_base_link

NODES
  /rtabmap/
    rtabmap (rtabmap_ros/rtabmap)
    rtabmapviz (rtabmap_ros/rtabmapviz)
    visual_odometry (rtabmap_ros/rgbd_odometry)
  /
    kinect2_base_link (tf/static_transform_publisher)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[kinect2_base_link-1]: started with pid [17743]
process[rtabmap/visual_odometry-2]: started with pid [17744]
process[rtabmap/rtabmap-3]: started with pid [17755]
process[rtabmap/rtabmapviz-4]: started with pid [17756]
[ INFO] [1527742277.722655324]: Starting node...
[ INFO] [1527742277.741481195]: Initializing nodelet with 4 worker threads.
[ INFO] [1527742277.796478681]: Initializing nodelet with 4 worker threads.
[ INFO] [1527742277.899716752]: Starting node...
[ INFO] [1527742277.943456801]: /rtabmap/rtabmap(maps): map_filter_radius          = 0.000000
[ INFO] [1527742277.943512932]: /rtabmap/rtabmap(maps): map_filter_angle           = 30.000000
[ INFO] [1527742277.943541592]: /rtabmap/rtabmap(maps): map_cleanup                = true
[ INFO] [1527742277.943563802]: /rtabmap/rtabmap(maps): map_negative_poses_ignored = true
[ INFO] [1527742277.943597106]: /rtabmap/rtabmap(maps): map_negative_scan_ray_tracing = true
[ INFO] [1527742277.943619661]: /rtabmap/rtabmap(maps): cloud_output_voxelized     = true
[ INFO] [1527742277.943649957]: /rtabmap/rtabmap(maps): cloud_subtract_filtering   = false
[ INFO] [1527742277.943672717]: /rtabmap/rtabmap(maps): cloud_subtract_filtering_min_neighbors = 2
[ INFO] [1527742277.951186921]: /rtabmap/rtabmap(maps): octomap_tree_depth         = 16
[ INFO] [1527742277.978310754]: rtabmap: frame_id      = kinect2_base_link
[ INFO] [1527742277.978552817]: rtabmap: map_frame_id  = map
[ INFO] [1527742277.978660749]: rtabmap: tf_delay      = 0.050000
[ INFO] [1527742277.978757907]: rtabmap: tf_tolerance  = 0.100000
[ INFO] [1527742277.978858492]: rtabmap: odom_sensor_sync   = false
[ INFO] [1527742278.018282181]: rtabmapviz: Using configuration from "/home/administrator/catkin_ws/src/rtabmap_ros/launch/config/rgbd_gui.ini"
[ INFO] [1527742278.043249601]: Odometry: frame_id               = kinect2_base_link
[ INFO] [1527742278.043339764]: Odometry: odom_frame_id          = odom
[... 53 lines cut ...]
   /rtabmap/odom,
   /kinect2/qhd/image_color_rect,
   /kinect2/qhd/image_depth_rect,
   /kinect2/qhd/camera_info,
   /rtabmap/odom_info
[ INFO] [1527742280.311710001]: rtabmap 0.17.2 started...
[ INFO] [1527742280.535264994]: Odom: quality=700, std dev=0.030802m|0.054012rad, update time=0.113628s
[ INFO] [1527742280.716539102]: Odom: quality=685, std dev=0.013440m|0.024930rad, update time=0.121960s
[ INFO] [1527742280.825918421]: Odom: quality=685, std dev=0.011229m|0.000032rad, update time=0.106650s
[ INFO] [1527742281.036623222]: Odom: quality=697, std dev=0.020077m|0.046875rad, update time=0.098918s
[ INFO] [1527742281.218089265]: Odom: quality=712, std dev=0.014399m|0.040663rad, update time=0.104841s
[rtabmap/rtabmap-3] process has died [pid 17755, exit code -11, cmd /home/administrator/catkin_ws/devel/lib/rtabmap_ros/rtabmap --delete_db_on_start rgb/image:=/kinect2/qhd/image_color_rect depth/image:=/kinect2/qhd/image_depth_rect rgb/camera_info:=/kinect2/qhd/camera_info __name:=rtabmap __log:=/home/administrator/.ros/log/6e9b35b4-648b-11e8-804c-9689319299bc/rtabmap-rtabmap-3.log].
log file: /home/administrator/.ros/log/6e9b35b4-648b-11e8-804c-9689319299bc/rtabmap-rtabmap-3*.log
[ INFO] [1527742281.514779944]: Odom: quality=701, std dev=0.011797m|0.000032rad, update time=0.118101s
[ INFO] [1527742281.644237177]: Odom: quality=691, std dev=0.013655m|0.033069rad, update time=0.100385s
[ INFO] [1527742281.819614221]: Odom: quality=715, std dev=0.010526m|0.000032rad, update time=0.105454s
[ INFO] [1527742282.056783770]: Odom: quality=679, std dev=0.018841m|0.038257rad, update time=0.098143s
^C[ INFO] [1527742282.415742842]: Odom: quality=701, std dev=0.010353m|0.000032rad, update time=0.099883s
[rtabmap/rtabmapviz-4] killing on exit
[ INFO] [1527742282.481752714]: rtabmapviz: ctrl-c catched! Exiting Qt app...
[rtabmap/visual_odometry-2] killing on exit
[kinect2_base_link-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
`

matlabbe, 2018-05-31:
I cannot reproduce the problem, did you update both rtabmap and rtabmap_ros?

ruifanwu, 2018-05-31:
Yes, I updated both rtabmap and rtabmap_ros

ruifanwu, 2018-05-31:
Hi matlabbe,
I just found in the output of launch command the `rtabmap: Database version = "0.17.1"`. Is this the cause of this problem?

ruifanwu, 2018-05-31:
Hi Mathieu,

I finally know how to use gdb. And I will post the gdb.txt. It is quite long. 

[gdb.txt](https://github.com/introlab/rtabmap_ros/files/2059456/gdb.txt)

matlabbe, 2018-05-31:
The Statistics object have changed, which could cause the error for `rtabmap_ros::infoToROS`:
```
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffcce29700 (LWP 17217)]
0x00007fffbd32c170 in std::vector<int, std::allocator<int> > uKeys<int, float>(std::map<int, float, std::less<int>, std::allocator<std::pair<int const, float> > > const&) () from /home/administrator/catkin_ws/devel/lib//librtabmap_ros.so
#0  0x00007fffbd32c170 in std::vector<int, std::allocator<int> > uKeys<int, float>(std::map<int, float, std::less<int>, std::allocator<std::pair<int const, float> > > const&) ()
   from /home/administrator/catkin_ws/devel/lib//librtabmap_ros.so
#1  0x00007fffbd31c40c in rtabmap_ros::infoToROS(rtabmap::Statistics const&, rtabmap_ros::Info_<std::allocator<void> >&) ()
   from /home/administrator/catkin_ws/devel/lib//librtabmap_ros.so
...
```

 Can you copy/paste results of:
```bash
$ ldd /home/administrator/catkin_ws/devel/lib//librtabmap_ros.so
```
to see if it is a linker problem (not using the right libraries).

cheers,
Mathieu

ruifanwu, 2018-05-31:
Hi Mathieu,

The result of above comment is quite long, so I put it into a file named `error.docx`

[error.docx](https://github.com/introlab/rtabmap_ros/files/2059826/error.docx)

matlabbe, 2018-05-31:
It looks like the linker is using rtabmap binaries from `ros-indigo-rtabmap` before those you built from source.
```
librtabmap_core.so.0.17 => /opt/ros/indigo/lib/x86_64-linux-gnu/librtabmap_core.so.0.17 (0x00007fe2eb1b0000)
librtabmap_utilite.so.0.17 => /opt/ros/indigo/lib/x86_64-linux-gnu/librtabmap_utilite.so.0.17 (0x00007fe2eaf92000)
```
Remove rtabmap binaries:
```bash
$ apt-get remove ros-indigo-rtabmap
```

ruifanwu, 2018-05-31:
Hi Mathieu,

Thank you very much! It is working properly right now. I really appreciate your time for helping me find the reason. Thank you!

closed 2018-05-31 by matlabbe