introlab/rtabmap_ros #250: A question about recent published paper.

DingFeng0103, 2018-06-05:
Hi.

I have read your recent paper published at 2017. Your lab combined 2 planning module to help navigation as I understand. The paper says we can find the code here. So is the code already merged into rtabmap_ros? if not when will it be ready?

Thanks 
Feng

matlabbe, 2018-06-07:
Yes the code is there. However, there is no tutorial yet. For example, the patrol behavior is there: https://github.com/introlab/rtabmap_ros/blob/master/scripts/patrol.py

and az3_nav.launch uses it too: https://github.com/introlab/rtabmap_ros/blob/master/launch/azimut3/az3_nav.launch

The main parameters related to planning:
```
<param name="use_action_for_goal" type="bool" value="true"/>
<remap from="goal_out"  to="current_goal"/>	
<remap from="move_base" to="/planner/move_base"/>
<remap from="grid_map"  to="/map"/>
```
`goal_out` is for visualization in RVIZ to see actual rtabmap goal. `use_action_for_goal` is used to connect TPP to MPP with ros [actionlib](http://wiki.ros.org/actionlib). `grid_map` is the local map. `move_base` is a remap to actual move_base node name.

The user goal of the paper is connected to "/rtabmap/goal" input topic of rtabmap node. In code, TPP is implemented directly in rtabmap node, while MPP is external (which is move_base).

cheers,
Mathieu

DingFeng0103, 2018-06-08:
Thanks very much for you reply. I am really interested in your project and hope I can put it to use to my robot.

DingFeng0103, 2018-06-20:
Is it possible that I only use TPP to navigate with simple instruction or paramter chance?

DingFeng0103, 2018-06-21:
Hi, 
As I am reading az3_nav.launch file and trying to configure parameters to get good result for my robot, I found the default parameter for rtabmap_ros:
`Param: Vis/FeatureType = "6"                               [0=SURF 1=SIFT 2=ORB 3=FAST/FREAK 4=FAST/BRIEF 5=GFTT/FREAK 6=GFTT/BRIEF 7=BRISK 8=GFTT/ORB 9=KAZE.]`
so default feature is GFIT/BRIEF instead of SURF however in your az3_nav.launch you set:
`<param name="SURF/HessianThreshold" type="string" value="500"/>`
So did you change some default parameters for rtabmap_ros but didn't update it on `rtabmap --params`
Or the az3_nav.launch is outdated as I saw it is updated 2 years ago while your paper is published last year.

DingFeng0103, 2018-06-21:
Hi,
As I am reading az3_nav.launch file and trying to configure parameters to get good result for my robot, I found the default parameter for rtabmap_ros:
Param: Vis/FeatureType = "6" [0=SURF 1=SIFT 2=ORB 3=FAST/FREAK 4=FAST/BRIEF 5=GFTT/FREAK 6=GFTT/BRIEF 7=BRISK 8=GFTT/ORB 9=KAZE.]
so default feature is GFIT/BRIEF instead of SURF however in your az3_nav.launch you set:
<param name="SURF/HessianThreshold" type="string" value="500"/>
So did you change some default parameters for rtabmap_ros but didn't update it on rtabmap --params
Or the az3_nav.launch is outdated as I saw it is updated 2 years ago while your paper is published last year.

Best
Feng