introlab/rtabmap_ros #251: remote mapping delay

RajathSwaroop, 2018-06-05:
I tried performing remote mapping by streaming the camera RGBD data from a local machine at 30 frame per second and run the rtabmap on another machine that receives the RGB and depth data with a delay of 50 ms. The rtabmap fails to perform real-time mapping, becomes laggy and the rate of mapping goes down. Any way of getting real-time mapping using remote mapping solution?

matlabbe, 2018-06-07:
What is the network bandwidth used? Is it stable?

RajathSwaroop, 2018-06-07:
Yeah my network is stable. The bandwidth utilization for the compressed topics of color and depth when checked with rostopic bw command was 1.8MB per second and for the uncompressed topic was around 6-7 MB per second

matlabbe, 2018-06-18:
Is rtabmap or rgbd_odometry node that is laggy? What frame rate do you have of rtabmap/odometry or rtabmap/mapData topics?

Even a little network lag could make ROS not able to synchronize topics for a couple of seconds.

cheers,
Mathieu

RajathSwaroop, 2018-06-21:
@matlabbe the issue was resolved. I was publishing depth topic as /compressed instead of /compressedDepth. I figured it out and changed the depth compressed topic publish type to compressedDepth and things were fine.

I wanted to know if I can record a map and then change the frame_id of it before saving the map? How do I improve the number of loop closures it does to get a fine map for small regions ? (like a clear table or rack mapped)

RajathSwaroop, 2018-06-25:
When I changed the detection rate to 20 Hz and run the rtabmap on an intel i7 8-core CPU with 32 Gb ram, most of the times I get a warning message that says processing time is more than detection time and some times the processing time goes as high as 0.2 seconds. Is this normal behavior? a low processing and detection rate affects the resolution of map building??

RajathSwaroop, 2018-06-27:
[FATAL] (2018-06-27 13:16:14.258) OptimizerG2O.cpp:1092::optimizeBA() Condition (optimizer.verifyInformationMatrices()) not met!

I get this and the process gets killed for rtabmap. This started happening after I compiled rtabmap with g2o and launch the rtabmap ros node. How do I solve it??

RajathSwaroop, 2018-06-29:
Is there a way of re-initializing the map's co-ordinates after the map has been built? Like shifting the map from 0,0,0 as center to 9,9,8 as center

matlabbe, 2018-07-06:
To shift the map, you could add a static_transform between "world" and "map" of 9,9,8.

For the g2o error, see http://official-rtab-map-forum.67519.x6.nabble.com/g2o-error-on-loop-closure-td4707.html#a4725

For the detection rate, 20 Hz is quite high, and yes processing time can go over 200 ms, in particular when the vocabulary sohuld be rebuilt or if there are many loop closures found + graph optimization. I don't know what is your application, but most of mines don't really require more than 1 or 2 Hz detection rate. 

cheers,
Mathieu

RajathSwaroop, 2018-07-10:
I fixed the G2O error and trying to map with the bundle adjustment On. If i map using base_footprint as the frame id and I loose odometry when I try localizing in the map (RED screen). "Reset odom to pose" will set the base_footprint to the given co-ordinate or will it set the "odom" to that given co-ordinate?

matlabbe, 2018-07-10:
It will reset `base_footprint` pose accordingly to `odom` frame.

RajathSwaroop, 2018-08-27:
The function to save or export the 3-D map generated in rtabmap to ply or pcd file has an option of choosing between dense point cloud and organized point cloud. I tried generating organized point cloud for a map of a room and the ply  file has height = 1. Is this an error or is this the expected result for such a case?

matlabbe, 2018-08-29:
To make sure the output is also organized, the clouds should not be assembled and voxel filter should be disabled:
[image]

Uncheck binary and select PCD format to better see the result:
```
# .PCD v0.7 - Point Cloud Data file format
VERSION 0.7
FIELDS x y z rgb normal_x normal_y normal_z curvature
SIZE 4 4 4 4 4 4 4 4
TYPE F F F U F F F F
COUNT 1 1 1 1 1 1 1 1
WIDTH 160
HEIGHT 120
VIEWPOINT 0 0 0 1 0 0 0
POINTS 19200
DATA ascii
nan nan nan 2143289344 nan nan nan nan
nan nan nan 2143289344 nan nan nan nan
nan nan nan 2143289344 nan nan nan nan
nan nan nan 2143289344 nan nan nan nan
...
```
The NaN values are for invalid points.

RajathSwaroop, 2018-09-11:
Thanx for the solution. When I try to build Rtabmap on a different machine I get this error when I do a cmake . . 
Qt5WebKitWidgetsConfig.cmake not found.

I did a sudo apt-get install libqt5webkit but nothing changed. Is it dependent on Qt5? can u pls provide a solution to fix it

matlabbe, 2018-09-11:
What is the OS of the different machine? Try deleting the build directory to reset cmake cache. RTAB-Map doesn't use Qt5WebKit.

RajathSwaroop, 2018-09-11:
The OS is Ubuntu 16.0.4 same in both. I tried deleting the build directory and rebuild but that does'nt work.

RajathSwaroop, 2018-09-12:
Also what could be the reason for this error

"Odometry.cpp:313::process() Guess from motion is set but dt is invalid! Odometry is then computed without guess. (dt=-0.491127 previous transform=xyz=0.031484,-0.141772,-0.406672 rpy=0.007029,-0.086933,0.030537)"

matlabbe, 2018-09-14:
What are the VTK paths? Maybe having only one VTK installed may solve the problem.