introlab/rtabmap_ros #259: Camera nodelet_manager process dies

frushton, 2018-07-06:
I am launching the demo_two_kinects.launch file, and rtabmapviz pops up and begins collecting data and showing the point cloud, but after about 20 seconds, the camera nodelet manager (always for camera 2) dies. Also, just before data starts being collected, I also get this error: 

[ERROR] (2018-07-06 12:25:20.213) OdometryF2M.cpp:200::computeTransform() Odometry bundle adjustment doesn't work with multi-cameras. It is disabled.

Does anyone know what's going on here?

``... logging to /home/e4e/.ros/log/4f3975f8-8152-11e8-b98f-02429def565d/roslaunch-e4e-XPS-15-9560-9462.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://e4e-XPS-15-9560:46391/

SUMMARY
========

PARAMETERS
 * /camera1/camera1_nodelet_manager/num_worker_threads: 4
 * /camera1/depth_rectify_depth/interpolation: 0
 * /camera1/depth_registered_rectify_depth/interpolation: 0
 * /camera1/disparity_depth/max_range: 4.0
 * /camera1/disparity_depth/min_range: 0.5
 * /camera1/disparity_registered_hw/max_range: 4.0
 * /camera1/disparity_registered_hw/min_range: 0.5
 * /camera1/disparity_registered_sw/max_range: 4.0
 * /camera1/disparity_registered_sw/min_range: 0.5
 * /camera1/driver/data_skip: 0
 * /camera1/driver/debug: False
 * /camera1/driver/depth_camera_info_url: 
 * /camera1/driver/depth_frame_id: camera1_depth_opt...
 * /camera1/driver/depth_registration: True
 * /camera1/driver/device_id: #1
 * /camera1/driver/diagnostics_max_frequency: 30.0
 * /camera1/driver/diagnostics_min_frequency: 30.0
 * /camera1/driver/diagnostics_tolerance: 0.05
 * /camera1/driver/diagnostics_window_time: 5.0
 * /camera1/driver/enable_depth_diagnostics: False
 * /camera1/driver/enable_ir_diagnostics: False
 * /camera1/driver/enable_rgb_diagnostics: False
 * /camera1/driver/rgb_camera_info_url: 
 * /camera1/driver/rgb_frame_id: camera1_rgb_optic...
 * /camera2/camera2_nodelet_manager/num_worker_threads: 4
 * /camera2/depth_rectify_depth/interpolation: 0
 * /camera2/depth_registered_rectify_depth/interpolation: 0
 * /camera2/disparity_depth/max_range: 4.0
 * /camera2/disparity_depth/min_range: 0.5
 * /camera2/disparity_registered_hw/max_range: 4.0
 * /camera2/disparity_registered_hw/min_range: 0.5
 * /camera2/disparity_registered_sw/max_range: 4.0
 * /camera2/disparity_registered_sw/min_range: 0.5
 * /camera2/driver/data_skip: 0
 * /camera2/driver/debug: False
 * /camera2/driver/depth_camera_info_url: 
 * /camera2/driver/depth_frame_id: camera2_depth_opt...
 * /camera2/driver/depth_registration: True
 * /camera2/driver/device_id: #2
 * /camera2/driver/diagnostics_max_frequency: 30.0
 * /camera2/driver/diagnostics_min_frequency: 30.0
 * /camera2/driver/diagnostics_tolerance: 0.05
 * /camera2/driver/diagnostics_window_time: 5.0
 * /camera2/driver/enable_depth_diagnostics: False
 * /camera2/driver/enable_ir_diagnostics: False
 * /camera2/driver/enable_rgb_diagnostics: False
 * /camera2/driver/rgb_camera_info_url: 
 * /camera2/driver/rgb_frame_id: camera2_rgb_optic...
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /rtabmap/rgbd_odometry/Odom/FillInfoData: true
 * /rtabmap/rgbd_odometry/Odom/Strategy: 0
 * /rtabmap/rgbd_odometry/OdomF2M/MaxSize: 1000
[... 546 lines cut ...]
[camera1/disparity_depth-19] killing on exit
[camera1/depth_registered_metric-18] killing on exit
[camera1/depth_registered_hw_metric_rect-17] killing on exit
[camera1/points_xyzrgb_hw_registered-16] killing on exit
[camera1/depth_registered_rectify_depth-15] killing on exit
[camera1/depth_registered_sw_metric_rect-14] killing on exit
[camera1/points_xyzrgb_sw_registered-13] killing on exit
[camera1/register_depth_rgb-12] killing on exit
[camera1/depth_points-11] killing on exit
[camera1/depth_metric-10] killing on exit
[camera1/depth_metric_rect-9] killing on exit
[camera1/ir_rectify_ir-7] killing on exit
[camera1/depth_rectify_depth-8] killing on exit
[camera1/rgb_rectify_color-6] killing on exit
[camera1/rgb_rectify_mono-5] killing on exit
[camera1/rgb_debayer-4] killing on exit
[camera1/driver-3] killing on exit
[camera1/camera1_nodelet_manager-2] killing on exit
device 0x7f4bf4036ce0 not found in linked list for this context!
rtabmap: Saving database/long-term memory...done! (located at /home/e4e/.ros/rtabmap.db, 7 MB)
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

matlabbe, 2018-07-06:
The bundle error is more a warning, odometry is still computed. For the crash, it is related to freenect:
```bash
what(): freenect_process_events error
```
See also https://github.com/ros-drivers/freenect_stack/issues/40

closed 2018-07-11 by frushton

zhw5772, 2018-09-27:
@frushton 

Have solved the problem？

I also have the error：
[ERROR] (2018-07-06 12:25:20.213) OdometryF2M.cpp:200::computeTransform() Odometry bundle adjustment doesn't work with multi-cameras. It is disabled

Could you tell me how to deal with it?

Thank you!