introlab/rtabmap_ros #260: hello,i can not use catkin_make while rtabmap_ros is in my workspace,here are the details

w8128835, 2018-07-10:
at rtabmap_ros/CMakeLists.txt:22 (find_package):
  By not providing "FindRTABMap.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "RTABMap", but
  CMake did not find one.

  Could not find a package configuration file provided by "RTABMap"
  (requested version 0.17.2) with any of the following names:

    RTABMapConfig.cmake
    rtabmap-config.cmake

  Add the installation prefix of "RTABMap" to CMAKE_PREFIX_PATH or set
  "RTABMap_DIR" to a directory containing one of the above files.  If
  "RTABMap" provides a separate development package or SDK, be sure it has
  been installed.

could  you please tell me how to deal with it?thanks a lot

matlabbe, 2018-07-10:
You should install rtabmap standalone libraries before trying to build rtabmap_ros. See **step 2** [here](https://github.com/introlab/rtabmap_ros#build-from-source).

matlabbe, 2018-07-11:
The version should be okay. What is the output when you do `make install` of the library?

w8128835, 2018-07-11:
oh,thank you,here is the last information of the make record in my terminal:
In file included from /home/eric/rtabmap/examples/NoEventsExample/main.cpp:32:0:
/usr/include/pcl-1.7/pcl/visualization/cloud_viewer.h:202:14: warning: ‘template<class> class std::auto_ptr’ is deprecated [-Wdeprecated-declarations]
         std::auto_ptr<CloudViewer_impl> impl_;
              ^
In file included from /usr/include/c++/5/bits/locale_conv.h:41:0,
                 from /usr/include/c++/5/locale:43,
                 from /usr/include/c++/5/iomanip:43,
                 from /opt/ros/kinetic/include/opencv-3.3.1-dev/opencv2/flann/lsh_table.h:40,
                 from /opt/ros/kinetic/include/opencv-3.3.1-dev/opencv2/flann/lsh_index.h:49,
                 from /opt/ros/kinetic/include/opencv-3.3.1-dev/opencv2/flann/all_indices.h:42,
                 from /opt/ros/kinetic/include/opencv-3.3.1-dev/opencv2/flann/flann_base.hpp:43,
                 from /opt/ros/kinetic/include/opencv-3.3.1-dev/opencv2/flann.hpp:48,
                 from /opt/ros/kinetic/include/opencv-3.3.1-dev/opencv2/opencv.hpp:62,
                 from /home/eric/rtabmap/corelib/include/rtabmap/core/CameraModel.h:31,
                 from /home/eric/rtabmap/corelib/include/rtabmap/core/SensorData.h:33,
                 from /home/eric/rtabmap/corelib/include/rtabmap/core/Rtabmap.h:34,
                 from /home/eric/rtabmap/examples/NoEventsExample/main.cpp:28:
/usr/include/c++/5/bits/unique_ptr.h:49:28: note: declared here
   template<typename> class auto_ptr;
                            ^
[ 99%] Building CXX object examples/NoEventsExample/CMakeFiles/noEventsExample.dir/moc_MapBuilder.cpp.o
[100%] Linking CXX executable ../../../bin/rtabmap-noEventsExample
[100%] Built target noEventsExample
eric@eric:~/rtabmap/build$ 

And here is the information about make install:
[ 95%] Built target rgbd_mapping
[ 98%] Built target wifi_mapping
[100%] Built target noEventsExample
Install the project...
-- Install configuration: "Release"
-- Installing: /home/eric/catkin_ws/devel/lib/rtabmap-0.17/RTABMapConfig.cmake
-- Installing: /home/eric/catkin_ws/devel/lib/rtabmap-0.17/RTABMapConfigVersion.cmake
-- Installing: /home/eric/catkin_ws/devel/share/rtabmap/package.xml
-- Installing: /home/eric/catkin_ws/devel/lib/librtabmap_utilite.so.0.17.3
。。。。。。。
-- Installing: /home/eric/catkin_ws/devel/bin/rtabmap-report
-- Set runtime path of "/home/eric/catkin_ws/devel/bin/rtabmap-report" to ""
eric@eric:~/rtabmap/build$ 

It is like this,is there any problem with it? it seems that i can install all the library,but while i fellowed the instruction of catkin_make rtabmap_ros in my workspace,the problem occurs like that:

CMake Error: Could not open file for write in copy operation /home/eric/catkin_ws/devel/share/rtabmap_ros/cmake/rtabmap_ros-msg-paths.cmake.tmp
CMake Error: : System Error: No such file or directory
CMake Error at /opt/ros/kinetic/share/genmsg/cmake/genmsg-extras.cmake:218 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  rtabmap_ros/CMakeLists.txt:96 (generate_messages)
CMake Error: Could not open file for write in copy operation /home/eric/catkin_ws/devel/share/rtabmap_ros/cmake/rtabmap_rosConfig.cmake.tmp
CMake Error: : System Error: No such file or directory
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:381 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  rtabmap_ros/CMakeLists.txt:130 (catkin_package)
CMake Error: Could not open file for write in copy operation /home/eric/catkin_ws/devel/share/rtabmap_ros/cmake/rtabmap_rosConfig-version.cmake.tmp
CMake Error: : System Error: No such file or directory
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:386 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  rtabmap_ros/CMakeLists.txt:130 (catkin_package)
-- WITH costmap_2d
-- WITH octomap
-- WITH rviz
-- Configuring incomplete, errors occurred!
See also "/home/eric/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/eric/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
eric@eric:~/catkin_ws$ 

I am so sorry to show you so many mistakes.Did i missed any dependency？

w8128835, 2018-07-11:
hello matlabbe,I have gitcloneed the rtabmap_ros in my workspace,but mistakes occur while i catkin_make -j1 like this:
CMake Error: Could not open file for write in copy operation /home/eric/catkin_ws/devel/share/rtabmap_ros/cmake/rtabmap_ros-msg-paths.cmake.tmp
....

After i checked the fold /home/eric/catkin_ws/devel/share,there is no fold named rtabmap_ros,but the git clone shows me ireceived all the data like this：
100% (6225/6225), 3.99 MiB | 858.00 KiB/s

Is there any possible that my system is ubuntu16.04 and lost some file during transmission？

matlabbe, 2018-07-11:
Those errors are ros related, msgs could not be created. Did you source both /opt/ros/kinetic/setup.bash and /home/eric/catkin_ws/devel/setup.bash before doing catkin_make?

w8128835, 2018-07-11:
yes,i sourced it  before,and I checked it in my ~/.bashrc just now.But it still can't be catkin_make.Should i change my system to ubuntu 14.04,or follow the instruction of installing it in ROS Distribution with command :sudo apt-get install ros-kinetic-rtabmap-ros?Will it occur some conflicts?

matlabbe, 2018-07-11:
If you already have ros-kinetic-rtabmap-ros, you don't need to build it from source. If you build from source, make sure to uninstall ros-kinetic-rtabmap and ros-kinetic-rtabmap-ros as stated in the build from source instructions.

w8128835, 2018-07-11:
i am so sorry to interupt you again,i just get the command:
sudo apt-get remove ros-kinetic-rtabmap ros-kinetic-rtabmap-ros
in the source instructions.Is this commend enough?
Should i delete the file ratmap in my local home folder and ratmap_ros in my catkin_ws workspace as well?

PrabhjotKaurGosal, 2020-07-18:
I got the same error while I was trying to build a package that was dependent on rtabmap. I was able to find the root cause for my problem. See details below.
The package that I created and was building dependent on the packages that are mentioned in the package.xml below.
[https://github.com/introlab/rtabmap_ros/blob/master/package.xml]
As can be seen, one of the dependencies is 'rtabmap'. 

So, I got the exact same error as @w8128835 . I was able to fix it by changing the name of the this package to rtabmap_ros in both package.xml and CMakeLists.txt. The reason for doing that was, I had installed the rtabmap package for Melodic using this command:  $ sudo apt-get install ros-melodic-rtabmap-ros
This package was installed on my machine as rtabmap_ros. 

I believe if we install rtabmap_ros using sudo apt-get, we don't have to again install by 'Build from Source' as mentioned in [https://github.com/introlab/rtabmap_ros]. Build from Source and installing using apt-get are two ways of installing the same package I believe.

So, if we install by using apr-get, this package gets installed as rtabmap_ros. But if we install by using 'Buid from SOUrce', then, it gets installed as rtabmap. This is just my 2 cents and I would like to confirm with @matlabbe if my observation is correct.

matlabbe, 2020-07-31:
You don't have to build from source if you get rtabmap from apt-get. Note that there is two packages: `rtabmap` and `rtabmap_ros`. If you have a ros package, you are more likely to depend on `rtabmap_ros` instead of `rtabmap`.

When building from source, you have one project called [rtabmap](https://github.com/introlab/rtabmap_ros) (the library) and the other one [rtabmap](https://github.com/introlab/rtabmap_ros) (the ros interface).