introlab/rtabmap_ros #261: Can't run rtabmap_ros with or without pointcloud_to_depthimage

AnGPU, 2018-07-10:
Hi Matt 

I decided to reinstall the ros wrapper because I need the ros connection to modify the gain and exposure to get better performance in outdoors, but as I manage to control it several weeks ago, now I cant make it work. 

The standalone application works perfectly but I erased it to avoid it configuration interfere somehow with the rtabmap_ros and following https://github.com/introlab/rtabmap_ros doesn't shows any errors during installation.
From the first approach I can't obtain response from rtabmap/odominfo and from the second :
"roslaunch realsense_camera zr300_nodelet_rgbd.launch"
"rosrun rtabmap_ros pointcloud_to_depthimage cloud:=/camera/depth/points camera_info:=/camera/rgb/camera_info image_raw:=/camera/depth/points/image_raw image:=/camera/depth/points/image _approx:=false _fill_holes_size:=2"

I just get this
[image]
Without any GUI
I can see the fixed frame id is empty but if this is the problem I can't see why it doesn't show this from the code
ROS_FATAL("fixed_frame_id should be set when using approximate "
					"time synchronization (approx=true)! If the robot "
					"is moving, it could be \"odom\". If not moving, it "
					"could be \"base_link\".");

matlabbe, 2018-07-10:
You launch with `_approx:=false`, the error will not be shown. With approximate synchronization, a fixed frame is required (e.g., odom) to project correctly the cloud where the RGB camera is at a different time. So what is the problem? Do you have any outputs with `rostopic hz`? In your screenshot, the remapping of image_raw seems a little messed up.

AnGPU, 2018-07-11:
As you can see in the new image 
[image]

You are right, with approx=true there is the error message, by the way, the problem is that I can't use the application. It doesn't appear. I just thought this could be the reason but all the topics within this commands seems to run. 
Yes, I didn't notice that but that capture was made when changing the topics after trying with the suggested in the previous post 
(rosrun rtabmap_ros pointcloud_to_depthimage cloud:=/camera/depth/points camera_info:=/camera/rgb/camera_info image_raw:=/camera/depth/points/image_raw image:=/camera/depth/points/image _approx:=false _fill_holes_size:=2)
with the same results, all "running" but without the application.

matlabbe, 2018-07-11:
> the problem is that I can't use the application. It doesn't appear.

In your example, you didn't start rtabmap. By "It doesn't appear", do you mean when your start:
`$ roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start" depth_topic:=/camera/depth/points/image_raw`
there is nothing shown in rtabmapviz? If so, are there any errors/warnings on the terminal?

AnGPU, 2018-07-11:
I learnt about ros during the last months and when I did this before I wrote the steps to start the rtabmap ros but I didn't use this for months so I didn't understand/remember the whole process and I thought just the "rosrun part" was the second approach.
Sorry and thanks for your time.

closed 2018-07-12 by AnGPU